sllurp/sllurp #41: Fix parameter shifting bug.

jettan, 2015-02-17:
When startAccess is called from another function, sometimes the parameters of startAccess would be shifted to the right for no apparent reason. This also happens when changing accessSpecs frequently and can be seen by this log message:

2015-02-17 16:58:11,477 sllurp.llrp: ERROR: panic(): ('ADD_ACCESSSPEC failed',)
2015-02-17 16:58:11,477 sllurp.llrp: ERROR: 6
2015-02-17 16:58:11,477 sllurp.llrp: ERROR: Traceback (most recent call last):
Failure: __builtin__.int: 6

Basically, it received the STATE of the system instead of the real 1st parameter of startAccess causing the weird behavior.

ewfuentes, 2015-02-17:
Hah, this is interesting, I was dealing with this exact same problem yesterday.

The parameters are not getting shifted.  What is happening is that the reader is refusing to update the AccessSpec.  When Sllurp gets the ADD_ACCESSSPEC_RESPONSE with the failure, it calls the errback function with it's state (LLRP.py:303).  I'll try to get a log of this happening with verbose logging.

ewfuentes, 2015-02-17:
Here is the log:
```
DEBUG:sllurp.llrp:deserializing ADD_ACCESSSPEC_RESPONSE command
DEBUG:sllurp.llrp_proto:decode_AddROSpecResponse
DEBUG:sllurp.llrp_proto:decode_LLRPStatus
DEBUG:sllurp.llrp_proto:decode_LLRPStatus: 011f00560065003e4c4c5250205b3239395d203a202f2f416363657373537065632f416363657373537065634944203a20494420616c726561647920636f6e666967757265640121001000cf00c9012000080001012c
DEBUG:sllurp.llrp_proto:decode_LLRPStatus (type=287 len=86)
DEBUG:sllurp.llrp_proto:decode_FieldError
DEBUG:sllurp.llrp_proto:decode_ParameterError
DEBUG:sllurp.llrp_proto:decode_ParameterError (type=289 len=16 data='\x00\xcf\x00\xc9\x01 \x00\x08\x00\x01\x01,')
DEBUG:sllurp.llrp_proto:decode_FieldError
DEBUG:sllurp.llrp_proto:decode_FieldError (type=288 len=8 data='\x00\x01\x01,')
DEBUG:sllurp.llrp_proto:decode_ParameterError
DEBUG:sllurp.llrp:done deserializing ADD_ACCESSSPEC_RESPONSE command
DEBUG:sllurp.llrp:LLRPMessage received in state 6: <ADD_ACCESSSPEC_RESPONSE>
	<Ver>1</Ver>
	<Type>50</Type>
	<ID>0</ID>
	<LLRPStatus>
		<StatusCode>FieldError</StatusCode>
		<ErrorDescription>LLRP [299] : //AccessSpec/AccessSpecID : ID already configured</ErrorDescription>
		<ParameterError>
			<ParameterType>207</ParameterType>
			<ErrorCode>201</ErrorCode>
			<FieldError>
				<FieldNum>1</FieldNum>
			</FieldError>
		</ParameterError>
	</LLRPStatus>
</ADD_ACCESSSPEC_RESPONSE>
DEBUG:sllurp.llrp:starting message callbacks for ADD_ACCESSSPEC_RESPONSE
DEBUG:sllurp.llrp:done with message callbacks for ADD_ACCESSSPEC_RESPONSE
DEBUG:sllurp.llrp:in handleMessage(ADD_ACCESSSPEC_RESPONSE), there are 1 Deferreds
DEBUG:sllurp.llrp:running 1 Deferreds for ADD_ACCESSSPEC_RESPONSE; isSuccess=False
ERROR:sllurp.llrp:panic(): ('ADD_ACCESSSPEC failed',)
ERROR:sllurp.llrp:6
ERROR:sllurp.llrp:Traceback (most recent call last):
Failure: __builtin__.int: 6
```

You can see my workaround at ewfuentes/sllurp@7ce56c7d2f143ae9878968f289e1931934b93557

I would call startAccess with an access count of 1 and I would wait for the response before calling startAccess again.

At a higher level, I don't particularly like how the accessSpec api looks but I don't really have any suggestions on how to improve it.  It is definitely better than nothing, but it still requires you to understand the LLRP protocol.

jettan, 2015-02-17:
Interesting... I'll definitely try it out.
On another note, are you using an accessSpec to send over data in blockWrite by any chance?
particularly this line: DEBUG:sllurp.llrp_proto:decode_ParameterError (type=289 len=16 data='\x00\xcf\x00\xc9\x01 \x00\x08\x00\x01\x01,')
Because to my understanding blockWrite has never worked for me.

ewfuentes, 2015-02-17:
To be honest I hadn't heard of blockWrite until now.  I'm still pretty new to this RFID stuff.  I did find this though: http://www.rfidjournal.com/blogs/experts/entry?10614

It basically says that both the reader and the tag must support block write for it to be successful.  I don't know what tag IC you are using, but it might be worth looking into.  Block write is an optional part of the spec.

jettan, 2015-02-18:
I'm using sllurp exclusively for the WISP5 (see http://wisp5.wikispaces.com/WISP+Home)

ransford, 2015-02-23:
Hmm, the message `//AccessSpec/AccessSpecID : ID already configured` suggests that the AccessSpec you're adding has the same AccessSpec ID as the AccessSpec that's currently enabled.  Note that `LLRPClient.startAccess()` currently hard-codes `accessSpecID = 1`.  If you call startAccess twice in a row without a `DELETE_ACCESSSPEC` command to the reader in between, this is the error message that will arise.  I'm pretty sure you'll need to intersperse `DELETE_ACCESSSPEC` commands if you're going to generate a series of AccessSpecs.

I don't really understand what you mean by parameter shifting.  Do you mean that a call like `f(a,b,c)` is actually calling `f(b,c)` or `f(x,a,b,c)`?  Or something else?

ransford, 2015-02-23:
In other words, I think that for a series of writes representing a big transmission, you'll have to do a sequence like:

    1. ADD_ACCESSSPEC (with parameter representing the data you want to write)
    2. ENABLE_ACCESSSPEC
    3. (wait for the write to happen)
    4. (optionally) DISABLE_ACCESSSPEC
    5. DELETE_ACCESSSPEC
    6. Go to step 1

jettan, 2015-02-23:
Yes, this is exactly what has been happening Ben.
That sequence is implemented in my own (not streamlined for pull request yet) fork: https://github.com/jettan/sllurp

The big difference is in llrp.py and access.py (the last one is a bit ugly atm).
In llrp.py, you will find that f(a, b, c) will have become f(x, a, b, c).

Using the accessSpecStopTrigger only fixes it temporarily and makes the sequence go smooth. The occurences of parameter shifting has become less, but can still occur.

ransford, 2015-02-23:
I recommend using `traceback.print_tb` to print a stack trace when you detect that the bug has occurred.  Are the erroneous calls happening via `task.deferLater`?

jettan, 2015-02-23:
Nope, they happen way before.
Remember how llrp.py has a ClientFactory and LLRPClient.
https://github.com/jettan/sllurp/blob/master/sllurp/llrp.py
Everything passed from ClientFactory to the first call in LLRPClient goes perfectly, but when passing around parameters within the LLRPClient is going wrong for some reason, starting in line 712 (Don't worry about line 708 and 709 for now since that fixes the compatibility with accessSpecStopTrigger)
This is also the reason why I added a placeHolderArgument in send_DELETE_ACCESSSPEC in line 615.

The task.deferLater is the only thing working, since appending it using any other way immediately returns an error when invoked (which also turns out the parameter has been shifted).

ransford, 2015-04-07:
Is this bug still occurring?

jettan, 2015-04-07:
It is, but it's rare now as long as you control the flow by using the
AccessSpecStopParam thingie.

closed 2017-09-13 by ransford