sllurp/sllurp #44: Get info from the reader

osk386, 2015-05-07:
Hi ransford,

I'm on a project to build a tag based management system that runs on the web, and I use Sllurp to get data from tags, that's cool!. all is working well. I haven't much knowledge about RFID systems and I'am a beginner of python, I just want you ask a simple question, is possible to get data from reader inside the TagReportData?

[image]

What happens is I got two readers, I am reading from both an listing the tags, I get the antenna id from each reader, no problem. But I need to know from where reader comes data.

I mean maybe  host IP, mac address or some id that identifies a reader. If this can be done, how can I get that data into tag data returned by the report.

I tested these parameters but no one distinguishes one reader from another

                                     'EnableROSpecID': True,
                                     'EnableSpecIndex': True,
                                     'EnableInventoryParameterSpecID': True,
                                     'EnableAntennaID': True,
                                     'EnableChannelIndex': True,
                                     'EnablePeakRRSI': True,
                                     'EnableFirstSeenTimestamp': True,
                                     'EnableLastSeenTimestamp': True,
                                     'EnableTagSeenCount': True,
                                     'EnableAccessSpecID': True

I appreciate your answer, sorry for my english, is not my native language.

Thank you.

ransford, 2015-08-04:
Sorry, I didn't notice this issue!  Grab the hostname or IP address from the `peername` field of the LLRPMessage provided to your tag report callback.  For example, you could modify `tagReportCallback` in `sllurp/inventory.py` to say

    logger.info('saw tag(s): {}'.format(pprint.pformat(tags)))

instead of

ransford, 2015-08-04:
Closing since the above comment should solve the problem.  Reopen if you have trouble!

closed 2015-08-04 by ransford