sllurp/sllurp #45: Support for ThingMagic Astra-EX reader

futnuh, 2015-07-19:
Hi Folks.  I'd like to get sslurp working with the ThingMagic Astra EX reader (http://www.thingmagic.com/index.php/integrated-readers/445-astra-ex).  The reader specs state that it supports "LLRP v1.0.1 with multi-protocol extensions".  Here's what I get when trying to use inventory "out-of-the-box":

$ ./inventory 10.0.80.63
2015-07-19 00:41:01,892 sllurp: INFO: log level: INFO
2015-07-19 00:41:01,893 sllurp.llrp: INFO: connecting...
2015-07-19 00:41:01,896 sllurp.llrp: INFO: will reset reader state on connect
2015-07-19 00:41:01,896 sllurp.llrp: INFO: will start inventory on connect
2015-07-19 00:41:01,896 sllurp.llrp: INFO: using antennas: [1]
2015-07-19 00:41:01,897 sllurp.llrp: INFO: connected to ('10.0.80.63', 5084)
2015-07-19 00:41:02,104 sllurp.llrp: INFO: requested modulation: M4
2015-07-19 00:41:02,104 sllurp.llrp: INFO: using reader mode: {'C': 0, 'MaxTari': 25000, 'M': 3, 'PIE': 1700, 'StepTari': 12500, 'R': 1, 'FLM': 0, 'MinTari': 12500, 'ModeIdentifier': 1, 'BDR': 250000, 'Mod': 2}
2015-07-19 00:41:02,105 sllurp.llrp: INFO: stopping politely
2015-07-19 00:41:02,118 sllurp.llrp: INFO: reader finished inventory
2015-07-19 00:41:02,118 sllurp.llrp: INFO: starting inventory
2015-07-19 00:41:02,128 sllurp.llrp: CRITICAL: Error FieldError adding ROSpec: ADD_ROSPEC: Error: Transmit Power index 107 is out of range for Ant 1 in NA region, valid range is 1 to 101

Any help is much appreciated.  The "Mercury" API that ThingMagic supplies only supports Java, C and C#.  I could probably generate python wrappers for the C interface but if I can get sslurp working, that will save me a lot of time.

ewfuentes, 2015-07-19:
Try using:

    ./inventory 10.0.80.63 -X 100

The -X flag sets the transmit power.

futnuh, 2015-07-20:
Thanks Erick, I ran the above and, at least intermittently, was able to see a few tags.  Unfortunately most of the time the inventory script doesn't return any tags.  It just sits there after the last line below:

```
$ bin/inventory 10.0.80.63 -X 100
2015-07-19 21:27:56,100 sllurp: INFO: log level: INFO
2015-07-19 21:27:56,100 sllurp.llrp: INFO: connecting...
2015-07-19 21:27:56,104 sllurp.llrp: INFO: will reset reader state on connect
2015-07-19 21:27:56,104 sllurp.llrp: INFO: will start inventory on connect
2015-07-19 21:27:56,104 sllurp.llrp: INFO: using antennas: [1]
2015-07-19 21:27:56,105 sllurp.llrp: INFO: connected to ('10.0.80.63', 5084)
2015-07-19 21:27:56,396 sllurp.llrp: INFO: requested modulation: M4
2015-07-19 21:27:56,397 sllurp.llrp: INFO: using reader mode: {'C': 0, 'MaxTari': 25000, 'M': 3, 'PIE': 1700, 'StepTari': 12500, 'R': 1, 'FLM': 0, 'MinTari': 12500, 'ModeIdentifier': 1, 'BDR': 250000, 'Mod': 2}
2015-07-19 21:27:56,397 sllurp.llrp: INFO: stopping politely
2015-07-19 21:27:56,410 sllurp.llrp: INFO: reader finished inventory
2015-07-19 21:27:56,410 sllurp.llrp: INFO: starting inventory
```

After a few minutes, I usually end the program with a ctrl-c.  As a check, I have used ThingMagic's web-based Java app (which seems to also connect on LLRP default port 5084) to confirm that the reader is finding tags.  I try the ThingMagic app, see a bunch of tags, disconnect so that the inventory script can connect, run the inventory script and get no tags.  Any thoughts what could be wrong?

In the hopes that it helps, I'm attaching a debug log for the same command as above.  I can't see anything problematic in terms of communication errors or deserialization of the protocol.  It makes me wonder if there is a problem with the configuration of the reader?  Again, any help is much appreciated.

```
2015-07-19 20:14:06,421 sllurp: DEBUG: log level: DEBUG
2015-07-19 20:14:06,421 sllurp.llrp: INFO: connecting...
2015-07-19 20:14:06,423 sllurp.llrp: INFO: will reset reader state on connect
2015-07-19 20:14:06,423 sllurp.llrp: INFO: will start inventory on connect
2015-07-19 20:14:06,424 sllurp.llrp: INFO: using antennas: [1]
2015-07-19 20:14:06,424 sllurp.llrp: INFO: connected to ('10.0.80.63', 5084)
2015-07-19 20:14:06,425 sllurp.llrp: DEBUG: got 32 bytes from reader: 043f000000200000006600f600160080000c00051b45ee30d2b0010000060000
2015-07-19 20:14:06,425 sllurp.llrp: DEBUG: expect 32 bytes (have 32)
2015-07-19 20:14:06,425 sllurp.llrp: DEBUG: deserializing READER_EVENT_NOTIFICATION command
2015-07-19 20:14:06,449 sllurp.llrp_proto: DEBUG: decode_ReaderEventNotification
2015-07-19 20:14:06,450 sllurp.llrp_proto: DEBUG: decode_ReaderEventNotificationData (type=246 len=22)
2015-07-19 20:14:06,451 sllurp.llrp_proto: DEBUG: decode_UTCTimestamp
2015-07-19 20:14:06,452 sllurp.llrp_proto: DEBUG: decode_UTCTimestamp (type=128 len=12)
2015-07-19 20:14:06,453 sllurp.llrp_proto: DEBUG: decode_ConnectionAttemptEvent
2015-07-19 20:14:06,453 sllurp.llrp_proto: DEBUG: decode_ConnectionAttemptEvent (type=256 len=6)
2015-07-19 20:14:06,454 sllurp.llrp_proto: DEBUG: decode_AntennaEvent
2015-07-19 20:14:06,454 sllurp.llrp: DEBUG: done deserializing READER_EVENT_NOTIFICATION command
2015-07-19 20:14:06,455 sllurp.llrp: DEBUG: LLRPMessage received in state 1: <READER_EVENT_NOTIFICATION>
	<Ver>1</Ver>
	<Type>63</Type>
	<ID>102</ID>
	<ReaderEventNotificationData>
		<ConnectionAttemptEvent>
			<Status>Success</Status>
		</ConnectionAttemptEvent>
	</ReaderEventNotificationData>
</READER_EVENT_NOTIFICATION>
2015-07-19 20:14:06,455 sllurp.llrp: DEBUG: starting message callbacks for READER_EVENT_NOTIFICATION
2015-07-19 20:14:06,455 sllurp.llrp: DEBUG: done with message callbacks for READER_EVENT_NOTIFICATION
2015-07-19 20:14:06,455 sllurp.llrp: DEBUG: in handleMessage(READER_EVENT_NOTIFICATION), there are 0 Deferreds
2015-07-19 20:14:06,455 sllurp.llrp: DEBUG: serializing GET_READER_CAPABILITIES command
2015-07-19 20:14:06,455 sllurp.llrp: DEBUG: serialized bytes: 04010000000b0000000000
2015-07-19 20:14:06,455 sllurp.llrp: DEBUG: done serializing GET_READER_CAPABILITIES command
2015-07-19 20:14:06,455 sllurp.llrp: DEBUG: state change: STATE_DISCONNECTED -> STATE_SENT_GET_CAPABILITIES
2015-07-19 20:14:06,459 sllurp.llrp: DEBUG: got 1401 bytes from reader: 040b0000060d00000000011f0008000000000089003d00024000000067ba000000300009342e31392e322e3431008b000800010000008d000800040004008c00090001000101008c00090002000101008e001c080100000000002000000010000000040000008000000001008f04fd00000001009004f500910008000101f4009100080002020d0091000800030226009100080004023f00910008000502580091000800060271009100080007028a00910008000802a300910008000902bc00910008000a02d500910008000b02ee00910008000c030700910008000d032000910008000e033900910008000f0352009100080010036b0091000800110384009100080012039d00910008001303b600910008001403cf00910008001503e80091000800160401009100080017041a0091000800180433009100080019044c00910008001a046500910008001b047e00910008001c049700910008001d04b000910008001e04c900910008001f04e200910008002004fb0091000800210514009100080022052d0091000800230546009100080024055f0091000800250578009100080026059100910008002705aa00910008002805c300910008002905dc00910008002a05f500910008002b060e00910008002c062700910008002d064000910008002e065900910008002f0672009100080030068b00910008003106a400910008003206bd00910008003306d600910008003406ef00910008003507080091000800360721009100080037073a0091000800380753009100080039076c00910008003a078500910008003b079e009 [... cut ...]
2015-07-19 20:14:31,347 sllurp.llrp: DEBUG: running 1 Deferreds for DELETE_ROSPEC_RESPONSE; isSuccess=True
2015-07-19 20:14:31,348 sllurp.llrp: DEBUG: state change: STATE_DISCONNECTED -> STATE_DISCONNECTED
2015-07-19 20:14:31,348 sllurp.llrp: INFO: lost connection: Connection to the other side was lost in a non-clean fashion: Connection lost.
2015-07-19 20:14:31,348 sllurp: INFO: total # of tags seen: 0
2015-07-19 20:14:31,348 sllurp.llrp: INFO: disconnecting
```

(Oh, by the way, the Astra doesn't support the "show" command used in the get_reader_config script.)

ewfuentes, 2015-07-20:
Ahh I see what happened with power level.  The reader advertises that it can do up to 31.5 dBm, but the built in regulatory restrictions will only allow it to do 30 dBm of power.  The default configuration will choose the max power that is advertised, hence the critical error that you saw earlier.

Looking at the product page, I see that the integrated antenna is limited to 30 dBm of transmit power and the external antenna port is limited to 31.5 dBm. Am I correct in assuming that you want to use the integrated antenna?

It's not immediately obvious why you are not picking up any tags. The only thing that I can suggest is try setting the power to 101 and seeing if anything different (I don't expect it to).

Could you also post a screenshot of the web app that you mentioned? We can try to replicate their configuration.

ransford, 2015-07-20:
I agree with @ewfuentes; try different power levels and please do post the screenshot he requested.  It's likely that sllurp is selecting a set of configuration values *different* from those the web app is selecting.

futnuh, 2015-07-20:
Yes, at the moment I am trying to use the integrated antenna (I don't actually have an external antenna attached at all).  Here are a selection of screenshots from the app.  The first one probably doesn't give much info other than show that tags really are being read.  The other 4 are of various "settings" panes.  To my untrained eye, the settings look to match the sllurp.inventory defaults pretty well.

[image]

[image]

[image]

[image]

[image]

ransford, 2015-07-20:
Hmm, yes.  Notice that `inventory --help` describes the various settings you can change.  In your situation, I'd probably fiddle with the settings in this order:
* `-X` (transmit power)
* `-M` (modulation, although it looks like both the reader and sllurp claim you're using M4; perhaps try FM0?)
* `-T` (Tari, try 25000)
* `-s` (try 0 and 2)

futnuh, 2015-07-21:
I tried all of the above in various combinations but without success.  What's strange is that when I originally tried sllurp, I had a few bin/inventory runs where I saw some tags.  Recently though, nothing.  And yet the ThingMagic Java app still works, reliably finding all the tags I have in range.

ewfuentes, 2015-07-21:
Unless @ransford has any other ideas, I think the best thing to do at this point is to email ThingMagic Support.  I would have dug deeper if I had one of the readers on hand, but these things are tricky to debug otherwise.  If you're familiar with Wireshark, it might help to send them a capture of the traffic between you and the reader.

If you can figure out what's missing or wrong, I can try to help you out then. Best of luck!

ransford, 2015-07-21:
@ewfuentes thanks for helping out!  I don't have a ThingMagic reader on hand, so I can't help much more either.  Here's what I would ask them: how can you tell exactly what settings the reader is using to query tags?

A tcpdump or wireshark trace would also be hugely helpful here even if you don't contact ThingMagic support; we can probably reverse-engineer the protocol conversation and then match the settings the ThingMagic app is selecting.  The process on a Mac would be something like:

```shell
# capture traffic on the LLRP port to the file llrp.pcap
$ sudo tcpdump -i en0 -w llrp.pcap 'port 5084'
```
Then run the ThingMagic app, then `Ctrl-c` to stop tcpdump & share the pcap file in this issue.

I'll reach out to ThingMagic about improving sllurp's support for their readers.

futnuh, 2015-07-25:
Hi Guys, I've done a wireshark packet capture for the ThingMagic Java web app mentioned above.  I kept it as simple as possible: connect; (continuous) read for a few seconds; stop reading; disconnect.  Here's a link to the file:

https://www.dropbox.com/s/mkzxg858994l7na/thingmagic-astraex.pcap?dl=0

For comparison, here's a packet capture corresponding to running "./bin/inventory 10.0.80.63 -X 100":

https://www.dropbox.com/s/xl6cdo34hrqnwhz/sslurp.pcap?dl=0

While I'm going to start looking at the ThingMagic capture tomorrow, this will mostly be as a way to familiarize myself with LLRP - I'm not really expecting to spot the problem.  If someone with experience wants to jump in, please feel free ;-)  As always, any help is much appreciated.

ransford, 2015-07-26:
The comparable packets to focus on are number 72 in the thingmagic-astraex trace and number 17 in the sllurp trace.  Wireshark's LLRP decoding capabilities are excellent.  I haven't found the discrepancy yet, but it's somewhere in there.

futnuh, 2015-07-26:
Unless I'm mistaken, it looks like thingmagic-astrex packet 72 doesn't do much in the way of antenna setup whereas sllurp number 17 does.  I wonder if it is worth modifying my local copy of sllurp temporarily such that it skips the antenna configuration in favour of the defaults?   (If so, could you advise on how to do this?  It's not obvious to me how to modify llrp.py to accomplish this.)  

For the sllurp trace, packet 72, presumably wire shark's listing of "Transmit Power Value: 100" refers to the transmission table index, right?  It isn't an actual power value.

[image]

The only differences that I can spot so far are the antenna configuration (or lack of for the thingmagic app), the start triggering (sllurp jumps straight in with a "start inventory" while the thingmagic does this as a separate step via the "connect" button), and some of the flags in the Tag Report Content Selector (which I don't think impact us right now as we aren't even getting any Tag Reports).

As always, any further help is much appreciated.

ransford, 2015-08-04:
Just catching up here... I agree with your reading of the traces, @futnuh.  The antenna configuration is probably the issue; perhaps we're accidentally messing up the default values.  You can comment out the `AntennaConfiguration` part of `sllurp/llrp_proto.py` around line 2857:

https://github.com/ransford/sllurp/blob/master/sllurp/llrp_proto.py#L2857

ransford, 2017-09-13:
Closing for staleness; I'd be curious if this problem still happens with recent sllurp releases.

closed 2017-09-13 by ransford