sllurp/sllurp #56: ERROR: Problem with READER_EVENT_NOTIFICATION message format

spyle, 2016-12-07:
$ bin/inventory 169.254.1.1
2016-12-07 14:40:04,190 sllurp: INFO: log level: INFO
2016-12-07 14:40:04,191 sllurp.llrp: INFO: connecting...
2016-12-07 14:40:04,192 sllurp.llrp: INFO: will reset reader state on connect
2016-12-07 14:40:04,192 sllurp.llrp: INFO: will start inventory on connect
2016-12-07 14:40:04,192 sllurp.llrp: INFO: using antennas: [1]
2016-12-07 14:40:04,193 sllurp.llrp: INFO: connected to ('169.254.1.1', 5084) (169.254.1.1:5084)
2016-12-07 14:40:04,461 sllurp.llrp: INFO: requested modulation: M4
2016-12-07 14:40:04,461 sllurp.llrp: INFO: using reader mode: {'C': 0, 'MaxTari': 20000, 'M': 3, 'PIE': 2000, 'StepTari': 0, 'R': 1, 'FLM': 0, 'MinTari': 20000, 'ModeIdentifier': 2, 'BDR': 274000, 'Mod': 2}
2016-12-07 14:40:04,461 sllurp.llrp: INFO: stopping politely
2016-12-07 14:40:04,472 sllurp.llrp: INFO: reader finished inventory
2016-12-07 14:40:04,472 sllurp.llrp_proto: INFO: will report every ~N=1 tags
2016-12-07 14:40:04,472 sllurp.llrp: INFO: starting inventory
2016-12-07 14:40:04,505 sllurp.llrp: ERROR: Problem with READER_EVENT_NOTIFICATION message format
Traceback (most recent call last):
  File "sllurp/llrp.py", line 84, in deserialize
    name: dict(decoder(body))
  File "sllurp/llrp_proto.py", line 645, in decode_ReaderEventNotification
    raise LLRPError('junk at end of message: ' + bin2dump(body))
LLRPError: junk at end of message: 00 f9 00 0d 00 00 00 00 01 00 00 00 00            .............
2016-12-07 14:40:04,505 sllurp.llrp: ERROR: unexpected response None when enabling ROSpec
Unhandled Error
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/twisted/python/log.py", line 103, in callWithLogger
    return callWithContext({"system": lp}, func, *args, **kw)
  File "/usr/local/lib/python2.7/site-packages/twisted/python/log.py", line 86, in callWithContext
    return context.call({ILogContext: newCtx}, func, *args, **kw)
  File "/usr/local/lib/python2.7/site-packages/twisted/python/context.py", line 118, in callWithContext
    return self.currentContext().callWithContext(ctx, func, *args, **kw)
  File "/usr/local/lib/python2.7/site-packages/twisted/python/context.py", line 81, in callWithContext
    return func(*args,**kw)
--- <exception caught here> ---
  File "/usr/local/lib/python2.7/site-packages/twisted/internet/selectreactor.py", line 149, in _doReadOrWrite
    why = getattr(selectable, method)()
  File "/usr/local/lib/python2.7/site-packages/twisted/internet/tcp.py", line 208, in doRead
    return self._dataReceived(data)
  File "/usr/local/lib/python2.7/site-packages/twisted/internet/tcp.py", line 214, in _dataReceived
    rval = self.protocol.dataReceived(data)
  File "/usr/local/lib/python2.7/site-packages/twisted/protocols/basic.py", line 578, in dataReceived
    why = self.rawDataReceived(data)
  File "sllurp/llrp.py", line 544, in rawDataReceived
    self.handleMessage(lmsg)
  File "sllurp/llrp.py", line 444, in handleMessage
    status = lmsg.msgdict[msgName]['LLRPStatus']['StatusCode']
exceptions.TypeError: 'NoneType' object has no attribute '__getitem__'

2016-12-07 14:40:04,506 sllurp.llrp: INFO: lost connection: 'NoneType' object has no attribute '__getitem__'
2016-12-07 14:40:04,507 sllurp: INFO: total # of tags seen: 0 (0 tags/second)

ransford, 2016-12-07:
Thank you!  Can you reproduce with debug-level logging?  You can do `bin/inventory --debug --logfile issue56.txt 169.254.1.1` and attach `issue56.txt` here.

spyle, 2016-12-07:
Great. As soon as I ran bin/inventory --debug --logfile issue56.txt 169.254.1.1, the error went away.
It appears to be running, but not reading anything yet. Let me play around with the args/settings and see if I can get it to work.
Thanks for the quick reply

spyle, 2016-12-08:
[issue56.txt](https://github.com/ransford/sllurp/files/639729/issue56.txt)

Ok, so today, I ran MultiReader from Impinj on this Speedway 420 and it works perfectly fine. I ran the inventory command and the same error happened. Here's the log file.

spyle, 2016-12-08:
Lol, sorry I acted in haste. When I rant he command to save the log file, its now throwing a different error. I cant seem to recreate the "junk at end of message error" now.
Now I'm stuck at:
2016-12-08 09:58:32,326 sllurp.llrp: ERROR: unexpected response READER_EVENT_NOTIFICATION when enabling ROSpec
Unhandled Error
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/twisted/python/log.py", line 103, in callWithLogger
    return callWithContext({"system": lp}, func, *args, **kw)
  File "/usr/local/lib/python2.7/site-packages/twisted/python/log.py", line 86, in callWithContext
    return context.call({ILogContext: newCtx}, func, *args, **kw)
  File "/usr/local/lib/python2.7/site-packages/twisted/python/context.py", line 118, in callWithContext
    return self.currentContext().callWithContext(ctx, func, *args, **kw)
  File "/usr/local/lib/python2.7/site-packages/twisted/python/context.py", line 81, in callWithContext
    return func(*args,**kw)
--- <exception caught here> ---
  File "/usr/local/lib/python2.7/site-packages/twisted/internet/selectreactor.py", line 149, in _doReadOrWrite
    why = getattr(selectable, method)()
  File "/usr/local/lib/python2.7/site-packages/twisted/internet/tcp.py", line 208, in doRead
    return self._dataReceived(data)
  File "/usr/local/lib/python2.7/site-packages/twisted/internet/tcp.py", line 214, in _dataReceived
    rval = self.protocol.dataReceived(data)
  File "/usr/local/lib/python2.7/site-packages/twisted/protocols/basic.py", line 578, in dataReceived
    why = self.rawDataReceived(data)
  File "sllurp/llrp.py", line 544, in rawDataReceived
    self.handleMessage(lmsg)
  File "sllurp/llrp.py", line 444, in handleMessage
    status = lmsg.msgdict[msgName]['LLRPStatus']['StatusCode']
exceptions.KeyError: 'LLRPStatus'

2016-12-08 09:58:32,327 sllurp.llrp: INFO: lost connection: 'LLRPStatus'

ransford, 2016-12-24:
still stuck?

kikekr, 2017-05-10:
Same problem. Solved?

kikekr, 2017-05-10:
Mi log file:

2017-05-10 13:00:39,500 sllurp: DEBUG: log level: DEBUG
2017-05-10 13:00:39,500 sllurp.llrp: INFO: connecting...
2017-05-10 13:00:39,501 sllurp.llrp: INFO: will reset reader state on connect
2017-05-10 13:00:39,501 sllurp.llrp: INFO: will start inventory on connect
2017-05-10 13:00:39,501 sllurp.llrp: INFO: using antennas: [1]
2017-05-10 13:00:39,501 sllurp.llrp: INFO: connected to ('10.0.1.57', 5084) (10.0.1.57:5084)
2017-05-10 13:00:39,502 sllurp.llrp: DEBUG: got 32 bytes from reader: 043f000000205ff5d6a300f600160080000c00054f296b53fa4b010000060000
2017-05-10 13:00:39,502 sllurp.llrp: DEBUG: expect 32 bytes (have 32)
2017-05-10 13:00:39,502 sllurp.llrp: DEBUG: deserializing READER_EVENT_NOTIFICATION command
2017-05-10 13:00:39,511 sllurp.llrp_proto: DEBUG: decode_ReaderEventNotification
2017-05-10 13:00:39,512 sllurp.llrp_proto: DEBUG: decode_ReaderEventNotificationData (type=246 len=22)
2017-05-10 13:00:39,514 sllurp.llrp_proto: DEBUG: decode_UTCTimestamp
2017-05-10 13:00:39,515 sllurp.llrp_proto: DEBUG: decode_UTCTimestamp (type=128 len=12)
2017-05-10 13:00:39,516 sllurp.llrp_proto: DEBUG: decode_ConnectionAttemptEvent
2017-05-10 13:00:39,517 sllurp.llrp_proto: DEBUG: decode_ConnectionAttemptEvent (type=256 len=6)
2017-05-10 13:00:39,518 sllurp.llrp_proto: DEBUG: decode_AntennaEvent
2017-05-10 13:00:39,518 sllurp.llrp: DEBUG: done deserializing READER_EVENT_NOTIFICATION command
2017-05-10 13:00:39,518 sllurp.llrp: DEBUG: LLRPMessage received in state 1: <READER_EVENT_NOTIFICATION>
	<Ver>1</Ver>
	<Type>63</Type>
	<ID>1609946787</ID>
	<ReaderEventNotificationData>
		<ConnectionAttemptEvent>
			<Status>Success</Status>
		</ConnectionAttemptEvent>
	</ReaderEventNotificationData>
</READER_EVENT_NOTIFICATION>
2017-05-10 13:00:39,518 sllurp.llrp: DEBUG: starting message callbacks for READER_EVENT_NOTIFICATION
2017-05-10 13:00:39,518 sllurp.llrp: DEBUG: done with message callbacks for READER_EVENT_NOTIFICATION
2017-05-10 13:00:39,518 sllurp.llrp: DEBUG: in handleMessage(READER_EVENT_NOTIFICATION), there are 0 Deferreds
2017-05-10 13:00:39,518 sllurp.llrp: DEBUG: serializing GET_READER_CAPABILITIES command
2017-05-10 13:00:39,518 sllurp.llrp: DEBUG: serialized bytes: 04010000000b0000000000
2017-05-10 13:00:39,519 sllurp.llrp: DEBUG: done serializing GET_READER_CAPABILITIES command
2017-05-10 13:00:39,519 sllurp.llrp: DEBUG: state change: STATE_DISCONNECTED -> STATE_SENT_GET_CAPABILITIES
2017-05-10 13:00:39,521 sllurp.llrp: DEBUG: got 1448 bytes from reader: 040b0000070f00000000011f00080000000000890347003440000000651a001e886f0009352e342e302e323430008b000800010000008b00080002000a008b00080003000b008b00080004000c008b00080005000d008b00080006000e008b00080007000f008b000800080010008b000800090011008b0008000a0012008b0008000b0013008b0008000c0014008b0008000d0015008b0008000e0016008b0008000f0017008b000800100018008b000800110019008b00080012001a008b00080013001b008b00080014001c008b00080015001d008b00080016001e008b00080017001f008b000800180020008b000800190021008b0008001a0022008b0008001b0023008b0008001c0024008b0008001d0025008b0008001e0026008b0008001f0027008b000800200028008b000800210029008b00080022002a008b00080023002b008b00080024002c008b00080025002d008b00080026002e008b00080027002f008b000800280030008b000800290031008b0008002a0032008d000800000000008c00090001000101008c00090002000101008c00090003000101008c00090004000101008c00090005000101008c00090006000101008c00090007000101008c00090008000101008c00090009000101008c0009000a000101008c0009000b000101008c0009000c000101008c0009000d000101008c0009000e000101008c0009000f000101008c00090010000101008c00090011000101008c00090012000101008c00090013000101008c00090014000101008c00090015000101008c00090016000101008c00090017000101008c00090018000101008c00090019000101008c0009001a000101008c0009001b000101008c0009001c000101008c0009001d000101008c0009001e000101008c0009001f000101008c00090020000101008c00090021000101008c00090022000101008c00090023000101008c00090024000101008c00090025000101008c00090026000101008c00090027000101008c00090028000101008c00090029000101008c0009002a000101008c0009002b000101008c0009002c000101008c0009002d000101008c0009002e000101008c0009002f000101008c00090030000101008c00090031000101008c00090032000101008c00090033000101008c00090034000101008e001c48010000000000010000002000000001000005e400000008008f0393000000020090038b00910008000103e80091000800020401009100080003041a0091000800040433009100080005044c0091000800060465009100080007047e009100080008049700910008000904b000910008000a04c900910008000b04e200910008000c04fb00910008000d051400910008000e052d00910008000f0546009100080010055f0091000800110578009100080012059100910008001305aa00910008001405c300910008001505dc00910008001605f5009100080017060e0091000800180627009100080019064000910008001a065900910008001b067200910008001c068b00910008001d06a400910008001e06bd00910008001f06d600910008002006ef00910008002107080091000800220721009100080023073a0091000800240753009100080025076c0091000800260785009100080027079e00910008002807b700910008002907d000910008002a07e900910008002b080200910008002c081b00910008002d083400910008002e084d00910008002f0866009100080030087f009100080031089800910008003208b100910008003308ca00910008003408e300910008003508fc0091000800360915009100080037092e0091000800380947009100080039096000910008003a097900910008003b099200910008003c09ab00910008003d09c400910008003e09dd00910008003f09f60091000800400a0f0091000800410a280091000800420a410091000800430a5a0091000800440a730091000800450a
2017-05-10 13:00:39,521 sllurp.llrp: DEBUG: expect 1807 bytes (have 1448)
[... cut ...]
		<MaxNumInventoryParametersSpecsPerAISpec>1</MaxNumInventoryParametersSpecsPerAISpec>
		<MaxNumAccessSpec>1508</MaxNumAccessSpec>
		<MaxNumOpSpecsPerAccessSpec>8</MaxNumOpSpecsPerAccessSpec>
	</LLRPCapabilities>
	<RegulatoryCapabilities>
		<CountryCode>0</CountryCode>
		<CommunicationsStandard>2</CommunicationsStandard>
		<UHFBandCapabilities>
			<FrequencyInformation>
				<Hopping>False</Hopping>
			</FrequencyInformation>
			<UHFRFModeTable>
			</UHFRFModeTable>
		</UHFBandCapabilities>
	</RegulatoryCapabilities>
	<AirProtocolLLRPCapabilities>G

ransford, 2017-08-09:
Still having this problem @spyle @kikekr?

closed 2017-08-12 by ransford