sllurp/sllurp #57: ThingMagic Sargas: Unsupported StopTrigger 3

will2142, 2017-02-27:
Hi,

I was trying to use the `inventory` script with the ThingMagic Sargas and get the following error.

> 2017-02-27 11:57:01,481 sllurp: INFO: log level: INFO
> 2017-02-27 11:57:01,482 sllurp.llrp: INFO: connecting...
> 2017-02-27 11:57:01,485 sllurp.llrp: INFO: will reset reader state on connect
> 2017-02-27 11:57:01,485 sllurp.llrp: INFO: will start inventory on connect
> 2017-02-27 11:57:01,485 sllurp.llrp: INFO: using antennas: [1]
> 2017-02-27 11:57:01,485 sllurp.llrp: INFO: connected to ('sargas-836c20.local', 5084) (10.129.1.197:5084)
> 2017-02-27 11:57:01,649 sllurp.llrp: INFO: requested modulation: M4
> 2017-02-27 11:57:01,649 sllurp.llrp: INFO: using reader mode: {'C': 0, 'MaxTari': 25000, 'M': 3, 'PIE': 1700, 'StepTari': 12500, 'R': 1, 'FLM': 0, 'MinTari': 12500, 'ModeIdentifier': 1, 'BDR': 250000, 'Mod': 2}
> 2017-02-27 11:57:01,649 sllurp.llrp: INFO: stopping politely
> 2017-02-27 11:57:01,656 sllurp.llrp: INFO: reader finished inventory
> 2017-02-27 11:57:01,657 sllurp.llrp_proto: INFO: will report every ~N=0 tags
> 2017-02-27 11:57:01,657 sllurp.llrp: INFO: starting inventory
> 2017-02-27 11:57:01,677 sllurp.llrp: CRITICAL: Error ParameterError adding ROSpec: ADD_ROSPEC: Error: Adding AISpec with unsupported StopTrigger 3

It's not obvious to me how to change the StopTrigger.  I'd appreciate any help.

Thanks!

ransford, 2017-02-27:
It's possible that this reader doesn't support the `'Tag observation'` stop trigger.  Can you run `inventory` with `-d -l sargas.log`, find the reader capabilities in `sargas.log`, and paste them here?  Look for something like:
```
2016-09-21 09:26:52,473 sllurp.llrp: DEBUG: LLRPMessage received in state 9: <GET_READER_CAPABILITIES_RESPONSE>
[... lots of XML follows ...]
</GET_READER_CAPABILITIES_RESPONSE>
```

will2142, 2017-02-28:
Thanks. How do you disable the `'Tag observation'` stop trigger? 

Here's the requested log.

```
2017-02-28 10:53:27,733 sllurp.llrp: DEBUG: LLRPMessage received in state 9: <GET_READER_CAPABILITIES_RESPONSE>
	<Ver>1</Ver>
	<Type>11</Type>
	<ID>0</ID>
	<LLRPStatus>
		<StatusCode>Success</StatusCode>
		<ErrorDescription></ErrorDescription>
	</LLRPStatus>
	<GeneralDeviceCapabilities>
		<MaxNumberOfAntennaSupported>2</MaxNumberOfAntennaSupported>
		<CanSetAntennaProperties>False</CanSetAntennaProperties>
		<HasUTCClockCapability>True</HasUTCClockCapability>
		<DeviceManufacturerName>26554</DeviceManufacturerName>
		<ModelName>13360</ModelName>
		<FirmwareVersionByteCount>8</FirmwareVersionByteCount>
		<ReaderFirmwareVersion>5.1.3.51</ReaderFirmwareVersion>
		<ReceiveSensitivityTableEntry>
			<Index>1</Index>
			<ReceiveSensitivityValue>0</ReceiveSensitivityValue>
		</ReceiveSensitivityTableEntry>
		<GPIOCapabilities>
			<NumGPIs>2</NumGPIs>
		</GPIOCapabilities>
		<PerAntennaAirProtocol>
			<AntennaID>1</AntennaID>
			<NumProtocols>1</NumProtocols>
		</PerAntennaAirProtocol>
	</GeneralDeviceCapabilities>
	<LLRPCapabilities>
		<CanDoRFSurvey>False</CanDoRFSurvey>
		<CanReportBufferFillWarning>False</CanReportBufferFillWarning>
		<SupportsClientRequestOpSpec>False</SupportsClientRequestOpSpec>
		<CanDoTagInventoryStateAwareSingulation>False</CanDoTagInventoryStateAwareSingulation>
		<SupportsEventAndReportHolding>True</SupportsEventAndReportHolding>
		<MaxPriorityLevelSupported>1</MaxPriorityLevelSupported>
		<ClientRequestOpSpecTimeout>0</ClientRequestOpSpecTimeout>
		<MaxNumROSpec>32</MaxNumROSpec>
		<MaxNumSpecsPerROSpec>16</MaxNumSpecsPerROSpec>
		<MaxNumInventoryParametersSpecsPerAISpec>4</MaxNumInventoryParametersSpecsPerAISpec>
		<MaxNumAccessSpec>128</MaxNumAccessSpec>
		<MaxNumOpSpecsPerAccessSpec>1</MaxNumOpSpecsPerAccessSpec>
	</LLRPCapabilities>
	<RegulatoryCapabilities>
		<CountryCode>0</CountryCode>
		<CommunicationsStandard>1</CommunicationsStandard>
		<UHFBandCapabilities>
			<FrequencyInformation>
				<Hopping>True</Hopping>
			</FrequencyInformation>
			<UHFRFModeTable>
			</UHFRFModeTable>
		</UHFBandCapabilities>
	</RegulatoryCapabilities>
	<AirProtocolLLRPCapabilities>G\00\C0\00\FF\00
\00\00g\BA\00\00\00K\00\FF\00\86\00\00g\BA\00\00\00\00000000000000\0045160123701051\001.0\00UM6e Micro HWVer:20.00.00.01 BootVer:12.12.13.00 AppVer:01.07.03.25 AppDate:2015.12.21\FF\00\00\00g\BA\00\00\00\FF\00
\00\00g\BA\00\00\00</AirProtocolLLRPCapabilities>
</GET_READER_CAPABILITIES_RESPONSE>
```

ransford, 2017-02-28:
Thanks for the capabilities blob -- very helpful!  If you search `sllurp/llrp_proto.py` for `Tag observation`, you'll quickly find a spot where `AISpecStopTriggerType` is hard-coded to `Tag observation` when you've enabled every-N-tags reporting; there you can substitute any other value from `StopTrigger_Name2Type`:

```python
# 10.2.1.1.2 ROSpec Stop trigger
StopTrigger_Name2Type = {
    'Null':                 0,
    'Duration':             1,
    'GPI with timeout':         2,
    'Tag observation':          3
}
```

will2142, 2017-02-28:
Thanks.  I see that if I make the default value `None` for `--report-every-n-tags` that I don't get the error about the unsupported StopTrigger.  Unfortunately I also don't get any tag reads... :disappointed: 

I suspect it has to do with nothing triggering the `tagReportCallback()` function.  Do you have any examples that do not use the `'Tag observation'` stop trigger?

ransford, 2017-02-28:
I recommend you try `'Duration'` instead.  I won't get more time to work on this issue this week, unfortunately.

will2142, 2017-03-09:
I've tried `Duration` by setting `--report-every-n-tags=None` and a duration of `--time=1`.

It seems to work very sporadically.  That is the tags are only read once every 20 to 30 tries.  Other times, it returns saying it's detected no tags.  The tags are directly in front of the antenna and have not been moved in this test.

Insights?

ransford, 2017-03-11:
Can you post the entire command line that yields that result, along with a `--debug`-level log file?  That will allow me to inspect the messages going back and forth.  The sad reality is that there are differences in protocol support between different models of reader.

will2142, 2017-03-17:
I ran the command 

```bash
./inventory sargas-836c20.local --time 1 -d -l duration3.log
```

With the following patch
```diff
diff --git a/sllurp/inventory.py b/sllurp/inventory.py
index 34ba348..2af279d 100644
--- a/sllurp/inventory.py
+++ b/sllurp/inventory.py
@@ -56,7 +56,8 @@ def tagReportCallback(llrpMsg):
         logger.info('no tags seen')
         return
     for tag in tags:
-        numTags += tag['TagSeenCount'][0]
+    #    numTags += tag['TagSeenCount'][0]
+        print(tag)
 
 
 def parse_args():
@@ -70,7 +71,7 @@ def parse_args():
                         help='seconds to inventory (default forever)')
     parser.add_argument('-d', '--debug', action='store_true',
                         help='show debugging output')
-    parser.add_argument('-n', '--report-every-n-tags', default=1, type=int,
+    parser.add_argument('-n', '--report-every-n-tags', default=None, type=int,
                         dest='every_n', metavar='N',
                         help='issue a TagReport every N tags')
     parser.add_argument('-a', '--antennas', default='1',
```

Log files are https://gist.github.com/will2142/359b3b80c2291004473f3d6a4c64d24c

Thanks for you help!

ransford, 2017-09-13:
I'm sorry I didn't get to this sooner.  I wonder -- is this problem still occurring with recent releases, e.g., the latest, 0.3.5?

closed 2021-01-24 by ransford