sllurp/sllurp #61: Antenna configuration problems

kikekr, 2017-05-31:
Hi! I have some problems to configure the antennas in my LLRP reader, in order to run a simple inventory. I would like to receive only data from antennas 5, 6 and 7. Then, I'm running the inventory like this:

```
python inventory.py -t 2 -a 5,6,7 10.0.1.57
``` 

When I do this, only receive data from the first antenna that I set up, in this case, the 5.

```
2017-05-31 11:20:53,525 sllurp: INFO: no tags seen
2017-05-31 11:20:53,530 sllurp.llrp: WARNING: message None received in unknown state!
2017-05-31 11:20:53,534 sllurp.llrp: WARNING: message None received in unknown state!
2017-05-31 11:20:53,569 sllurp: INFO: saw tag(s): [{'AccessSpecID': (0,),
  'AntennaID': (5,),
  'ChannelIndex': (1,),
  'EPCData': {'EPC': '08000000000000001dc184bf8b218400',
              'EPCLengthBits': 128},
  'FirstSeenTimestampUTC': (1496222831128366,),
  'InventoryParameterSpecID': (1,),
  'LastSeenTimestampUTC': (1496222831128366,),
  'PeakRSSI': (-59,),
  'ROSpecID': (1,),
  'SpecIndex': (1,),
  'TagSeenCount': (1,)}]
2017-05-31 11:20:53,571 sllurp: INFO: no tags seen
2017-05-31 11:20:53,576 sllurp.llrp: WARNING: message None received in unknown state!
2017-05-31 11:20:53,581 sllurp.llrp: WARNING: message None received in unknown state!
2017-05-31 11:20:53,697 sllurp: INFO: saw tag(s): [{'AccessSpecID': (0,),
  'AntennaID': (5,),
  'ChannelIndex': (1,),
  'EPCData': {'EPC': '080000000000000020a13c3233c28400',
              'EPCLengthBits': 128},
  'FirstSeenTimestampUTC': (1496222831192767,),
  'InventoryParameterSpecID': (1,),
  'LastSeenTimestampUTC': (1496222831192767,),
  'PeakRSSI': (-58,),
  'ROSpecID': (1,),
  'SpecIndex': (1,),
  'TagSeenCount': (1,)}]
``` 

```
python inventory.py -t 2 -a 6,7,8 10.0.1.57
```

In this case, only antenna 6:

```
2017-05-31 11:22:46,310 sllurp: INFO: no tags seen
2017-05-31 11:22:46,315 sllurp.llrp: WARNING: message None received in unknown state!
2017-05-31 11:22:46,320 sllurp.llrp: WARNING: message None received in unknown state!
2017-05-31 11:22:46,433 sllurp: INFO: saw tag(s): [{'AccessSpecID': (0,),
  'AntennaID': (6,),
  'ChannelIndex': (1,),
  'EPCData': {'EPC': '0800000000000000314489b334008400',
              'EPCLengthBits': 128},
  'FirstSeenTimestampUTC': (1496222943928948,),
  'InventoryParameterSpecID': (1,),
  'LastSeenTimestampUTC': (1496222943928948,),
  'PeakRSSI': (-60,),
  'ROSpecID': (1,),
  'SpecIndex': (1,),
  'TagSeenCount': (1,)}]
2017-05-31 11:22:46,435 sllurp: INFO: no tags seen
2017-05-31 11:22:46,441 sllurp.llrp: WARNING: message None received in unknown state!
2017-05-31 11:22:46,449 sllurp.llrp: WARNING: message None received in unknown state!
2017-05-31 11:22:46,472 sllurp: INFO: saw tag(s): [{'AccessSpecID': (0,),
  'AntennaID': (6,),
  'ChannelIndex': (1,),
  'EPCData': {'EPC': '08000000000000002f3b33930e808400',
              'EPCLengthBits': 128},
  'FirstSeenTimestampUTC': (1496222944051346,),
  'InventoryParameterSpecID': (1,),
  'LastSeenTimestampUTC': (1496222944051346,),
  'PeakRSSI': (-71,),
  'ROSpecID': (1,),
  'SpecIndex': (1,),
  'TagSeenCount': (1,)}]
```

ransford, 2017-06-01:
Hmm.  I'd probably try two things here.

First, if you have the MultiReader software from Impinj (which runs only on Windows iirc), leave your physical setup as it is, and see if MultiReader gives you tag reports from all three antennas.  If it does, then you've probably found a sllurp bug.

Second, increase the log level to `DEBUG` (you can use the `--debug --logfile foo.log` switches on the inventory script) and look carefully at the outgoing ROSpec (`vim -R -c '/ROSpec:' foo.log`).  Are there the right number of `<AntennaConfiguration>` blocks (should equal the number of antennas you've chosen)?

I just connected to an Impinj R220 reader with `inventory -lfoo.log -d -a1,2 my.reader.hostname` and here's what my `ROSpec` looked like:

```
2017-05-31 20:30:38,685 sllurp.llrp: DEBUG: ROSpec: <ROSpec>
	<ROSpecID>1</ROSpecID>
	<Priority>0</Priority>
	<CurrentState>Disabled</CurrentState>
	<ROBoundarySpec>
		<ROSpecStartTrigger>
			<ROSpecStartTriggerType>Immediate</ROSpecStartTriggerType>
		</ROSpecStartTrigger>
		<ROSpecStopTrigger>
			<ROSpecStopTriggerType>Null</ROSpecStopTriggerType>
			<DurationTriggerValue>0</DurationTriggerValue>
		</ROSpecStopTrigger>
	</ROBoundarySpec>
	<AISpec>
		<AntennaIDs>1 2</AntennaIDs>
		<AISpecStopTrigger>
			<AISpecStopTriggerType>Null</AISpecStopTriggerType>
			<DurationTriggerValue>0</DurationTriggerValue>
		</AISpecStopTrigger>
		<InventoryParameterSpec>
			<InventoryParameterSpecID>1</InventoryParameterSpecID>
			<ProtocolID>1</ProtocolID>
			<AntennaConfiguration>
				<AntennaID>1</AntennaID>
				<RFTransmitter>
					<HopTableId>1</HopTableId>
					<ChannelIndex>1</ChannelIndex>
					<TransmitPower>91</TransmitPower>
				</RFTransmitter>
				<C1G2InventoryCommand>
					<TagInventoryStateAware>False</TagInventoryStateAware>
					<C1G2RFControl>
						<ModeIndex>1000</ModeIndex>
						<Tari>20000</Tari>
					</C1G2RFControl>
					<C1G2SingulationControl>
						<Session>2</Session>
						<TagPopulation>4</TagPopulation>
						<TagTransitTime>0</TagTransitTime>
					</C1G2SingulationControl>
				</C1G2InventoryCommand>
			</AntennaConfiguration>
			<AntennaConfiguration>
				<AntennaID>2</AntennaID>
				<RFTransmitter>
					<HopTableId>1</HopTableId>
					<ChannelIndex>1</ChannelIndex>
					<TransmitPower>91</TransmitPower>
				</RFTransmitter>
				<C1G2InventoryCommand>
					<TagInventoryStateAware>False</TagInventoryStateAware>
					<C1G2RFControl>
						<ModeIndex>1000</ModeIndex>
						<Tari>20000</Tari>
					</C1G2RFControl>
					<C1G2SingulationControl>
						<Session>2</Session>
						<TagPopulation>4</TagPopulation>
						<TagTransitTime>0</TagTransitTime>
					</C1G2SingulationControl>
				</C1G2InventoryCommand>
			</AntennaConfiguration>
		</InventoryParameterSpec>
	</AISpec>
	<ROReportSpec>
		<N>0</N>
		<ROReportTrigger>Upon_N_Tags_Or_End_Of_AISpec</ROReportTrigger>
		<TagReportContentSelector>
			<EnableROSpecID>False</EnableROSpecID>
			<EnableSpecIndex>False</EnableSpecIndex>
			<EnableInventoryParameterSpecID>False</EnableInventoryParameterSpecID>
			<EnableAntennaID>True</EnableAntennaID>
			<EnableChannelIndex>False</EnableChannelIndex>
			<EnablePeakRRSI>True</EnablePeakRRSI>
			<EnableFirstSeenTimestamp>False</EnableFirstSeenTimestamp>
			<EnableLastSeenTimestamp>True</EnableLastSeenTimestamp>
			<EnableTagSeenCount>True</EnableTagSeenCount>
			<EnableAccessSpecID>False</EnableAccessSpecID>
		</TagReportContentSelector>
	</ROReportSpec>
</ROSpec>
```

kikekr, 2017-06-01:
I think I've found the problem. The strange behavior described above occurs when the AISpecStopTriggerType field has the value 'Tag observation' but I dont know the root cause of this. If I set 'Null' (like your example) it works fine. The problem is that the sllurp code automatically overwrites the AISpecStopTriggerType field with the value 'Tag observation' in the LLRPROSpec class constructor. I just commented this part of the code and the problem has been solved:

```
        if report_every_n_tags is not None:
            if report_timeout_ms:
                logger.info('will report every ~N=%d tags or %d ms',
                            report_every_n_tags, report_timeout_ms)
            else:
                logger.info('will report every ~N=%d tags',
                            report_every_n_tags)
            self['ROSpec']['ROReportSpec'].update({
                'ROReportTrigger': 'Upon_N_Tags_Or_End_Of_AISpec',
                'N': report_every_n_tags,
            })
            # self['ROSpec']['AISpec']['AISpecStopTrigger'].update({
            #         'AISpecStopTriggerType': 'Tag observation',
            #         'TagObservationTrigger': {
            #             'TriggerType': 'UponNTags',
            #             'NumberOfTags': report_every_n_tags,
            #             'NumberOfAttempts': 0,
            #             'T': 0,
            #             'Timeout': report_timeout_ms,  # milliseconds
            #         },
            # })

            print(self['ROSpec'])
```

ransford, 2017-06-01:
I have to confess that I've never fully understood the _entire_ range of configurations possible with ROSpec and AISpec; there are simply too many possibilities (and I have regular access to only two models of reader).

Check out the latest sllurp from `develop` and see if

    $ sllurp inventory <your.reader.ip>

has the desired effect?  Also try with `--report-every-n-tags N` (any `N` ≥1)?

ransford, 2017-07-28:
Still happening with latest `sllurp` release?

ransford, 2017-09-13:
Closing; please reopen if you'd like to investigate further.

closed 2017-09-13 by ransford