sllurp/sllurp #62: Getting TID from RFID tags through LLRP readers (although not Impinj).

kikekr, 2017-06-01:
Hi! I would like to get the TID from the RFID through any LLRP reader (although not Impinj).

First of all, I don't know if it is possible, because I've seen that this data is requested in a _Custom Parameter_ by the Octane SDK in LLRP frames. **Any idea?** Example:

[image]

Secondly, if I can request this data from any reader through a Custom Parameter in the ROSpec, how could I do it? I started the implementation of this field but I have some problems.

First, I defined the Custom Parameter struct:

```
Message_struct['CustomParameter'] = {
    'type': 1023,
    'fields': [
        'VendorID',
    #    'Subtype',
    #    'CustomParameter'
    ],
    'encode': encode_CustomParameter
}
```
 Second, I've implemented the encoding function like yours:

```
def encode_CustomParameter(par):

    msgtype = Message_struct['CustomParameter']['type']

    msg_header = '!HH'

    flags = 0
    i = 15
    for field in Message_struct['CustomParameter']['fields']:
        if field in par and par[field]:
            print(field)
            flags = flags | (1 << i)
        i = i - 1

    data = struct.pack('!H', flags)
    data = struct.pack(msg_header, msgtype,
                       len(data) + struct.calcsize(msg_header)) + data

    return data
``` 

And then, I added the new field to the 'ROSpec':

```
# 16.2.7.1 ROReportSpec Parameter
def encode_ROReportSpec(par):
    msgtype = Message_struct['ROReportSpec']['type']
    n = int(par['N'])
    roReportTrigger = ROReportTrigger_Name2Type[par['ROReportTrigger']]

    msg_header = '!HHBH'
    msg_header_len = struct.calcsize(msg_header)

    data = encode('TagReportContentSelector')(par['TagReportContentSelector'])

    custom = encode('CustomParameter')(par['CustomParameter'])

    data = struct.pack(msg_header, msgtype,
                       len(data) + len(custom) + msg_header_len,
                       roReportTrigger, n) + data + custom

    return data

Message_struct['ROReportSpec'] = {
    'type': 237,
    'fields': [
        'N',
        'ROReportTrigger',
        'TagReportContentSelector',
        'CustomParameter'
    ],
    'encode': encode_ROReportSpec
}
```

Finally, the reader replies that there is an error with any of the fields:

[image]

Could you help me a little bit? :)
Thank you!

ransford, 2017-06-01:
What kind of reader are you connecting to, and is there a public spec that describes the parameters you want to use?

I'm happy to help.  I know it's a pain to write encoders & decoders; that's the biggest chunk of code we inherited from the LLRPyC project when we forked.  I am gradually building up a declarative parser/builder for LLRP messages in `feature/construct`, but that's coming along slowly (I'm the bottleneck).

ghiewa, 2019-10-29:
kikekr, have you solved it?

closed 2021-01-24 by ransford