sllurp/sllurp #63: Hardcoded ModeIndex breaks communication with WISP5

anparks, 2017-06-14:
The value of mode_index (sllurp/llrp_proto.py) has been hardcoded to a value of '1000', breaking communication with WISP5 which requires a value of '0'.

ransford, 2017-06-14:
Would you rather:
- be able to specify a `mode_index` argument to the `LLRPClientFactory`,
- be able to request a specific `mode_index` via a command-line argument to `inventory`, or
- be able to specify an entire pile of configuration variables, including `ModeIndex`, in a configuration file that you can provide to `LLRPClientFactory` and/or `inventory`?

anparks, 2017-06-16:
Hi Ben,

Seems like the simplest route would be if we can specify `mode_index` via a command line argument to `inventory`. Basically, when we are trying to test the operation of WISPs, we are unable to get them working with sllurp using the once-standard:

`sllurp inventory -n 1 -T 7140 -M WISP5 [ip address]`

We haven't yet done any testing (of the latest version of sllurp) beyond simply using `inventory`.

Thanks!

ransford, 2017-06-17:
Fixed, I claim.  Just got lots of tag reads on a Speedway Revolution R420 by going like this:

```shell
$ sllurp inventory -n 1 -T 7140 -M WISP5 --mode-index 0 [ipaddr]
```

ransford, 2017-06-17:
Caveat (which I'm adding to the README): ModeIndex is _supposed_ to be an index into the `UHFC1G2RFModeTable`, which the reader returns as part of its "capabilities" blob.  So a `ModeIndex` of 0 *should* refer to the first entry in that table.  But the Impinj Speedway readers I have seem to want me to specify the `ModeIdentifier` from those entries instead.

In other words, `--mode-index 0` *should* select `UHFC1G2RFModeTable[0]`, but instead `--mode-index 0` specifies `UHFC1G2RFModeTable[n] where UHFC1G2RFModeTable[n].ModeIdentifier == 0`.  Ugh!

Here's what that table looks like when I connect to a Speedway Revolution R220:

```python
'UHFRFModeTable': {
    'UHFC1G2RFModeTableEntry0': {
        'BDR': 274000,
        'C': 0,
        'FLM': 0,
        'M': 3,
        'MaxTari': 20000,
        'MinTari': 20000,
        'Mod': 2,
        'ModeIdentifier': 2,
        'PIE': 2000,
        'R': 1,
        'StepTari': 0
    },
    'UHFC1G2RFModeTableEntry1': {
        'BDR': 170600,
        'C': 0,
        'FLM': 0,
        'M': 3,
        'MaxTari': 20000,
        'MinTari': 20000,
        'Mod': 3,
        'ModeIdentifier': 3,
        'PIE': 2000,
        'R': 1,
        'StepTari': 0
    },
    'UHFC1G2RFModeTableEntry2': {
        'BDR': 40000,
        'C': 0,
        'FLM': 0,
        'M': 0,
        'MaxTari': 6250,
        'MinTari': 6250,
        'Mod': 0,
        'ModeIdentifier': 1000,
        'PIE': 1500,
        'R': 0,
        'StepTari': 0
    }
}
```

To select the first one, we should use `--mode-index 2`.

I'm looking into a way to make mode selection more straighforward.

ransford, 2017-06-18:
OK, try this:

    sllurp inventory -n 1 -T 7140 -M WISP5 --mode-identifier 0 [ipaddr]

(note ``mode-identifier`` not ``mode-index``)

anparks, 2017-06-27:
This works! Thanks so much!

closed 2017-06-27 by anparks

ransford, 2017-07-06:
Just to be sure, you get this warning about `Requested Tari is greater than MaxTari for selected mode`, right?

```
$ git describe
v0.2.4-2-g0097f37
$ sllurp version
0.2.4
$ sllurp inventory -n 1 -T 7140 -M WISP5 --mode-identifier 0 10.218.95.113
2017-07-05 21:30:32,852 sllurp.verb.inventory: WARNING: recommended Tari for WISP5 is 6250
2017-07-05 21:30:32,852 sllurp.llrp: INFO: connecting to 10.218.95.113:5084...
2017-07-05 21:30:32,858 sllurp.llrp: INFO: will reset reader state on connect
2017-07-05 21:30:32,859 sllurp.llrp: INFO: will start inventory on connect
2017-07-05 21:30:32,859 sllurp.llrp: INFO: using antennas: [1]
2017-07-05 21:30:32,860 sllurp.llrp: INFO: connected to (u'10.218.95.113', 5084) (10.218.95.113:5084)
2017-07-05 21:30:33,147 sllurp.llrp: ERROR: Capabilities mismatch
Traceback (most recent call last):
  File "/Users/ben/vc/sllurp/sllurp/llrp.py", line 418, in handleMessage
    self.parseCapabilities(self.capabilities)
  File "/Users/ben/vc/sllurp/sllurp/llrp.py", line 326, in parseCapabilities
    ' {}'.format(self.reader_mode))
ReaderConfigurationError: Requested Tari is greater than MaxTari for selected mode {'C': 0, 'MaxTari': 6250, 'M': 2, 'PIE': 1500, 'StepTari': 0, 'R': 1, 'FLM': 2, 'MinTari': 6250, 'ModeIdentifier': 0, 'BDR': 640000, 'Mod': 0}
2017-07-05 21:30:33,147 sllurp.llrp: ERROR: Failed to decode LLRPMessage; will not decode 1641 remaining bytes
Traceback (most recent call last):
  File "/Users/ben/vc/sllurp/sllurp/llrp.py", line 573, in rawDataReceived
    self.handleMessage(lmsg)
  File "/Users/ben/vc/sllurp/sllurp/llrp.py", line 421, in handleMessage
    raise err
ReaderConfigurationError: Requested Tari is greater than MaxTari for selected mode {'C': 0, 'MaxTari': 6250, 'M': 2, 'PIE': 1500, 'StepTari': 0, 'R': 1, 'FLM': 2, 'MinTari': 6250, 'ModeIdentifier': 0, 'BDR': 640000, 'Mod': 0}
^C2017-07-05 21:30:35,610 sllurp.llrp: INFO: stopping politely
2017-07-05 21:30:35,610 sllurp.llrp: INFO: will disconnect when stopped
2017-07-05 21:30:35,629 sllurp.llrp: INFO: lost connection: Connection to the other side was lost in a non-clean fashion: Connection lost.
2017-07-05 21:30:35,629 sllurp.verb.inventory: INFO: total # of tags seen: 0 (0 tags/second)
2017-07-05 21:30:35,629 sllurp.llrp: INFO: disconnecting
$
```

zkapetanovic, 2017-07-10:
Hi Ben, 

We are getting the same message, 

```
D-173-250-156-139:sllurp zerinak$ sllurp inventory -n 1 -T 7140 -M WISP5 --mode-identifier 0 192.168.1.2
2017-07-10 11:47:09,098 sllurp.verb.inventory: WARNING: recommended Tari for WISP5 is 6250
2017-07-10 11:47:09,099 sllurp.llrp: INFO: connecting to 192.168.1.2:5084...
2017-07-10 11:47:09,101 sllurp.llrp: INFO: will reset reader state on connect
2017-07-10 11:47:09,101 sllurp.llrp: INFO: will start inventory on connect
2017-07-10 11:47:09,101 sllurp.llrp: INFO: using antennas: [1]
2017-07-10 11:47:09,102 sllurp.llrp: INFO: connected to (u'192.168.1.2', 5084) (192.168.1.2:5084)
2017-07-10 11:47:11,015 sllurp.llrp: INFO: using reader mode: {'C': 0, 'MaxTari': 7140, 'M': 2, 'PIE': 1500, 'StepTari': 0, 'R': 1, 'FLM': 0, 'MinTari': 7140, 'ModeIdentifier': 0, 'BDR': 640000, 'Mod': 0}

```

yogeshv12s, 2025-03-17:
already scanned rfids are not populating again when re run the program`
# Minimal example; see sllurp/verb/inventory.py for more.
from sllurp import llrp
from sllurp.llrp import LLRPReaderConfig, LLRPReaderClient, LLRP_DEFAULT_PORT
#import logging
#logging.getLogger().setLevel(logging.INFO)

def tag_report_cb (reader, tag_reports):
    for tag in tag_reports:
        print('tag: %r' % tag)

config = LLRPReaderConfig()
reader = LLRPReaderClient("192.168.10.166", LLRP_DEFAULT_PORT, config,5)
reader.add_tag_report_callback(tag_report_cb)

reader.connect()
# We are now connected to the reader and inventory is running.

try:
   # logging.info("Waiting for tag reports...")
    reader.join(None)  # Blocks and waits for tag reports
except (KeyboardInterrupt, SystemExit):
   # logging.info("Exiting due to user interrupt.")
    reader.disconnect()
`

commit a9c02597 (2017-06-17): accept --mode-index on inventory command

fixes #63