sllurp/sllurp #64: Modulation problem

whitecloudy, 2017-06-23:
I tried to use R420 reader to communicate with my WISP5 tag. My command line was this

>sllurp inventory -M WISP5 -T 7140 (my reader's ip address)

But unfortunately I got this massage

>WARNING: Could not find reader mode matching modulation=WISP5 and Tari=7140

And sllurp just ignore my modulation setting. I tried other modulation settings like FM0 or M2 etc but none of these are working.

Can you help me out with this problem?

ransford, 2017-06-24:
Is this with v0.2.4 or later?  I made the parameter selection stricter.

With 0097f37e759e51d2cb713350638087641dbaf8ad against an R220 with `sllurp inventory -n1 -M FM0 -T 7140 speedwayr-xx.local.` I get:
```
2017-06-23 23:46:18,428 sllurp.llrp: INFO: connecting to speedwayr-xx.local.:5084...
2017-06-23 23:46:18,784 sllurp.llrp: INFO: will reset reader state on connect
2017-06-23 23:46:18,784 sllurp.llrp: INFO: will start inventory on connect
2017-06-23 23:46:18,784 sllurp.llrp: INFO: using antennas: [1]
2017-06-23 23:46:18,785 sllurp.llrp: INFO: connected to (u'speedwayr-xx.local.', 5084) (10.218.95.191:5084)
2017-06-23 23:46:19,090 sllurp.llrp: ERROR: Capabilities mismatch
Traceback (most recent call last):
  File "/Users/ben/vc/sllurp/sllurp/llrp.py", line 418, in handleMessage
    self.parseCapabilities(self.capabilities)
  File "/Users/ben/vc/sllurp/sllurp/llrp.py", line 326, in parseCapabilities
    ' {}'.format(self.reader_mode))
ReaderConfigurationError: Requested Tari is greater than MaxTari for selected mode {'C': 0, 'MaxTari': 6250, 'M': 0, 'PIE': 1500, 'StepTari': 0, 'R': 0, 'FLM': 0, 'MinTari': 6250, 'ModeIdentifier': 1000, 'BDR': 40000, 'Mod': 0}
```

ransford, 2017-06-24:
You can see from the `ReaderConfigurationError` that `Requested Tari is greater than MaxTari` and `MaxTari == 6250`.

whitecloudy, 2017-06-26:
>2017-06-26 14:28:20,616 root: INFO: log level: INFO
>2017-06-26 14:28:20,617 sllurp.llrp: INFO: connecting to 169.254.0.60:5084...
>2017-06-26 14:28:21,615 sllurp.llrp: INFO: will reset reader state on connect
>2017-06-26 14:28:21,615 sllurp.llrp: INFO: will start inventory on connect
>2017-06-26 14:28:21,615 sllurp.llrp: INFO: using antennas: [1]
>2017-06-26 14:28:21,616 sllurp.llrp: INFO: connected to (u'169.254.0.60', 5084) (169.254.0.60:5084)
>2017-06-26 14:28:21,754 sllurp.llrp: INFO: requested modulation: FM0
>2017-06-26 14:28:21,754 sllurp.llrp: WARNING: Could not find reader mode matching modulation=FM0 and Tari=7140
>2017-06-26 14:28:21,754 sllurp.llrp: INFO: using reader mode: {'C': 0, 'MaxTari': 16670, 'M': 2, 'PIE': 1500, 'StepTari': 0, 'R': 1, 'FLM': 0, 'MinTari': 16670, 'ModeIdentifier': 0, 'BDR': 436000, 'Mod': 0}
>2017-06-26 14:28:21,754 sllurp.llrp: INFO: stopping politely
>2017-06-26 14:28:21,761 sllurp.llrp_proto: INFO: will report every ~N=1 tags
>2017-06-26 14:28:21,761 sllurp.llrp: INFO: starting inventory

these massages are what I got. And I used updated version by 'git pull' and reinstall.

ransford, 2017-06-26:
I'm not sure why it's choosing those reader mode settings; I'll debug against my own R420.

whitecloudy, 2017-06-27:
I found that this problem is occur only when I tried to change my modulation setting and Tari setting. This command is working

>sllurp inventory -m FM0 (my reader's ip address)

But when I try to change both at the same time, that problem comes out.

ransford, 2017-06-27:
If you're using the WISP5, perhaps you could try the command line that worked in #63.

whitecloudy, 2017-07-03:
I tired but there is an error

> Error : no such option: --mode-identifier

ransford, 2017-07-03:
`sllurp version` should say `0.2.4`:
```shell
% sllurp version
0.2.4
% sllurp inventory --help
Usage: sllurp inventory [OPTIONS] [HOST]...

Options:
  -p, --port INTEGER
  -t, --time FLOAT                seconds to inventory
  -n, --report-every-n-tags INTEGER
                                  issue a TagReport every N tags
  -a, --antennas TEXT             comma-separated list of antennas to use
                                  (0=all; default 1)
  -X, --tx-power INTEGER          transmit power (default 0=max power)
  -M, --modulation [FM0|M2|M4|M8|WISP5|WISP5pre]
                                  modulation (default=M4)
  -T, --tari INTEGER              Tari value (default 0=auto)
  -s, --session INTEGER           Gen2 session (default 2)
  --mode-identifier INTEGER       ModeIdentifier value
  --mode-index INTEGER            ModeIndex value
  -P, --tag-population INTEGER    Tag Population value (default 4)
  -r, --reconnect                 reconnect on connection failure or loss
  --help                          Show this message and exit.
```

Please `git checkout v0.2.4` and try again?

whitecloudy, 2017-07-03:
I'm using Master branch right now. And I think there is no v0.2.4 branch in this repo.
And sllurp said there is no such command right now. Maybe I'm using older version.

ransford, 2017-07-03:
You're right, I'm sorry -- I had forgotten to push the `v0.2.4` tag (which is already merged into `master`).  You should be able to `git checkout v0.2.4` now.

whitecloudy, 2017-07-03:
Now I am using v0.2.4 and I can use 'mode identifier' but still there is a problem

> ~/sllurp$ sllurp inventory -n 1 -T 7140 -M WISP5 --mode-identifier 0 169.254.0.60
> 2017-07-03 14:36:47,150 sllurp.verb.inventory: WARNING: recommended Tari for WISP5 is 6250
> 2017-07-03 14:36:47,150 sllurp.llrp: INFO: connecting to 169.254.0.60:5084...
> 2017-07-03 14:36:47,151 sllurp.llrp: INFO: will reset reader state on connect
> 2017-07-03 14:36:47,151 sllurp.llrp: INFO: will start inventory on connect
> 2017-07-03 14:36:47,151 sllurp.llrp: INFO: using antennas: [1]
> 2017-07-03 14:36:47,152 sllurp.llrp: INFO: connected to (u'169.254.0.60', 5084) (169.254.0.60:5084)
> 2017-07-03 14:36:47,300 sllurp.llrp: INFO: using reader mode: {'C': 0, 'MaxTari': 16670, 'M': 2, 'PIE': 1500, 'StepTari': 0, 'R': 1, 'FLM': 0, 'MinTari': 16670, 'ModeIdentifier': 0, 'BDR': 436000, 'Mod': 0}
> 2017-07-03 14:36:47,301 sllurp.llrp: INFO: stopping politely
> 2017-07-03 14:36:47,307 sllurp.llrp_proto: INFO: will report every ~N=1 tags
> 2017-07-03 14:36:47,307 sllurp.llrp: INFO: starting inventory
> 2017-07-03 14:36:49,974 sllurp.llrp: INFO: stopping politely
> 2017-07-03 14:36:49,974 sllurp.llrp: INFO: will disconnect when stopped
> 2017-07-03 14:36:49,983 sllurp.verb.inventory: INFO: no tags seen
> 2017-07-03 14:36:49,986 sllurp.llrp: INFO: lost connection: Connection to the other side was lost in a non-clean fashion: Connection lost.
> 2017-07-03 14:36:49,986 sllurp.verb.inventory: INFO: total # of tags seen: 0 (0 tags/second)
> 2017-07-03 14:36:49,986 sllurp.llrp: INFO: disconnecting

Modulation mode is still not WISP5(or FM0) and Tari hasn't changed.

ransford, 2017-07-06:
Ah, interesting; mode 0 is probably wrong in your scenario.  Could you run `sllurp --debug --logfile pr64.log.txt inventory -n 1 -T 7140 -M WISP5 --mode-identifier 0 169.254.0.60` and post the file `?  Let's look through the capabilities blob for the modes your reader supports.

whitecloudy, 2017-07-06:
[pr64.log.txt](https://github.com/ransford/sllurp/files/1126839/pr64.log.txt)

Here is the log file.

ransford, 2017-07-06:
Ah, I see that your reader numbers the modes with Tari=6250 as 1000, 1002, 1003, and 1004.  Try `sllurp inventory -n 1 -M WISP5 -T 6250 --mode-identifier 1000 169.254.0.60 ` and see if that helps?  If that doesn't work, try `--mode-identifier 1002` and so on, up to 1004.

(Lines 543 through 641 of your `pr64.log.txt` show your reader's permissible modes.)

whitecloudy, 2017-07-07:
I checked it and it worked......(though it didn't answered my WISP5's respond) I guess this is my reader's limitation. I really really appreciate your help.

ransford, 2017-07-07:
Glad it worked.  I'm going to close this issue, since it sounds like the problem is fixed.

It sounds as though it might be helpful to have a `--list-modes` option that lists the mode settings for a given reader and then exits.

closed 2017-07-07 by ransford