sllurp/sllurp #70: Continuously reading TID from RFID tags

alpharen, 2017-08-30:
Hi Ben,
I am using Impinj Speedway Revolution R220, when reading TID instead of EPC from RFID tags, it only reads 5 times and then start reading EPC instead of TID, how to continuously reading TID from RFID tags, Thanks.

ransford, 2017-08-30:
Is this behavior you're observing with some invocation of `sllurp inventory`?  If so, could you please tell me exactly what command you ran so I can try to reproduce the issue?  Also, is this with a tagged release of sllurp (e.g. v0.3.0) or with `develop`?

alpharen, 2017-08-31:
I use sllurp v0.3.1, here is the command and output I get. It outputs with OpSpecResult field 5 times and then this field disappeared.

$ python access.py 192.168.1.100 -mb 2 -r6
2017-08-31 11:19:17,131 sllurp: INFO: log level: INFO
2017-08-31 11:19:17,131 sllurp.llrp: INFO: connecting to 192.168.1.100:5084...
2017-08-31 11:19:17,132 sllurp.llrp: INFO: will reset reader state on connect
2017-08-31 11:19:17,132 sllurp.llrp: INFO: will start inventory on connect
2017-08-31 11:19:17,132 sllurp.llrp: INFO: using antennas: (1,)
2017-08-31 11:19:17,132 sllurp.llrp: INFO: transmit power: {1: 0}
2017-08-31 11:19:17,133 sllurp.llrp: INFO: connected to ('192.168.1.100', 5084) (192.168.1.100:5084)
2017-08-31 11:19:17,272 sllurp.llrp: INFO: using reader mode: {u'C': 0, u'MaxTari': 20000, u'M': 3, u'PIE': 2000, u'StepTari': 0, u'R': 1, u'FLM': 0, u'MinTari': 20000, u'ModeIdentifier': 3, u'BDR': 320000, u'Mod': 3}
2017-08-31 11:19:17,411 sllurp.llrp: INFO: stopping politely

2017-08-31 11:19:17,512 sllurp: INFO: no tags seen
2017-08-31 11:19:17,546 sllurp: INFO: saw tag(s): [{u'AccessSpecID': (1,),
  u'AntennaID': (1,),
  u'EPC-96': '1122334455660147263010a9',
  u'LastSeenTimestampUTC': (1504149642286737,),
  u'OpSpecResult': {u'OpSpecID': 0,
                    u'ReadData': '\xe2\x004\x12\x010\xfa\x00\x0b\x96\x10\xa9',
                    u'ReadDataWordCount': 6,
                    u'Result': 0},
  u'PeakRSSI': (-32,),
  u'TagSeenCount': (1,)}]

2017-08-31 11:19:17,664 sllurp: INFO: no tags seen
2017-08-31 11:19:17,803 sllurp: INFO: saw tag(s): [{u'AccessSpecID': (0,),
  u'AntennaID': (1,),
  u'EPC-96': '112233445566334433f910c3',
  u'LastSeenTimestampUTC': (1504149642438196,),
  u'PeakRSSI': (-43,),
  u'TagSeenCount': (1,)}]

ransford, 2017-08-31:
I was able to reproduce this behavior: only 5 reads.

Try the `feature/accessop-forever` branch of this repo.  Does it solve the problem?  If so, I'll clean it up a little bit.

alpharen, 2017-08-31:
Just tried the feature/accessop-forever branch, it produce same result, only 5 times.

ransford, 2017-08-31:
Can you send a log with `--debug`?  Here's what I get:

```
2017-08-31 10:29:28,429 sllurp.llrp: DEBUG: AccessSpec: {u'AccessReportSpec': {u'AccessReportTrigger': 1}, u'C': False, u'AntennaID': 0, u'AccessSpecStopTrigger': {u'AccessSpecStopTriggerType': 0, u'OperationCountValue': 0}, u'ProtocolID': 1, u'AccessCommand': {u'OpSpecParameter': {u'WordPtr': 0, u'AccessPassword': 0, u'WordCount': 6, u'MB': 2, u'OpSpecID': 0}, u'TagSpecParameter': {u'C1G2TargetTag': {u'MB': 0, u'MaskBitCount': 0, u'M': 1, u'TagData': u'', u'DataBitCount': 0, u'Pointer': 0, u'TagMask': u''}}}, u'AccessSpecID': 1, u'Type': 207, u'ROSpecID': 0}
```

You should see `{u'AccessSpecStopTriggerType': 0, u'OperationCountValue': 0}`... that's what makes the reader run the AccessSpec forever.  As you can see from 6c27523c6b19c2a82b9cc6ebee470c953abae035, there was previously a 5-time limit in `OperationCountValue`.

alpharen, 2017-09-01:
we did change the parameter, it solved the problem. as you suggested it would be good to add a parameter to do this.
accessStopParam={'AccessSpecStopTriggerType': 0, 'OperationCountValue': 0})

alpharen, 2017-09-01:
we changed the parameter, it solved the problem. as you suggested it would be good to add a parameter to do this.
accessStopParam={'AccessSpecStopTriggerType': 0, 'OperationCountValue': 0})

ransford, 2017-09-07:
I just made two changes in 345371a:
- Use `sllurp access` instead of `python -m sllurp.access`... the command-line arguments should be largely the same.  (See `sllurp access --help`.)
- You can specify `-c`/`--count` and this value will be passed to `OperationCountValue`.

commit 345371ab (2017-09-07): Merge branch 'feature/accessop-forever' into develop

Fixes #70
commit 6c27523c (2017-09-07): quick hack to remove default 5-access limit

(a better approach would be to make the access limit a parameter to the
startAccess function)

fixes #70
closed 2017-09-07 by ransford