sllurp/sllurp #72: exception.TypeError: Data must not be unicode.

leonzhu1985, 2017-09-13:
Got this error while in send_KEEPALIVE_ACK. This function is in llrp.py. I am using a Zebra RFID reader. Would you please look at this issue? Thanks. 

The dump is below, 

 2017-09-12 22:25:05,510 sllurp.llrp: INFO: <READER_EVENT_NOTIFICATION>
        <Ver>1</Ver>
        <Type>63</Type>
        <ID>286997</ID>
        <ReaderEventNotificationData>
        </ReaderEventNotificationData>
</READER_EVENT_NOTIFICATION>
2017-09-12 22:25:05,538 sllurp.llrp: INFO: <READER_EVENT_NOTIFICATION>
        <Ver>1</Ver>
        <Type>63</Type>
        <ID>286998</ID>
        <ReaderEventNotificationData>
        </ReaderEventNotificationData>
</READER_EVENT_NOTIFICATION>
2017-09-12 22:25:05,569 sllurp.llrp: INFO: <KEEPALIVE>
        <Ver>1</Ver>
        <Type>62</Type>
        <ID>287000</ID>
</KEEPALIVE>

Unhandled Error
Traceback (most recent call last):
  File "c:\python27\lib\site-packages\twisted\python\log.py", line 103, in callWithLogger
    return callWithContext({"system": lp}, func, *args, **kw)
  File "c:\python27\lib\site-packages\twisted\python\log.py", line 86, in callWithContext
    return context.call({ILogContext: newCtx}, func, *args, **kw)
  File "c:\python27\lib\site-packages\twisted\python\context.py", line 122, in callWithContext
    return self.currentContext().callWithContext(ctx, func, *args, **kw)
  File "c:\python27\lib\site-packages\twisted\python\context.py", line 85, in callWithContext
    return func(*args,**kw)
--- <exception caught here> ---
  File "c:\python27\lib\site-packages\twisted\internet\selectreactor.py", line 149, in _doReadOrWrite
    why = getattr(selectable, method)()
  File "c:\python27\lib\site-packages\twisted\internet\tcp.py", line 208, in doRead
    return self._dataReceived(data)
  File "c:\python27\lib\site-packages\twisted\internet\tcp.py", line 214, in _dataReceived
    rval = self.protocol.dataReceived(data)
  File "c:\python27\lib\site-packages\twisted\protocols\basic.py", line 578, in dataReceived
    why = self.rawDataReceived(data)
  File "c:\python27\lib\site-packages\sllurp\llrp.py", line 661, in rawDataReceived
    self.handleMessage(lmsg)
  File "c:\python27\lib\site-packages\sllurp\llrp.py", line 375, in handleMessage
    self.send_KEEPALIVE_ACK()
  File "c:\python27\lib\site-packages\sllurp\llrp.py", line 687, in send_KEEPALIVE_ACK
    self.sendLLRPMessage(llrpmsg)
  File "c:\python27\lib\site-packages\sllurp\llrp.py", line 1153, in sendLLRPMessage
    self.transport.write(llrp_msg.msgbytes)
  File "c:\python27\lib\site-packages\twisted\internet\_newtls.py", line 191, in write
    FileDescriptor.write(self, bytes)
  File "c:\python27\lib\site-packages\twisted\internet\abstract.py", line 348, in write
    raise TypeError("Data must not be unicode")
exceptions.TypeError: Data must not be unicode

2017-09-12 22:25:05,571 sllurp.llrp: INFO: lost connection: Data must not be unicode
2017-09-12 22:25:05,572 sllurp: INFO: total # of tags seen: 700 (41 tags/second)

ransford, 2017-09-13:
Try the latest `develop`... did d6c9b9ef3ec80d009b3a3bb634feeb0d1a829374 solve the problem?

leonzhu1985, 2017-09-13:
👍 Nice job. It solved the problem.

closed 2017-09-13 by leonzhu1985