sllurp/sllurp #75: Basic tags reading

sync667, 2017-10-06:
Hello,

I know it's maybe stupid question, but we can't get it at all. 
We want to use Sllurp in our app and read every tag that is in antena range. So when tag is in range we check it for example every second and put it to array so we could use this information next.

Basically our problem is that when we try to read tags by sllurp command or by using code connection we just get only tags that are new to reader. So easily saying tags that where read before don't get read any more. What we do wrong?

ransford, 2017-10-06:
That's not by design.  You should see a callback *every* time the reader issues an RO_ACCESS_REPORT.  Could you provide more details to help me try to reproduce the problem?

https://www.chiark.greenend.org.uk/~sgtatham/bugs.html

sync667, 2017-10-06:
We use latest master clone. Our reader is Zebra FX7500 with is connected over local network. We could use there Windows Zebra API, but we need it on Linux so that's why LLRP. Normally API works well, but like i wrote Sllurp don't give us what we need.

Setup - reader, Ubuntu 16.04, but tested on Windows and get same. It's not any difference if we use your sllurp commands or just write connector with factory to read inventory by own.

So we use sllurp inventory ip -n 1 to 50 time parameter 1 second 
Then we get connection and tags starts to read. When we put in antenna range our big tags roll we get etags in console spamming, but when we use one tag on phone for example it's just read it just one time and after that it's just outputs no tags in range. It's not just when we have instance of command run, restart it don't read that tag anymore too. I think that only reader restart allow us to read that tag again, but i'm not sure because i can't test it right now.

We tried switching factory flags too, but nothing changed. I really don't know what to add more, because we use just simple inventory. We tested tornado example too and get the same result.

ransford, 2017-10-08:
So you're running `sllurp inventory <ip> -n 1 -t 50`?

ransford, 2017-10-08:
If you attach some debug logs (`sllurp -d -l myfile.log inventory ...`), I can take a look at them to see whether there's something obviously wrong.

I don't have a Zebra reader, unfortunately.

sync667, 2017-10-12:
https://pastebin.com/41cWJB5j

ransford, 2017-10-15:
It looks like there's a 45-second gap in your log:
```
2017-10-12 12:37:49,934 sllurp.llrp: DEBUG: state change: STATE_SENT_ENABLE_ROSPEC -> STATE_INVENTORYING
2017-10-12 12:38:32,970 sllurp.llrp: INFO: stopping politely
```
... after which there are 94 EPCs reported.  This is normal behavior when you *don't* specify a value for `-n`.

I see two tags with TagSeenCount > 1:
- e2806890000000001d7e7bef (seen 2 times)
- e2806890000000001d7e7bea (seen 3 times)

Try changing some of the variables you see when you run `sllurp inventory --help`.  Perhaps changing the tag population, session, or modulation scheme will help?

sync667, 2017-12-11:
Think's that issue is gone after last merge :)

closed 2017-12-11 by sync667

ransford, 2017-12-11:
Great!