sllurp/sllurp #76: Targeted Antenna Tag Readings

harokira, 2017-10-25:
Hello!

I am using Impinj R420 Reader. Everything is working fine until I tried to the scenario below:

My setup, I place:
- EPC: 3039606343b916c000154213 on antenna#1
- EPC: 30396062c3b02440000002ee on antenna#2

Below is the command i used and result given:

`sllurp inventory <reader ip> --antennas 2`

sllurp.verb.inventory: INFO: saw tag(s):
 [{u'AntennaID': (2,),
  u'EPC-96': '3039606343b916c000154213',
  u'LastSeenTimestampUTC': (1508917299346036,),
  u'PeakRSSI': (-34,),
  u'TagSeenCount': (38,)},
 {u'AntennaID': (2,),
  u'EPC-96': '30396062c3b02440000002ee',
  u'LastSeenTimestampUTC': (1508917299345483,),
  u'PeakRSSI': (-53,),
  u'TagSeenCount': (38,)}]

Instead of getting one RFID's EPC, i'm getting both. 
Is there a configuration for me to disable antenna#1 while antenna#2 is scanning?

Thank you!

harokira, 2017-10-25:
Hi Ransford, 

I’m using Impinj's Mini-Guardrail Antenna the read distance is 7.5cm... I place both Antenna around 2meters apart. Therefore, im very sure that it is no because of Antenna 2 is reading the tag placed on Antenna 1.

ransford, 2017-10-25:
Understood, but the *reader* is reporting that it saw both tags on Antenna 2.  Maybe you could turn down the transmit power?

harokira, 2017-10-25:
You gave me a good idea! I shall try turn down Antenna#1 transmit power when Antenna#2 is scanning and vice versa tomorrow! As im not at my work desk. Thanks!

harokira, 2017-10-26:
Thank ransford! Its works!