sllurp/sllurp #79: Trouble reading temperature sensor data from EM 4325 IC

redredder007, 2018-05-11:
Hi
EM 4325 IC has the on board temperature sensor on it. I am trying to read the 256th word in its user memory bank using the access.py .However, I am not understanding this tag read rata. Here is it for example:

2018-05-11 16:01:42,200 sllurp: INFO: saw tag(s): [{'AccessSpecID': (1,),
  'AntennaID': (1,),
  'EPC-96': b'0080b0443c0000001210b4b6',
  'LastSeenTimestampUTC': (1526056288899394,),
  'OpSpecResult': {'OpSpecID': 0,
                   'ReadData': b'\x00^\x00\x00',
                   'ReadDataWordCount': 2,
                   'Result': 0},
  'PeakRSSI': (-49,),
  'TagSeenCount': (1,)}]

Why is it reading \x00^\ does this mean there is a decoding issue? But it works well if I am reading any other data for example EPC code. I have tested reading using the impinj's multi reader software and it works perfectly fine. So I am guessing something is not happening right when I use Sllurp.

Did anyone successfully read EM 4325's temperature data using sllurp?

Also, how do I selectively read only one tag from access.py instead of reading all tags. Thanks for your help in advance.

ransford, 2018-05-11:
That `ReadData` contains a four-byte sequence, which you'll have to decode according to the EM 4325's datasheet.  It looks like the second byte is nonzero...
```python
>>> b = b'\x00^\x00\x00'
>>> list(b)
[0, 94, 0, 0]
```

There's not currently an option to read from only one tag; that could be a separate issue (or, better yet, a pull request).

ransford, 2018-05-11:
To give a more complete answer: if you're building an application to respond to temperature tags, you'll probably write your own program that `import`s sllurp.  The "Reader API" section of the README shows a skeletal example.  Your tag report callback will receive chunks like the one you're looking at, and your callback's job will be to extract & parse the `ReadData` field, then do whatever it wants with the value it finds.

If you're feeling more adventurous and you're building a web app, you might want to try the Tornado example in the sllurp repo; it uses WebSockets to send tag data to a webpage as fast as you want.

ransford, 2018-05-11:
One thing you can do in your tag callback: skip any tags in the tag report that don't match a set of tags you care about.

redredder007, 2018-05-11:
@ransford  this looks interesting. How did you convert  b = b'\x00^\x00\x00' to [0, 94, 0, 0]?
If the second byte is non-zero for the same case it should have been like b'\x00\x94\x00'? Why does the decoding fail only when reading user memory (temperature data) and not while reading EPC where the consecutive bytes are non zero as well?
Sorry for my ignorance with the sllurp but I am catching up!

closed 2021-01-24 by ransford