sllurp/sllurp #80: Trouble reading with rswr-400

thoquine, 2018-05-27:
Hi,

I'm currently having problem when running _sllurp inventory ip.add.re.ss_. I'm using the rswr-400 reader from Radiant Sensors. From what I can see, there is a problem with ADD_ROSPEC_RESPONSE but I have no clue how to resolve this. 

Here is the --debug result: https://pastebin.com/B6Cn7K5x

Help would be greatly appreciated.

Thanks!

ransford, 2018-05-27:
My guess is that something about the ROSpec message we send is making the reader unhappy.  If you look at line 848 of your log, you can see an easy-to-read XML version of the ROSpec that's being sent to the reader.  The reader responds with `an error occurred with a field of this message` and says the errant field is the field with type 224, which is `RFTransmitter`.

The `RFTransmitter` field we're sending looks like this:
```xml
                                        <HopTableId>1</HopTableId>
                                        <ChannelIndex>1</ChannelIndex>
                                        <TransmitPower>81</TransmitPower>
```

... so it's probably an issue with one of those fields.  Maybe try changing transmit power?  See the `-X` option of `sllurp inventory` (and try `-X 0` for maximum power).

Thanks for attaching the log -- that was helpful!

thoquine, 2018-05-27:
It looks like TransmitPower is the same with -X 0 when I tried it. (81) 
Is there any other way to change it? 

Here is the new log with -X 0 https://pastebin.com/FHW45xxA

Thanks again for helping me and for the fast response!

ransford, 2018-05-28:
It might also be complaining about HopTableId and ChannelIndex, especially if you're not in the US region.  Check out section 13.2.6.2 of the LLRP spec in `docs/`.  Currently the HopTableId and ChannelIndex are hard coded in https://github.com/ransford/sllurp/blob/develop/sllurp/llrp_proto.py#L3280.  Perhaps try hard-coding other values and then we can figure out a better solution (probably a new configuration parameter).

ransford, 2018-07-10:
Did you figure this out?

closed 2021-01-24 by ransford