sllurp/sllurp #82: Trouble getting sllurp to run

sophiaabraham96, 2018-06-27:
I have been trying to install the library from github and after cloning and entering sllup, all the commands are not recognized and I am having trouble fixing that, any insight as to how I can properly get it running?

ransford, 2018-06-27:
Here's the simplest sequence of steps:
```shell
$ python3 -m venv .venv
$ source .venv/bin/activate
(.venv) $ pip install sllurp
(.venv) $ sllurp inventory --help
```

I won't be able to help much beyond that, as I'm not sure what you've tried, what platform you're on, et cetera.

sophiaabraham96, 2018-06-27:
Thank you for such a rapid response! And I may have installed it incorrectly or missing something but really new to all of this so maybe I can provide greater clarity. I downloaded git and python and added to the path but then none of the commands are being found. Did I miss a step in the middle? I can enter the sllurp directory and then when I try to enter virtualenv .venv the response is "command not found". I am using a windows machine and running from a git bash terminal.

ransford, 2018-06-27:
sllurp isn't special with respect to installation, so I would suggest finding a good tutorial that specifically applies to your configuration.  If you can `pip install requests`, for example, then you can `pip install sllurp`!

sophiaabraham96, 2018-06-27:
Hmm so an error of "bash: .venv/bin/activate: No such file or directory" would be a configuration issue I would need to fix?

sophiaabraham96, 2018-06-27:
Wait I think I need Microsoft Visual installed and I did not have that! I will try it again, I really appreciate all your help!! :)

sophiaabraham96, 2018-06-27:
[image]

I was wondering if you did have a tutorial available on utilizing sllurp with the wisp5 in addition to the readme? Lol I seem to be very bad at this :)

thijmenketel, 2018-06-28:
Make sure that you are running it through python3 as some of the methods are only available in that.

sophiaabraham96, 2018-06-28:
I thought there were incompatibilities with python3? I guess I will attempt that and see if the issues are resolved, thank you :)

sophiaabraham96, 2018-06-29:
I was able to get it properly installed :) I just needed to download some window binaries and place them in the correct folders to properly get it working, thank you both for the suggestions and help! :)

ransford, 2018-07-01:
Yes, it's mostly a library for building your own applications.  Have fun!

closed 2018-07-01 by ransford