sllurp/sllurp #84: Error with sllurp access command

farisshajahan, 2018-07-02:
I'm not sure if I am doing something wrong but running `sllurp access <myreaderIP>` throws us some errors.

```
2018-07-02 14:32:25,708 sllurp.llrp: INFO: connecting to <myreaderIP>:5084...
2018-07-02 14:32:25,709 sllurp.llrp: INFO: will reset reader state on connect
2018-07-02 14:32:25,709 sllurp.llrp: INFO: will start inventory on connect
2018-07-02 14:32:25,709 sllurp.llrp: INFO: using antennas: (1,)
2018-07-02 14:32:25,709 sllurp.llrp: INFO: transmit power: {1: 0}
2018-07-02 14:32:25,710 sllurp.llrp: INFO: connected to (<myreaderIP>, 5084) (<myreaderIP>:5084)
2018-07-02 14:32:25,935 sllurp.llrp: INFO: using reader mode: None
2018-07-02 14:32:25,950 sllurp.llrp: INFO: stopping politely
2018-07-02 14:32:25,972 sllurp: INFO: no tags seen
2018-07-02 14:32:25,978 sllurp.llrp: INFO: Impinj search mode? None
2018-07-02 14:32:25,978 sllurp.llrp: INFO: starting inventory
2018-07-02 14:32:25,998 sllurp.llrp: ERROR: panic(): ('ENABLE_ROSPEC failed',)
2018-07-02 14:32:25,998 sllurp.llrp: ERROR: startAccess requires readWords or writeWords.
2018-07-02 14:32:25,999 sllurp.llrp: ERROR: Traceback (most recent call last):
  File "/home/dladmin/test/.venv/lib/python3.5/site-packages/sllurp/llrp.py", line 558, in handleMessage
    self.processDeferreds(msgName, lmsg.isSuccess())
  File "/home/dladmin/test/.venv/lib/python3.5/site-packages/sllurp/llrp.py", line 357, in processDeferreds
    d.callback(self.state)
  File "/home/dladmin/test/.venv/lib/python3.5/site-packages/twisted/internet/defer.py", line 459, in callback
    self._startRunCallbacks(result)
  File "/home/dladmin/test/.venv/lib/python3.5/site-packages/twisted/internet/defer.py", line 567, in _startRunCallbacks
    self._runCallbacks()
--- <exception caught here> ---
  File "/home/dladmin/test/.venv/lib/python3.5/site-packages/twisted/internet/defer.py", line 653, in _runCallbacks
    current.result = callback(current.result, *args, **kw)
  File "/home/dladmin/test/.venv/lib/python3.5/site-packages/sllurp/llrp.py", line 274, in _setState_wrapper
    self.setState(args[0], **kwargs)
  File "/home/dladmin/test/.venv/lib/python3.5/site-packages/sllurp/llrp.py", line 269, in setState
    fn(self)
  File "/home/dladmin/test/.venv/lib/python3.5/site-packages/sllurp/verb/access.py", line 62, in access
    accessStopParam=accessStopParam)
  File "/home/dladmin/test/.venv/lib/python3.5/site-packages/sllurp/llrp.py", line 907, in startAccess
    raise LLRPError('startAccess requires readWords or writeWords.')
sllurp.llrp_errors.LLRPError: startAccess requires readWords or writeWords.

Unhandled error in Deferred:
```

farisshajahan, 2018-07-02:
The inventory keeps running though and stops with ^C

sophiaabraham96, 2018-07-02:
make sure you substitute "<myreader IP> with the address of your reader. And then try "sllurp access --help" and you can see the options that come under the command

farisshajahan, 2018-07-02:
@sophiaabraham96 I did substitute. I put in `myreaderIP` here on purpose.

sophiaabraham96, 2018-07-02:
@farisshajahan yeah I figured, but just in case! The error states that you need readWords or writeWords the sllurp access --help will provide command line options on how to read and write into the memory unless your attempting to do something else with the access command?

farisshajahan, 2018-07-03:
Yeah. The issue was with me not specifying how many words to read with the `-r` option. Thanks.

closed 2018-07-03 by farisshajahan