sllurp/sllurp #85: Read status of GPIO contacts and set them.

gabrieletassoni, 2018-07-02:
Hi,
I'm testing the python API using tornado example from this repository, it looks that, as an inventory, the demo works, by outputting to the web interface the id of the read tags.
I'm trying to figure out how to interact with the GPIO onboard of the antenna I'm using (Zebra FX7500), from what I understood from LLRP specification is something doable, but I'm a bit stuck. Could you help me find a way? Is there a documented api method which I can import into python example to test a way to interact with GPIO while the antenna keeps reading tags?
Thank you very much, bye,
Gabriele

ransford, 2018-07-04:
sllurp doesn't have good GPIO support, but I could help you figure out how to improve it.  How would you like to "interact with" the onboard GPIO?

ransford, 2018-07-10:
Did you figure out what you'd like to do with GPIOs?  If you've decided not to pursue this route, you can close this issue, or I will after a while.

gabrieletassoni, 2018-07-11:
Hi thank you, sorry for late reply, but I started some holidays. ;-) Anyway, I'd like to interact with gpio in two ways: 

First One: I would be able to make a method to ask for the status of a gpi or a gpo and syncronously get that status. And a method to set syncronously a gpo value (high or low).
Second One: I would like to manage an event, which asyncronously gets informed by the fx7500 about status changes of a gpio, so that I could execute a method upon status changes.

ransford, 2018-07-11:
Congratulations, you're the first person to ask about GPIO support!  It's currently missing from sllurp.  I will take a close look at the LLRP spec and try to figure out how difficult it would be to add the mechanisms you're describing.

wcoltters, 2019-02-16:
@ransford 

Hi, I did some tests yesterday to activate the GPIO outputs of a Impinj R420 and it is working.
I just did some modifications to llrp.py and llrp_proto.py to add the message structure, encode function and a test function trying to follow your same architecture. It turns out the GPIO are sent with a LLRP configuration message and it is very simple in structure.
I'm not sure I understand your architecture with deferreds, I prefer inline callbacks, but I just used them in the test function send_SET_READER_CONFIG_GPO.
I attached the modified files for you to revise them and see if this can be add to your program.
Thank you for your time.
[llrp.zip](https://github.com/ransford/sllurp/files/2871994/llrp.zip)

athulpn, 2019-06-12:
Any update on this?

athulpn, 2019-06-12:
Is it possible to set a GPO value from the tag_report_callback function? I am trying to fire an alarm if a specific tag is found.

ransford, 2019-06-12:
We still don't have GPIO support. Pull requests welcome!

closed 2021-01-24 by ransford