sllurp/sllurp #86: Getting reader info after connection

djacobson, 2018-07-05:
Please excuse my newness to this library. I was wondering if a preferred way exists of getting any stats or properties of a reader after successfully connecting to it? i.e.: not via logger INFO but, maybe, props stored in LLRPClient(...) ? :

fac = LLRPClientFactory(...
...
reactor.connectTCP(host, args.port, fac, timeout=_READER_TCP_CONNECT_TIMEOUT_S)
...
reactor.addSystemEventTrigger('before', 'shutdown', polite_shutdown, fac, tagreader)
# Start server & connect to readers
reactor.run()
...
2018-07-05 10:17:30,525 sllurp.llrp: INFO: connected to ('speedwayr-10-3f-93', 5084) (192.168.1.51:5084)
2018-07-05 10:17:31,451 sllurp.llrp: INFO: using reader mode: {'ModeIdentifier': 2, 'Mod': 2, 'FLM': 0, 'M': 3, 'BDR': 274000, 'PIE': 2000, 'MinTari': 20000, 'MaxTari': 20000, 'StepTari': 0, 'R': 1, 'C': 0}

Like some of the props logged above. Or, also, (and I realize some of this might be manufacturer specific) other props that might be reported back from the reader like power settings, current reader time, antenna count, serial number, sw version, etc. I'm interfacing with an Impinj R420.

Thank you in advance for any pointers to sample code that might already do something like this.

ransford, 2018-07-10:
If you look at the code for `LLRPClient` you'll find it has a lot of fields.  If you have an instance of `LLRPClient` called `cli` in hand, you can refer to `cli.reader_mode`, `cli.antennas`, `cli.session`, and so on.  Hope that helps.  If there's something you don't see there, feel free to either mention it here or create a pull request.

Thanks!

closed 2018-07-10 by ransford

djacobson, 2018-07-17:
Thank you for this hint! Fyi, I discovered:

pprint.pformat(proto.capabilities)

This is what I was looking for... how to access the properties of the reader AFTER successful connection and its 'capabilities' have been parsed and stored. :)