sllurp/sllurp #87: Adding Tag Contents

sophiaabraham96, 2018-07-06:
If we wanted to add another component to the tag content section how would we go about doing that? And I have been using a WISP5 and modifying code within the simpleAckDemo, but does anyone know where exactly the epc data is taken and displayed when the sllurp inventory command is called?

djacobson, 2018-07-07:
With the inventory command, don't you still register a TagReportCallback func? ...i.e.:

factory.addTagReportCallback(cb)

If so, the epc data would be in the cb (arg):

def cb (tagReport):
    tags = tagReport.msgdict['RO_ACCESS_REPORT']['TagReportData']
    print 'tags:', tags

Were you thinking of adding custom data to the RFID spec? There are ways of encoding rfid tag memory with limited custom data. Otherwise, I personally use the unique epc chip # as a key to a standard dictionary with custom data for my app (on receipt of tags in my tagReport callback).

sophiaabraham96, 2018-07-07:
Haha thank you!! :) But I would like to add to this section basically: 

tag_content_selector={
            'EnableROSpecID': True,
            'EnableSpecIndex': False,
            'EnableInventoryParameterSpecID': False,
            'EnableAntennaID': True,
            'EnableChannelIndex': False,
            'EnablePeakRRSI': True,
            'EnableFirstSeenTimestamp': False,
            'EnableLastSeenTimestamp': True,
            'EnableTagSeenCount': True,
            'EnableAccessSpecID': False
        },

I want to add another specification to display when the tags are read

ransford, 2018-07-10:
@sophiaabraham96 can you be a little more explicit?  The tag content selector basically tells the reader "these are the fields I would like you to mention each time you report on tag sightings."  The reader knows about only a small set of fields.

sophiaabraham96, 2018-07-12:
@ransford I was wondering if I could add a field to this that would display when a tag is present? If I stored data in the FRAM of the micro controller would it be possible to have the application authenticate and then have a verification field to display?

ransford, 2018-07-12:
That's something you could build into an application *on top* of sllurp; this library is just for collecting data from tags.  Check out `sllurp access --help` to see how to read tag memory.

sophiaabraham96, 2018-07-12:
Okay that makes sense! But when I tried using the sllurp access command I was unable to attain the values I wrote to the fram. The command is just `sllurp access -r n <ipadress>` with any additional specifiers that we would want included listed under `sllurp access --help` correct?

sophiaabraham96, 2018-07-13:
[image]

Perhaps I am specifying the wrong word pointer? When I wrote to the FRAM and read it in code composer I can see the values I had written to the FRAM of the microcontroller,  but when I tried issuing the command from sllurp the output is the same as the `sllurp inventory` command

ransford, 2018-07-13:
Oh, I haven't read the WISP5 code recently... you'll have to ask in the WISP project how you should be structuring your access command.  I suspect what you want to do is possible.

sophiaabraham96, 2018-07-13:
Alright Ill ask!! Thank you!! :)

closed 2018-07-16 by sophiaabraham96