sllurp/sllurp #88: Integrating C executable within sllurp

sophiaabraham96, 2018-07-16:
Any suggestions where the best place within the sllurp inventory command to place code to execute a C executable file to run with the command? I am utilizing Python's OS module to do it, but when I placed the code into the sllurp directory and ran the command I did not see the outputs of the executable file and was wondering where I should place it so that it would run each time a new tag is called? 

Thanks :)

ransford, 2018-07-16:
If you're trying to run an executable with each tag sighting, you should probably try to aggregate tag sightings in some way; either decrease the reporting frequency so reports arrive less often, or perform a higher-level aggregation in your tag report callback to control the frequency with which you invoke the executable.

I'm not sure what the problem is from your description.  You can run any code you want inside the tag report callback.  Could you share your callback here, or a small test case illustrating what you're trying to do?

ransford, 2018-07-16:
You'll have to do a little more debugging.  Are you confident your callback is actually running at all?

sophiaabraham96, 2018-07-17:
I had created a separate python script to run the code and I can access and run it with the python command but it did not work when I integrated into the sllurp inventory. May be a dumb question, but do I need to edit the sllurp executable to have it run?

ransford, 2018-07-17:
I recommend following the example in the README.  You don't need to edit the sllurp sources to set up your own script with a tag report callback.  I won't be able to provide any more help on getting it to run; sorry!

sophiaabraham96, 2018-07-17:
I has followed the readme but there were no outputs. With the sllurp commands, the ip address can be specified, is that not needed when we call our own scripts to connect with the reader? We just call the python script with the 'python '_nameofscript_.py' to run it right?

ransford, 2018-07-17:
That really depends on what your `nameofscript.py` has in it.  You can hard-code IP addresses in there if you want.

sophiaabraham96, 2018-07-19:
I hardcoded it but is the minimal example supposed to just return an empty buffer for the tag though?

sophiaabraham96, 2018-07-19:
Yeah, using the default settings for the tag content: 

'tags: [ ]'

ransford, 2018-07-19:
I mean the entire function you're expecting to run every time there's a tag report.  In the [README](https://github.com/ransford/sllurp#reader-api) it's called `cb`.

sophiaabraham96, 2018-07-20:
Oh I just went ahead and followed the readme with the default settings for the tag
def cb (tagReport):
    tags = tagReport.msgdict['RO_ACCESS_REPORT']['TagReportData']
    print 'tags:', tags

ransford, 2018-07-20:
OK, but I don't see where in `cb()` you're attempting to call an external C program.  If you want something to run every time there's a tag report, you have to put it in your callback function (`cb`).

sophiaabraham96, 2018-07-20:
Oh I did not place it in the code yet, I was trying to just get tag data from the python scripts first but can only attain blank arrays for the tag reports.

ransford, 2018-07-20:
I see.  I don't know that I'll be able to help any further with this issue... if you have tags and `sllurp inventory -n1 <your.reader.ip.address>` shows tags being successfully read, then the bug is somewhere in your application code rather than this library.

closed 2018-07-20 by ransford

sophiaabraham96, 2018-07-20:
Well the sllurp inventory call works for every tag, but the python script issues tag callbacks for all the tags except the WISP (which just returns empty banks). I will keep messing around, thanks!