sllurp/sllurp #90: time.monotonic() command not found

abstractspace, 2018-07-25:
I keep getting an error message that involves the time.monotonic() function. I get this error both on my Windows machine and on a Raspberry PI.  I'm using Python 2.7. 

```
pi@raspberrypi:~ $ sllurp -d inventory 10.10.31.58
:0: UserWarning: You do not have a working installation of the service_identity module: 'No module named service_identity'.  Please install it from <https://pypi.python.org/pypi/service_identity> and make sure all of its dependencies are satisfied.  Without the service_identity module, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.
2018-07-25 14:08:11,928 sllurp.cli: DEBUG: inventory args: Args(host=(u'10.10.31.58',), port=5084, time=None, every_n=None, antennas=u'1', tx_power=0, modulation=None, tari=0, session=2, population=4, mode_identifier=None, reconnect=False, impinj_search_mode=None)
2018-07-25 14:08:11,929 sllurp.verb.inventory: INFO: Antenna map: {u'10.10.31.58': {'1': 'Antenna 1'}}
2018-07-25 14:08:11,931 sllurp.llrp: INFO: connecting to 10.10.31.58:5084...
Traceback (most recent call last):
  File "/usr/local/bin/sllurp", line 11, in <module>
    load_entry_point('sllurp==0.4.1', 'console_scripts', 'sllurp')()
  File "/usr/lib/python2.7/dist-packages/click/core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/click/core.py", line 696, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python2.7/dist-packages/click/core.py", line 1060, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python2.7/dist-packages/click/core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python2.7/dist-packages/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/sllurp-0.4.1-py2.7.egg/sllurp/cli.py", line 71, in inventory
    _inventory.main(args)
  File "/usr/local/lib/python2.7/dist-packages/sllurp-0.4.1-py2.7.egg/sllurp/verb/inventory.py", line 122, in main
    start_time = time.monotonic()
AttributeError: 'module' object has no attribute 'monotonic'
```

abstractspace, 2018-07-25:
I think I realized what is going on. time.monotonic() wasn't introduced until Python 3.3 and I'm using 2.7.

closed 2018-07-25 by abstractspace

ransford, 2018-07-26:
Unbroke this behavior in 0c0be39 by using a `monotonic` module that works in Python 2 but degenerates to `time.monotonic()` on Python is 2.0.3.

abhishekmadhu, 2019-06-20:
> I think I realized what is going on. time.monotonic() wasn't introduced until Python 3.3 and I'm using 2.7.

Same here.

Anyways, thank you @ransford for `0c0be39`.