sllurp/sllurp #100: Possible to query for reader status info?

djacobson, 2018-08-09:
I was wondering if it might be possible to query the reader for status info *without* a tag observation report. By examining the sllurp inventory, it seems it only parses information from a TagReportCallback. I'm specifcally looking to get the temperature status apparently exposed by the Impinj API on a periodic basis.

I've looked at the LLRPClient properties like antennas, reader_mode, capabilities, etc.; but, didn't find something close and, these properties seem to be only refreshed by a reset and/or client reconnect, correct?

Am I missing something here? Any advice on an approach I might take would be appreciated.

Thank you.

ransford, 2018-08-09:
You're right about how the client works.  Chapter 5 of `docs/llrp_1_1-standard-20101013.pdf` in this repo explains a "typical" timeline for a session.  The state machine in `LLRPClient` is designed to reflect this typical timeline.

Does the reader send these status messages periodically without being asked for them?  That would be easy to support with a slight modification to the big switch statement in `LLRPClient.handleMessage()`.

djacobson, 2018-08-09:
Thanks for your explanation. That makes sense. I will research that Api feature and get the exact tech detail on it and reopen this issue / enhancement then.

djacobson, 2018-08-11:
After researching this further, I found that the "temperature" indicator is indeed implemented as one of Impinj's many "custom" encoded messages; from their documentation below... I'm not sure if there's a way to extend sllurp to send and receive these type of messages?:

"Custom Extension Encoding"

6.1 Custom Messages

All LLRP custom messages are encoded with a common header to ensure a unique namespace
across all LLRP implementations. This header appears in each of the custom messages below for
completeness. Byte and bit order are shown as the first and second row of the table respectively.
The vendor ID field contains the Impinj Private Enterprise Number (PEN) - 25882. A unique
subtype indicator defines each custom extension message.

ransford, 2018-08-13:
That's helpful to know.  Does the temperature indicator have a message subtype?  If you're not sure what I mean: the message subtype for phase angle (another Impinj extension) is 52.

(Are you looking at a document that's public?)

ransford, 2018-08-13:
That's great!  At first glance, it looks like we have to update `LLRPClient.send_ENABLE_IMPINJ_EXTENSIONS` to ask for this extension at connection time.  I'll take a closer look.

ransford, 2018-08-28:
I started playing around with this in a branch, but it will be a few more days until I can make real progress on it.

djacobson, 2018-08-29:
Great news! I am researching this some too. I am looking into the possibility (user request) of whether the Impinj can automatically detect what antenna port have antennas plugged into them, and then initialize the ports (in other words, not knowing ahead oof time what ant ports you plan to use, just detect them... ???). It sounds backwards to me, but, they say they've seen it done. ? Anyways, the following custom parameters look interesting, in addition to the Temperature one:

4.3.44 ImpinjIntelligentAntennaManagement
4.3.45 ImpinjHubConfiguration
4.3.11 ImpinjReaderTemperature

Any small example you come up with would be invaluable.

ransford, 2018-09-06:
Support for fetching extended configuration values at connection time is happening in branch `feature/100-impinj-capabilities`.  Not quite working yet, but you can see where it's going.

ransford, 2018-09-08:
I just merged some changes into `develop` to implement generic parameter parsing and a new switch `inventory --impinj-extended-configuration` to correctly fetch all extended configuration parameters from Impinj readers.  One more thing to do: implement parsing for specific Impinj `CustomParameter`s such as temperature.

ransford, 2018-09-09:
You can now get temperature from `LLRPClient.configuration['temperature']`.  If you'd like to get access to other configuration parameters, look at the changes in fe6e112 and you can see how to parse other Impinj custom (and non-vendor-specific) fields.

closed 2018-09-09 by ransford

djacobson, 2018-09-13:
This is great! I will test some of this at my next opportunity. Thank you.

commit fe6e1124 (2018-09-09): Merge branch 'feature/100-decode-temperature' into develop

Fixes #100