sllurp/sllurp #97: Multi-antenna setup only reporting from single antenna

thijmenketel, 2018-08-09:
When I attach two (or more) antenna's to my reader (R420), sllurp only reports from a single one. Depending on which option I select (-a for antenna selection) it always selects the first one in the list. I can't really find any documentation on how this should be done. 

Side note, I reach a read rate (in dual target mode and tested multiple DRM modes) of about 45 tags per second, however according to the Impinj spec for the R420 the reader is capable of around 1100 tags per second. Is this a limitation of sllurp or am I not looking in the right place?

ransford, 2018-08-09:
Do you mean both antennas should be able to see the same tags, but your reports all seem to be from the first antenna?

In the applications I build atop sllurp, I often see read rate in the hundreds of tags per second.  Read rate depends on so many factors that it's hard to speculate.  If you're able to, it might be useful to benchmark against the ItemTest or MultiReader [applications](https://support.impinj.com/hc/en-us/articles/208589707) that Impinj offers.  See what it reports for read rate in your configuration and perhaps you'll also find some clues about why the reports seem to be coming from only one antenna.

thijmenketel, 2018-08-09:
Exactly that, depending on how I configure the antennas (-a 1,2) it’s always the first one in the list. 

That’s what I read as well on specs everywhere. I think I might try their software to see the difference. I also ordered some higher performance tags to see if that has an impact. 

In your applications, do you use the standard inventory mode or some other configuration? Maybe I need to strip some of the code. I also rely on the Impinj extensions (phase angle), which might have an impact.

Thanks for the tips!

ransford, 2018-08-13:
I don't have any brilliant insight into why the reader seems to "prefer" one antenna over the other.  I believe the Speedway readers use a "round-robin" strategy to activate antenna ports, so only one antenna is active at any time.  If antennas 1 & 2 are both pointed at the same tags, perhaps the tags are already busy responding to antenna 1 by the time they hear from antenna 2?

You should also confirm that the antennas are working properly individually.  What happens when you specify *only* antenna 2?

thijmenketel, 2018-08-13:
Antenna selection works perfectly in sllurp, the issue is just with multi antenna setups. So when I set -a to 0 only the antenna in the lowest port reports. When I set -a to 2, only antenna 2 reports. Same with -a 2,1. When I set -a to 1,2 it only reports from antenna 1.

ransford, 2018-08-13:
You get different results if you specify "2,1" instead of "1,2"?  (That's kind of surprising!)

thijmenketel, 2018-08-14:
Yes, the antenna that is first in the 'list' reports the tags it sees. Seemed quite weird to me as well.

ransford, 2018-08-19:
The best I can suggest is to use the "dual-target" Impinj search mode, the same kind you get when you specify `sllurp inventory --impinj-search-mode 2`.

Other than that, try a faster CPU?  You may be hitting the performance wall with the pure-Python implementation of protocol message handling.

thijmenketel, 2018-08-21:
I guess it is a performance issue as the highest I can get with a laptop is about 65 tags/s, probably the issue is twisted (and python in general). I don't see a better CPU getting that up to the speed I need unfortunately. On the ItemTest software I can get it over 150 tags/s with a single tag and over 600 tags/s with more tags but that software doesn't report phase info. So I might need to find another, faster way to communicate with the reader and parse the responses.

I'll leave the issue open for the Multi-antenna problem.

thijmenketel, 2018-09-06:
> I've found that my Impinj readers (I have R220 and R420) behave in ways I don't understand when I change the ROReportTrigger parameter.

The combination of the -n option and changing the ROReportTrigger can give some strange results indeed. 

> sllurp is trying to make RFID more accessible by removing the need to understand the details of LLRP, so the intention of -n is to say "send me a report on every Nth tag," and -t is supposed to mean "send me a report every T seconds." But there are so many protocol configuration variables that I'm not sure -n and -t are implemented reasonably now. Patches gladly accepted!

In issue #106 there is a discussion about just this actually. I think @fviard is currently working on a patch that includes a better (higher performance) implementation of the triggers.

ehanson585, 2019-10-22:
I'm currently using a multi-antenna set-up, and I'm experiencing similar errors. When I set the program to read from multiple antennas, the program only reads from the first antenna in the list. 

For example, when I specify to read antennas "1,2", the program will report from antenna 1 but not antenna 2 unless the tag is far enough out of the range of antenna 1 that only antenna 2 receives a signal. Similarly, when I specify to read antennas "2,1", the opposite happens as the program only reports from antenna 2 unless the tag is only in range of antenna 1. Additionally, when I specify the program to read from antenna "0", it behaves in the same way as to when I specify antenna "1,2".

Is there any way to generate reports from both antennas at the same time?

ransford, 2019-10-29:
It seems as though the reader respects the order in which you specify ROSpec > AISpec > InventoryParameterSpec > AntennaConfiguration parameters. The `LLRPROSpec` constructor respects the order of the comma-separated antenna arguments (no reason not to), so the message you're emitting to the reader will differ for `1,2` versus `2,1`.

I don't know why the behavior seems different, or whether the difference should be expected. You could ask your reader's manufacturer for support.

One theory is that tags have a hard time responding to Gen2 protocol messages from two antennas at once because the antennas are activated in order (time multiplexing the channel) but each for less time than it takes to do a full tag read. So, like, perhaps the reader is moving on to power the next antenna before the first antenna has finished inventorying, and tags don't feel they need to respond to anything the second antenna says. That's just a theory. I know time multiplexing is happening among antennas, but I don't know how the timing of the multiplexing interacts with the timing of the Gen2 protocol.

closed 2021-01-24 by ransford