sllurp/sllurp #98: sluurp install

qingbin123, 2018-08-09:
I am following the tutorial to install sllurp on my computer, but the following will occur when following the following steps, may I ask why?

SE@SE-PC026 MINGW64 ~
$ git clone https://github.com/ransford/sllurp.git
Cloning into 'sllurp'...
remote: Counting objects: 2809, done.
remote: Compressing objects: 100% (50/50), done.
remote: Total 2809 (delta 78), reused 96 (delta 67), pack-reused 2692
Receiving objects: 100% (2809/2809), 5.77 MiB | 692.00 KiB/s, done.
Resolving deltas: 100% (1875/1875), done.

SE@SE-PC026 MINGW64 ~
$ cd sllurp

SE@SE-PC026 MINGW64 ~/sllurp (develop)
$ virtualenv .venv
bash: virtualenv: command not found

SE@SE-PC026 MINGW64 ~/sllurp (develop)
$

thijmenketel, 2018-08-09:
You need to install virtualenv and pip:

```
sudo apt-get install python3-pip
sudo pip3 install virtualenv 
```

ransford, 2018-08-09:
@thijmenketel thanks!

Closing since this is a general Python package installation question.  There are some good tutorials on the web that explain all the ins and outs.

I opened #99 to update the README now that pipenv is available everywhere.

closed 2018-08-09 by ransford

qingbin123, 2018-08-10:
Pip and virtualenv must be installed through Python, I am sorry I don't know anything about this, but I am studying hard. 
 
------------------ Original ------------------
From: "Thijmen Ketel"; 
Date: 2018年8月9日(星期四) 晚上10:47
To: "ransford/sllurp"; 
Cc: "qingbin123"; "Author"; 
Subject: Re: [ransford/sllurp] sllurp install (#98)

 

You need to install virtualenv and pip:
 sudo apt-get install python3-pip sudo pip3 install virtualenv

qingbin123, 2018-08-10:
As you  said, I have installed pip and virtualenv, but the code that I used to  run the error in Git bash is still wrong. I don't know what the reason  is. Please enlighten one or two.

ransford, 2018-08-10:
You can also install without a virtualenv.  If you have pip, you can just run `pip install sllurp` to install to your system's Python library directory.

qingbin123, 2018-08-10:
Thank  you very much for your suggestion. My question should be that there is  one less twisted file. I fixed it. The result is as shown in the figure.  Has it been installed successfully?

ransford, 2018-08-13:
Sorry @qingbin123, I'm not sure what you mean, but it sounds like you sorted it out.  Good luck!