sllurp/sllurp #102: Reader API

qingbin123, 2018-08-24:
An error occurs when running to this step,indicating that a parenthesis is missing,but is still wrong after adding paranthesis.

Python 3.6.0 (v3.6.0:41df79263a11, Dec 23 2016, 08:06:12) [MSC v.1900 64 bit (AMD64)] on win32
Type "copyright", "credits" or "license()" for more information.
>>> from sllurp import llrp
>>> from twisted.internet import reactor
>>> import logging
>>> logging.getLogger().setLevel(logging.INFO)
>>> def cb (tagReport):
	tags = tagReport.msgdict['RO_ACCESS_REPORT']['TagReportData']
	print 'tags:',tags
	
SyntaxError: Missing parentheses in call to 'print'

thijmenketel, 2018-08-24:
Maybe an obvious thing but you’re using python 3, which means your print statement needs to be like this:
print(‘stuff’)

But you can also just use logger.info(‘tags: %s’, pprint.pformat(tags))

qingbin123, 2018-08-25:
There was a syntax error at run time, and there was an error at the beginning, and I don't know why, because I'm new to python.

 Python 3.6.0 (v3.6.0:41df79263a11, Dec 23 2016, 08:06:12) [MSC v.1900 64 bit (AMD64)] on win32
Type "copyright", "credits" or "license()" for more information.
>>> from sllurp import llrp
>>> from twisted.internet import reactor
>>> import logging
>>> logging.getLogger().setLevel(logging.INFO)
>>> def cb(tagReport):
 tags = tagReport.msgdict['RO_ACCESS_REPORT']['TagReportData']
 logger.info('tags:%s',pprint.pformat(tags))
 factory = llrp.LLRPClientFactory()
 factory.addTagReportCallback(cb)
 reactor.connectTCP('myreader',llrp.LLRP_PORT,factory)
 reactor.run()
 
[DEBUG ON]
>>> 
[DEBUG OFF]
>>> import MyReader
Traceback (most recent call last):
  File "<pyshell#11>", line 1, in <module>
    import MyReader
  File "D:\python36\MyReader.py", line 2
    >>> from sllurp import llrp
     ^
SyntaxError: invalid syntax
>>>

qingbin123, 2018-08-25:
thanks ! I think the question I have been solved.

qingbin123, 2018-08-25:
I have some questions that I don't understand. Slulp is an application (that is, owning a client) or just a Python library. Sllurp relies on Twisted to communicate with the reader. You need to create an LLRPClientFactory. I don't understand where the LLRPClientFactory is to be created. And I created it in the Python Shell.

thijmenketel, 2018-08-25:
The LLRPClientFactory needs to be created in the script. Take a look at [this](https://github.com/ransford/sllurp/blob/develop/sllurp/verb/inventory.py) inventory file.

I hope that will clear some stuff up.

qingbin123, 2018-08-25:
Do you mean building a script and running it in Python? If this is the case, why did I run the created script in Python without any results but no error, then I entered this minimal example in Python and still had no results.

qingbin123, 2018-08-26:
Do you know the wisp? The GUI in the figure applies to wisp4.1, does it also apply to wisp5?

qingbin123, 2018-08-29:
I am sorry that I couldn't explain it. I added a picture to the email but it didn't show it. It seems that I can't display the image here. The GUI app I mentioned is in the link below.[http://wisp.wikispaces.com/UsingGUIApp](url)

ransford, 2019-02-13:
Wikispaces went away. @qingbin123 if you made any progress here, please feel free to reopen this issue or create a pull request.

closed 2019-02-13 by ransford