sllurp/sllurp #106: Twisted dependency removal

fviard, 2018-08-29:
For using sllurp in embedded systems, i'm a little bit annoyed by the Twisted dependency.
The goal would be that sllurp be self sufficient without such a dependency for python2 and python3.

I have started such a rework, and, in my opinion, that simplify a little the code, allows to better control the behavior and offer more possibilities for evolution.

@ransford : I would appreciate to have your feedback on the current Work In Progress in the following branch:
https://github.com/fviard/sllurp/tree/develop_untwisted

Basically inventory, reset, log, should be working like before.
Tornado is not yet updated but should be easily without issue.
Lock and access still need a little bit more changes in the library.

On the overall thing, there are still a few things  and improvements to do.
But the big work to have something working as in develop should be there.

thijmenketel, 2018-08-30:
This looks very nice! I’ll try it next week on the reader I’ve got available. 

Do you see a possibility that this might be extended to allow multiprocessing for decoding the reports?

fviard, 2018-08-31:
FYI, access and lock are now ported in the branch.

Do you mean doing multiprocessing for the de-serialisation of the TagReports  or to process them by user space.

For user space, one can easily push them in a "multiprocessing pool" python queue once received in the callback.

For the de-serialization, i could be wrong, but I don't think that it is necessary. I think that this step should be fast enough. Have you seen something else?
Anyway this step should not slowdown the reader as his messages will be pushed to the client network connection recv queue. There, it could be an issue only if the recv queue grow too much and too fast compared to the handling of received messages.

ransford, 2018-09-02:
Very cool; I'll take a look.  Here is a short list of features I know are in use in production applications with the current Twisted implementation:
- Connecting to multiple readers at the same time (same Twisted client factory)
- Connect to different types of readers at the same time (different Twisted client factories, different ROSpecs)
- Other protocols being served by the Twisted reactor

Any event loop should be able to replace the Twisted reactor in theory, but Twisted does provide quite a few nice facilities...

thijmenketel, 2018-09-02:
> For the de-serialization, i could be wrong, but I don't think that it is necessary. I think that this step should be fast enough. Have you seen something else?

Well I've been able to speed up the tag reads per second using sllurp (see my comment [here](https://github.com/ransford/sllurp/issues/97#issuecomment-415072375)) and there is a performance bottleneck (at least on my laptop, I'm probably able to use a workstation soon) when reaching about 400 tags per seconds. Using the ItemTest software from impinj I can even get it up to 900 tags per second. 

But I'll try your untwisted code tomorrow and report back!

fviard, 2018-09-03:
Can you give me the exact command line that you used for your benchmark?
(antenna choice and all)
Was it exactly the same one that you used for running with twisted?
Do you see any warning or anything before interrupting?

Also, if you enable the "--debug" flag, is there a performance drop?
Finally, you can try to enable the event reporting for the following events:
                        'ReportBufferFillWarning': True,
                        'ReaderExceptionEvent': True,
in SET_READER_CONFIG, to see if you have anything reported.

fviard, 2018-09-03:
Also, a little bit something that I'm wondering, if you don't use the "-n" option, you should have the same performance between untwisted and twisted, and eventually with itemtest, can you confirm that?
Because in such a case, all the buffering is done on the reader side and a single message is reported for tagreports

thijmenketel, 2018-09-03:
So I run sllurp (twisted and untwisted) with the following command (in a virtualenv though), I do disable the printing of each tag as it's annoying:
```
sllurp inventory -s 2 -P 4 -M WISP5 --impinj-search-mode 2 --impinj-reports 1 192.168.0.100
```
I can confirm that when the -n option is not used the read rate can get as high as the ItemTest can reach. However, as I need the phase info from each tag read I set [this](https://github.com/ransford/sllurp/blob/54527eba0530193e7dea321693dee0273f724160/sllurp/llrp_proto.py#L3427) N to 1, which makes the reader sent every single report of every single tag but in a more efficient manner (see my comment [here](https://github.com/ransford/sllurp/issues/97#issuecomment-415072375)).

Furthermore, enabling debug will impact performance as it prints a lot of stuff. 

When testing this try to use, multiple tags as it will give the highest reads per second.

> Finally, you can try to enable the event reporting for the following events:
'ReportBufferFillWarning': True,
'ReaderExceptionEvent': True,
in SET_READER_CONFIG, to see if you have anything reported.

I can try this but I assume those flags are related to the buffering of reports on the reader side?

fviard, 2018-09-03:
Ok, first thing that is confusing me is how you reach a so high read rate with this reader with just 3 tags.
You have just 1 antenna plugged?

Running with the debug output will show that if it is python that is saturated as the "slow log" should slow down things.
And for enabling these events, I would like to see if it is the reader that is slowing things down because.

It is probably not related, but for N=1, I noticed that there is something that looks badly done currently in the code.
When you set -n 1, the N of ROReportTrigger stays to 0, but you set a AIStopTrigger with the value of N.
To me it looks like very sub optimal as it is like you ask the reader to stop and restart the inventory after each tag found.

So, I think that you should comment out the AIStopTrigger, and put the N value in the ROReportTrigger.
(The only issue when using N>1 with this is that, contrary to its name,  the N correspond to the number of different tags seen before a report is emitted and not a number of tag read events)

fviard, 2018-09-04:
I don't think that i'm limited by that, but for your information if you were, when enabling "impinj-reports", each tag report is approximatively the double the size of normal tag reports.
Because of how custom params fields are defined in the llrp protocol (that is not very well though in my opinion), you have around 14 bytes per custom field for a 2 bytes worth of data.
So for the 3 impinj custom params, that makes around +45 bytes. Whereas the basic report is just around 40 bytes.

thijmenketel, 2018-09-04:
I'll try it tomorrow! 

Yes, the customparams do increase the size of the reports, but  I actually need the phase angle information for my project so running without is not an option.

I will report back tomorrow.

thijmenketel, 2018-09-04:
Small follow up, I looked at your changes and if I understand correctly you set the tcp socket to no delay. If that’s the speed improvement that could theoretically also be applied to the twisted version of sllurp:

transport.setTcpNoDelay(True)

I might test that tomorrow as well.

fviard, 2018-09-04:
No no, it is the other thing.
tcp no delay, is just a small candy. Normally it affects more the "sending" side.
The main thing is the call to "func()" for debug in each function.
First the call is always done even when not in debug mode, but the thing is that to get the func name, this function get it from python stack inspection.
And if you look on internet, you will see that there are a few benchmarks that shows how slow this can be.

fviard, 2018-09-04:
(With an impinj r420 fcc)
> 2018-09-04 23:00:51,279 sllurp.verb.inventory: INFO: total # of tags seen: 7894 (1074 tags/second)
:-)

thijmenketel, 2018-09-05:
I'm having an issue connecting to my reader through the untwisted version. It gives me the following error:

```
2018-09-05 11:16:30,628 sllurp.verb.inventory: INFO: selected recommended Tari of 0 for WISP5
2018-09-05 11:16:30,628 sllurp.llrp: INFO: will reset reader state on connect
2018-09-05 11:16:30,628 sllurp.llrp: INFO: will start inventory on connect
2018-09-05 11:16:30,628 sllurp.llrp: INFO: Enabling Impinj extensions
2018-09-05 11:16:30,628 sllurp.llrp: INFO: using antennas: [1]
2018-09-05 11:16:30,628 sllurp.llrp: INFO: transmit power: {1: 0}
2018-09-05 11:16:30,629 sllurp.llrp: INFO: will reset reader state on connect
2018-09-05 11:16:30,629 sllurp.llrp: INFO: will start inventory on connect
2018-09-05 11:16:30,629 sllurp.llrp: INFO: Enabling Impinj extensions
2018-09-05 11:16:30,629 sllurp.llrp: INFO: using antennas: [1]
2018-09-05 11:16:30,629 sllurp.llrp: INFO: transmit power: {1: 0}
2018-09-05 11:16:30,629 sllurp.llrp: WARNING: Reader not connected. Disconnect is not needed.
2018-09-05 11:16:30,629 sllurp.verb.inventory: INFO: total # of tags seen: 0 (0 tags/second)
2018-09-05 11:16:30,629 sllurp.llrp: WARNING: Reader not connected. Disconnect is not needed.
2018-09-05 11:16:30,629 sllurp.verb.inventory: INFO: total # of tags seen: 0 (0 tags/second)
```

Any ideas?

fviard, 2018-09-05:
Hum, can you run the command with the --debug flag? Nothing in the patch should have such an effect.
Btw, stupid question, do you run sllurp on Linux? or in another os?

thijmenketel, 2018-09-05:
It doesn't give much more:
```
2018-09-05 11:34:19,799 sllurp.cli: DEBUG: inventory args: Args(host=('1', '192.168.0.100'), port=5084, time=None, every_n=None, antennas='1', tx_power=0, modulation='WISP5', tari=0, session=2, population=4, mode_identifier=None, reconnect=False, impinj_search_mode='2', impinj_reports=True)
2018-09-05 11:34:19,799 sllurp.verb.inventory: INFO: selected recommended Tari of 0 for WISP5
2018-09-05 11:34:19,799 sllurp.llrp: INFO: will reset reader state on connect
2018-09-05 11:34:19,799 sllurp.llrp: INFO: will start inventory on connect
2018-09-05 11:34:19,800 sllurp.llrp: INFO: Enabling Impinj extensions
2018-09-05 11:34:19,800 sllurp.llrp: INFO: using antennas: [1]
2018-09-05 11:34:19,800 sllurp.llrp: INFO: transmit power: {1: 0}
2018-09-05 11:34:19,800 sllurp.llrp: INFO: will reset reader state on connect
2018-09-05 11:34:19,800 sllurp.llrp: INFO: will start inventory on connect
2018-09-05 11:34:19,800 sllurp.llrp: INFO: Enabling Impinj extensions
2018-09-05 11:34:19,800 sllurp.llrp: INFO: using antennas: [1]
2018-09-05 11:34:19,800 sllurp.llrp: INFO: transmit power: {1: 0}
2018-09-05 11:34:19,800 sllurp.verb.inventory: INFO: connecting
2018-09-05 11:34:19,800 sllurp.llrp: WARNING: Reader not connected. Disconnect is not needed.
2018-09-05 11:34:19,800 sllurp.verb.inventory: INFO: total # of tags seen: 0 (0 tags/second)
2018-09-05 11:34:19,800 sllurp.llrp: WARNING: Reader not connected. Disconnect is not needed.
2018-09-05 11:34:19,800 sllurp.verb.inventory: INFO: total # of tags seen: 0 (0 tags/second)
```

I run linux actually. When print the exception on the disconnect it gives me an error 22. I had this error with regular twisted sllurp as well but it ran fine.

As I'm an idiot some times I might have run your develop branch yesterday instead of develop_untwisted but I don't know for sure.

fviard, 2018-09-05:
ok :-)
Un-related but I think that I have still a few refinement to do to connect timeout and connect error output for the default apps (verbs)

thijmenketel, 2018-09-05:
Works like a charm! No issues on the disconnect on this.

My max throughput is about 440 tags per second with 5 (shitty) tags. I will test again once I get some higher performance tags (still waiting for shipment). 

In the mean time it might be interesting to apply those performance speedups to the twisted version of sllurp as well. The -n option slows the reader down by a lot which is not that good so as a solution we can use a buffer on the client side instead of the reader side and then only print every n tags. This keeps tag read performance high while still allowing the option to print every n tags.

>Un-related but I think that I have still a few refinement to do to connect timeout and connect error output for the default apps (verbs)

I might check them out but I mostly work with the inventory command.

fviard, 2018-09-05:
Did you use my version directly?
If just with -n 1, you can try the following patch in llrp_proto.py just to see best perfs:

```
@@ -3936,6 +3965,7 @@ class LLRPROSpec(dict):
             else:
                 logger.info('will report every ~N=%d tags',
                             report_every_n_tags)
+            """
             self['ROSpec']['AISpec']['AISpecStopTrigger'].update({
                 'AISpecStopTriggerType': 'Tag observation',
                 'TagObservationTrigger': {
@@ -3946,6 +3976,9 @@ class LLRPROSpec(dict):
                     'Timeout': report_timeout_ms,  # milliseconds
                 },
             })
+            """
+            self['ROSpec']['ROReportSpec']['N'] = report_every_n_tags
```

For a general case, if one want a near realtime feedback, the best is probably to use the UP_TO_N_Milliseconds report trigger in my opinion. With something like 200ms.

thijmenketel, 2018-09-05:
Yes, direct clone from your repo. I ran it with:
```
sllurp inventory -s 2 -P 5 -M WISP5 --impinj-search-mode 2 --impinj-reports 192.168.0.100
```
Well that might make it nice and periodic but I think we should make it even more general and replace the -n option by something that combines the possible different triggers. This way it completes the possibilities as described in the LLRP. Something like this:

```
@click.option('--trig', nargs=2, type=int)
```
Where --trig takes 2 arguments, first for which trigger and second on what N it triggers. In the case of UP_TO_N_Milliseconds and 200 ms it would be --trig 5 200. Something like that?

fviard, 2018-09-05:
Hum, I have just re-read the llrp spec, and sadly UP_TO_{seconds,milliseconds} only came with the protocole v2 (llrp 1.1).
As far as I know, impinj does not support 1.1 yet also.

So the previous logic of using the AISpecStropTrigger looks valid in fact.
The tweak is probably to export report_timeout_ms through the cli, that would be more useful.

So I guess that maybe the cli can be changed to:
--trig-n MaxNumTags --trig-ms MaxTimeoutSec
That set the current "n" and the current "report_timeout_ms"
The 2 values in one argument would be more convenient, but no one expect something like that in a command line

If llrp1.1 would have been implemented on the 2 sides, what would have been great is something like:
--trig 5  (trigger after 5 tag events)
--trig 5s (trigger after 5s)
--trig 5ms (trigger after 5ms)

thijmenketel, 2018-09-05:
Ah yes you're right, I only checked 1.1.

So that means we have only three options for the trigger:
 - 0 = None
 - 1 = (Upon N TagReportData Parameters or End of AISpec) or (End of RFSurveySpec) - N=0 is unlimited
 - 2 = Upon N TagReportData Parameters or End of ROSpec - N=0 is unlimited.

>So the previous logic of using the AISpecStropTrigger looks valid in fact.
The tweak is probably to export report_timeout_ms through the cli, that would be more useful.

My opinion is that by using the AISpecStopTrigger (current implementation) there is a (too) big sacrifice in performance just to specify when a report is generated (in the case of n tags), so it needs to change. The report_timeout_ms is pretty much fine I think.

If we implement a --trig-n option I think its best to keep track of the tags seen on the client side to keep perfomance high. This would require keeping a dictionary or something like that for different EPC's.

fviard, 2018-09-05:
In my opinion, the --trig-n would be more like a security to avoid the reader buffer overflow.
typically, i would say report every 500ms, and also if we would exceed 500 tag seen.

Also, a subtle difference between ROReportTrigger N and the AIStopTrigger N is that the first one is kind of number of "report entries" (that is different epc, or same epc read on different antennas, ...), and the second one is number of "tag read" event (even same epc, conditions).

I think that the 3 things can be interesting independently to different persons.

fviard, 2018-09-05:
I don't understand, why you want to keep on the client side?
something like "if the rate is too fast, agglomerate on the client side to not refresh display tag list faster than a few times per second?"

fviard, 2018-09-05:
I don't know how sllurp is used in the wild, but i guess that current perf should be enough for people outputing the result directly. Someone wanting better car take inventory.py and do his own app based on that.
If it was me, i would put the read tag in a queue and then other another process doing my business feature on that.
My opinion is that the parsing perf is now quite good until the user callback.

Anyway, i need to put them in a nice form, but i have a few more change that should give a little more speed for the tagreport parsing.
This is empiric, but on my side, I think that i can manage to get from 0.0045s per report for handling the message to around 0.0022s.
The worst offender is a log of "debug" logging on the parsing code path.
Even if we are not in "--debug" mode, the logging module induce a little lag just to understand that it has not to output anything. And some times some operations are still executed like it was for the "func()".
(Ex.: hexlify). This is not big, but multiplied by 50k calls in a few dozen of seconds, that can make a difference.

fviard, 2018-09-05:
@ransford  I forgot to reply to you about that:

> Very cool; I'll take a look. Here is a short list of features I know are in use in production applications with the current Twisted implementation:
>
> Connecting to multiple readers at the same time [...]

Something that can be added to my branch is a facility to have "group" functions for multiple readers.
But otherwise, i think that it is an advantage that now each reader is independent in his space for being used in a "library" way.

Here the good thing is that each reader event loop work on his side without having a need or relation with any "main loop". You can use a reader object as is without having your client code to be modified to use twisted event loop.

thijmenketel, 2018-09-06:
> I don't understand, why you want to keep on the client side?
something like "if the rate is too fast, agglomerate on the client side to not refresh display tag list faster than a few times per second?"

Yes, something like that. Maybe in the future it might be interesting to make a simple gui for starting, pausing and stopping as well as keeping track of different tags and settings.

> Also, a subtle difference between ROReportTrigger N and the AIStopTrigger N is that the first one is kind of number of "report entries" (that is different epc, or same epc read on different antennas, ...), and the second one is number of "tag read" event (even same epc, conditions).

Yes, very true. But it's trivial to keep track of the tag read event so it doesn't make the AIStopTrigger essential for that purpose (as it does slow down reporting).

> Anyway, i need to put them in a nice form, but i have a few more change that should give a little more speed for the tagreport parsing.
This is empiric, but on my side, I think that i can manage to get from 0.0045s per report for handling the message to around 0.0022s.
The worst offender is a log of "debug" logging on the parsing code path.
Even if we are not in "--debug" mode, the logging module induce a little lag just to understand that it has not to output anything. And some times some operations are still executed like it was for the "func()".
(Ex.: hexlify). This is not big, but multiplied by 50k calls in a few dozen of seconds, that can make a difference.

Sounds very good!

fviard, 2018-09-06:
@thijmenketel Would be nice if you could give a try at the very last version of the untwisted branch.
I did push a few more changes to improve the parsing speed when you are not in "--debug" mode.
I stop doing optimizations now, I think that the parsing speed is now quite fast enough.
(My reader goes at around 1100 tags/s and the parsing looks negligible)

thijmenketel, 2018-09-07:
Cool! I'm not able to all weekend so on Monday I can run a test again.

fviard, 2018-09-10:
FYI, I have noticed that with the fastest settings, ETSI impinj gives around half the speed of FCC ones. (like 500/600 vs 1100).
But anyway, this just help to validate that the parsing is not anymore the limiting factor but the case is particular we try to read the fastest the same few tags. That I think is the case of any real use.

thijmenketel, 2018-09-10:
> FYI, I have noticed that with the fastest settings, ETSI impinj gives around half the speed of FCC ones. (like 500/600 vs 1100).

My reader is running in FCC mode so that isn't the difference. My tags are just quite shitty so I think the bottleneck is there.

> But anyway, this just help to validate that the parsing is not anymore the limiting factor but the case is particular we try to read the fastest the same few tags. That I think is the case of any real use.

Yes I think so as well.

I think this untwisted patch will be a nice addition to sllurp, are you going to make a PR?

fviard, 2018-09-10:
I'm thinking of adding just a few little things more before, but very soon I think.
But regarding the huge changes, if it would be acceptable, probably should be a new "major" number version because would break the api of people that could have used as a library.

The only small feature that is not "ported" is the "auto resume timer" argument of the resume function.
I don't know if it was ever used. Its easy to do something like that with a timer/ thread timer. But with this its easy to have rogue timer that could fire at a later time when you don't expect it and you are doing other operations. I think it was already a possible issue before, but letting something like without doing it properly annoys me.

fviard, 2018-09-14:
Do you have applied the following patch:
(I have not pushed any commit with that yet as this has to be done in a clean way)

> Did you use my version directly?
> If just with -n 1, you can try the following patch in llrp_proto.py just to see best perfs:
>
> ```
> @@ -3936,6 +3965,7 @@ class LLRPROSpec(dict): [...]

And i'm testing with this setup:
- around 8 tags very near of the antenna.
The following command:
`sllurp inventory -a 1 -X 30 -s 0 --mode-identifier 0 -P 2 192.168.1.2 --impinj-search-mode 2 --impinj-reports -n 1`

fviard, 2018-09-14:
But you did also comment out the "AISpecStopTrigger" so it will not restart the aiespec at each tag?

The ETSI reader that I have has the fw 5.12 (but is also saturating around 600 tags/s)
I don't have the FCC reader available now to check the version, but I think that it was an older, like 4.x.
I will have a look at it later, and also try to update it to ensure that firmware has no impact on performances.

thijmenketel, 2018-09-14:
Tried that, but also I don't use the -n option so the AISpecStopTrigger doesn't even get set.

I did found some firmware through the atlasrfidstore, but as it's our only reader I'm careful about upgrading the firmware haha. Dont want to brick it.

fviard, 2018-09-14:
You can find all the firmware there (even older ones):
https://support.impinj.com/hc/en-us/articles/202473963
I think that you can register for free on their website.

Their last update is : 5.12.2 
I think that for updating it is easy and safe, go to the webinterface and upload the new firmware.
And they have a double bank system, so normally, that should not break.

What is your current version? You can see that from the web interface too.

thijmenketel, 2018-09-14:
Ah yes Thanks! I read something about those banks but didn’t really understand how it worked, but sounds like a good fail safe.

The current octane version is 4.8.x something (can check on Monday again). I’ll do the update first thing on Monday and check the results. I am using better tags now (aln-9840) which are faster.

fviard, 2018-09-14:
I did just check my version.
I was in 5.2.1 and updated to 5.12.3, and so far no difference in perf.

Fyi, the double bank update logic is not complicated:
The reader system storage drive has 2 partiions for the system. It is like copies.
Only one, the current one, is used for the currently running firmware.
When you do an update, the "system image" (so the new filesystem) is entirely written to the second partition that is unused.
Then, a flag is set to say: next time start on the other bank system.
At the first boot on the other bank, if the boot is a success, the flag is permanently marked to use the second partition for the "running" system.
At the opposite, if there is an error trying to boot, the flag is reset and the second partition is discarded.
So next reboots will still point to the first "disk" system partition that was untouched.

And the next time you will do an update, it was target the newly unused partition. Swapping the active one at each update. And so on...

thijmenketel, 2018-09-17:
Just did the upgrade from 4.8.x to 5.12.3, what a difference! went from 700 tags/s to almost 1000 tags/s (with 8 aln-9840 tags). But what really blows me away is single tag read performance: I get about 500 tag reads per second on this.

fviard, 2018-09-17:
Awesome. By curiosity, could you run the sllurp origin MASTER to compare its speed with this version for you?

thijmenketel, 2018-09-17:
Okay I tested the origin MASTER sllurp with:
```
sllurp inventory -n 1 -s 0 -P 1 -M WISP5 --mode-identifier 0 --impinj-search-mode 2 --impinj-reports 192.168.0.100
```

and the untwisted with:
```
sllurp inventory -s 0 -P 1 -M WISP5 --mode-identifier 0 --impinj-search-mode 2 --impinj-reports 192.168.0.100
```
But with hardcoded ROReportSpec N set to 1.

Impinj Speedway R420 reader with firmware verison 5.12.3 and single ALIEN ALN-9840 (Higgs-EC) tag.

 - origin MASTER sllurp: 44 tag reads per second
 - untwisted: 551 tag reads per second

fviard, 2018-09-17:
Great :-)

thijmenketel, 2018-10-01:
@fviard quick note: sometimes the untwisted sllurp version doesn't react to keyboard interrupt and it requires me to press it again. Not a big issue but just a note!

I've been using untwisted sllurp for the past month and it works perfectly!

fviard, 2018-10-01:
Thanks, great to know.
There are a few more things I want to do, but I'm currently focus on another urgent task before being able to go back on it.

For the keyboard interrupt, what happen is that the first time you interrupt the main script, and next time the "reader" main loop.
Did you wait a few seconds before the second ctrl+c?
What happens is that when you do ctrl+c, we do the shutdown procedure with the reader and wait for acknowledgement for the reader. If there was a lot of tag events/events being streaming, it can take a few seconds for the reader to send everything to empty its queue.

If not, it would be interesting next ime that you wait a little after the first keyboard interrupt to see if it will stop. Otherwise, i would be interested to know the context when you did it. Like was it at the beginning of a run, a end, ...
For my own testings, it was never blocked forever assuming that the reader was not dead.

djacobson, 2018-11-06:
I just want to commend you guys on your work on this fork. Really impressive tag read performance stats.

Just wondering out loud if the untwisted version couldn't be refactored into the existing module, with twisted.io re-org'd into and option import? Do that many higher-level api functions become stubbed / unused (i.e.: thinking of the client_factory model)? Users of the untwisted ""mode" might not care if some twisted supporting function become irrelevant? Just a thought...

Again, really intriguing work.

ransford, 2018-11-07:
Agreed, the stats are impressive.  Excellent work!

I'm not married to Twisted at all.  Removing that dependency is just fine with me.  We could bump the version number to 1.0 with such a big change!

What work remains to completely replace Twisted, and what help do you need @fviard?  I am having trouble keeping track of the "roadmap" for this work.

fviard, 2018-11-07:
Normally it should be almost good, sadly i was preempted on another unrelated subject for some time preventing me to complete it yet.
What I see as remaining and not yet started if you want to have a look at it:
- Related to twisted:
Fix the "tornado" example. Should be easy by using the basic ioloop of tornado and fixing the sllurp glue with the new api.
- Small rework of trigger/N settings to really differentiate report trigger and stop triggers.
- Backport your few recent changes to be in sync again before merge.

djacobson, 2018-11-07:
On your first point, I might also suggest a really simple "un-tornado'ized" example that simply outputs tag reports to the console. Would really focus on sllurp alone. That's basically what I based my sllurp app off of; the tornado demo with tornado dependencies removed. But, then again, tornado is a nice demo way to update a web page.

fviard, 2018-11-07:
@djacobson I don't know if you noticed, but that is provided by sllurp by default as a cli.
Example:
`sllurp inventory -a 4 -X 1 -s 2 --mode-identifier 2 -P 200 --impinj-search-mode 1 --impinj-reports -n 1 192.168.0.100`
"-n 1" will result in a direct output for each tag read event instead of dumping all the read tags when you "ctrl+c" interrupt it.

There is also the "log" command that can create a csv file as output.

djacobson, 2018-11-07:
@fviard Ah, thank you. I still needed the API approach for my app specific algorithm of storing and sorting tags based on RSSI value and some other logic, and integration into a custom UI (PyQT), but, this is something I should have explored further.

fviard, 2018-11-07:
Have a look at the following folder:
`sllurp/verb`
inventory and log "cli applications" are self contained in a single py file.
I think it should be easy to read and copy to use in your application.

The good thing now is that the reader has its self managed own thread, and so you don't have to worry about integrating with whatever io main loop is used by PyQT.

fviard, 2019-01-18:
@thijmenketel Do you plan of open sourcing your GUI? maybe as an example would be nice.
The lazy me would appreciate to use a GUI sometimes instead of always the command line.

fviard, 2019-01-18:
@ransford I have fixed the tornado example and merged all the recent commits of develop.
(A rebase was not possible...)
In my opinion, things are now good enough so I have created a PR with my "untwisted" branch.

There are still things that could be improved, but I think that it is unrelated to this work, and so it can be done later.

thijmenketel, 2019-01-20:
@fviard I might open source it, however I will first try to make my spaghetti bunch of a code that I copied from your repo and edited to be a little more presentable!

iadv, 2019-03-15:
@fviard @thijmenketel Thank you for this work and detailed comments. Really very helpful. I am student and trying to use sllurp to read tags using R420 readers. I am struggling to implement this modifications. I ran it as follows:
1. cloned it from here develop_untwisted
2. Followed these commands: 
$ virtualenv .venv
$ source .venv/bin/activate
$ pip install .
$ sllurp inventory ip.add.re.ss
3. The running with sllurp inventory -s 0 --mode-identifier 0 --impinj-search-mode 2 --impinj-reports xx.xx.xx.xxx

However, couple of things I don't understand are - 
1. Even if I delete the inventory file in the sllurp>verb>inventory.py, the command is working? So is the filed compiled already or am I using my old sllurp files. I am not sure how to differentiate between original sllurp and how to use this modification
2. Tag readrates I am getting are 20 tags/sec. 
3. which file is called when I use this command "sllurp inventory ...."
I am just a beginner with coding, so please pardon my lack of using correct words.
Thanks for the help

thijmenketel, 2019-03-18:
By using the commands you create a virtual environment (sort of sandbox) where Sllurp (the cloned repo version) is installed as a pip package. This means that inventory will keep working even if you delete it. I would advice you to create your own program by looking at the inventory.py file, this way you can get more familiar with the code. I'm also student working with this stuff, so I would advice you to look at the protocol itself and at the llrp.py and llrp_proto.py files to understand what is happening.

Short answer on the low read rates, can be the following:

 - Old firmware: my reader was considerably faster when I upgraded.
 - Bad tags: I'm using top of the line Alien and Impinj tags which are very sensitive and quite fast
 - Look at [this](https://github.com/ransford/sllurp/blob/351abb3726ace12624f40e6c2f019cf751fa2446/sllurp/llrp_proto.py#L4061) line, if there's a 0 there, change it to a 1 and rerun the commands for the venv (first removing the .venv/ dir ofcourse)

Hope this helps

iadv, 2019-03-18:
That's a nice explanation. However, when I run the code, I am getting the following error:
(.venv) idav-3:sllurp idav$ sllurp reset xx.xx.xx.xxx

raceback (most recent call last):
  File "/Users/idav/sllurp/.venv/bin/sllurp", line 6, in <module>
    from sllurp.cli import cli
  File "/Users/idav/sllurp/.venv/lib/python2.7/site-packages/sllurp/cli.py", line 11, in <module>
    from .verb import inventory as _inventory
  File "/Users/idav/sllurp/.venv/lib/python2.7/site-packages/sllurp/verb/inventory.py", line 11, in <module>
    from sllurp.llrp_proto import Modulation_DefaultTari
ImportError: cannot import name Modulation_DefaultTari

Did you face any issues with Modulation_DefaultTari?

iadv, 2019-03-18:
@thijmenketel 

Quick note on updating the firmware.. I had 4.0.0 running and I upgraded to the latest 5.14. Actually, my read rate went down from 44 tags/sec to 6 tags/sec. I don't have the old firmware file, so I kinda worried now. I had contacted the impinj support. Waiting for their response...

fviard, 2019-03-18:
@iadv Inside your virtualenv, instead of pip install ., you can do "python3 setup.py install" after each modification that you do to the source code.
This will kind of "install" the script and libraries in the right place. 
Regarding the your speed issue, I think that you should read the pages on Impinj website and their documentation to understand better the different modes and what they imply.

https://support.impinj.com/hc/en-us/articles/202756158-Understanding-EPC-Gen2-Search-Modes-and-Sessions

Reported speed will also depend on the number of tags that you have in the field.

iadv, 2019-03-21:
@fviard @thijmenketel Thanks for your help. I was able to make it run and even output the impinj extension features. I modified the inventory.py so that it can run by itself using the python inventory.py host command. I needed this because it is easy to modify. However, I do not understand the RSSI, phase and doppler values it outputs. RSSI I think it skips the decimal and outputs, for example, -6540 instead of 65.40. I am confused with the phase values. What does the 2000 phase mean? shouldn't phase be bounded from 0 to 180?

[2 more replies]

closed 2021-01-24 by ransford