sllurp/sllurp #107: Using sllurp to change a tag's EPC

PHarman4, 2018-08-31:
I have been using sllurp to operate an Impinj Multi-reader R420 and recently received a shipment of Confidex silverline Micro RFID labels. Trouble is, all the tags have the same exact EPC so the webpage doesn't write a new entry when I scan different tags. 

I have played with the tag_seen_callback in Main but I haven't been successful in changing a tag's EPC yet. Is there a way to do this? I know it can be done using Impinj's software but I am cheap and would prefer a solution that does not include purchasing their software

ransford, 2018-09-02:
Sounds like you need to use access commands to write to the EPC memory bank of each tag.  See `sllurp access --help`.  My guess is that you want to use relatively low power and make sure there's only one tag in front of the reader when you're attempting to write an EPC, or else you might end up with the same problem (duplicate EPCs).

Please share what you find!

PHarman4, 2018-09-27:
No I did not. Following your suggestion, I checked out `sllurp access --help ` and found options within sllurp access for a write-words command as well as different memory bank options to where these words could be stored within the tag, but I was not able to successfully execute those commands. I tried to write words (or really integers) to the user memory bank, the EPC, and the TID and none of which were successful. For example, from Linux command line: `sllurp access -w 1122 -mb 1 ` 

-w is the flag for write-words, -mb is the flag for memory bank and the options for memory bank are 3 for user, 2 for TID, and 1 for EPC. 

After trying these commands I would notice no changes when I went back and scanned a tag again no matter where in the tag memory I tried to store the new EPC

PHarman4, 2018-09-27:
@ransford If you have any suggestions as to what I could do to make it work, I am more than open to it. Otherwise I was thinking about ordering new tags and requesting they came with different EPC's. 

It is very possible I am missing something obvious here

ransford, 2019-02-13:
I ran out of time to work on this, sorry. If you managed to make any progress and are able to share your results, please reopen this issue or consider a pull request. Thank you!

closed 2019-02-13 by ransford

ukrutt, 2019-10-01:
FWIW, I've been able to write to a tag's EPC.  I'm on current `develop` (a6079d1).  The trickiest part was to (a) figure out how to specify the actual data I wanted to write (it is read from stdin) and (b) convert it from "hexl" form to binary (I used a tiny Python script that I specified on the command line).  Below is what I did; it writes `0xCAFE` to the "last visible" word of the EPC-96 of a SMARTRAC FROG 3D tag (based on Impinj Monza 4D).  Note that this tag has room for 128 bit EPC, thus the offset/word pointer is set to `-wp 7` -- not `5`.
```
echo cafe | python3 -c "import sys, binascii; sys.stdout.buffer.write(binascii.unhexlify(input().strip()))" | sllurp access tag-reader-name.local -mb 1 -wp 7 -w 1 -X 51 -c 1
```

ukrutt, 2019-10-01:
@PHarman4 -- you write:

>  [...] For example, from Linux command line: `sllurp access -w 1122 -mb 1 `

The `-w` flag is in fact indicating "how many 16 bit words to write", not the actual word/data itself.  The data is read from STDIN (as binary.)  My previous comment show the hoops I jumped through.