sllurp/sllurp #108: Launch access command to only a subset of Tag

rouxb, 2018-09-04:
I have been using sllurp API to control an impinj speedway revolution reader.
Based on the available examples, I was able to write a custom script that start inventory and, for each seen tags, send a list of callbacks such as read/write access commands and return the the desired data to the user.

Now I wanna isolate a particular tag (through a filter on its EPC value for example) and send a list of access command only to the matching tag.
At the first glance I think that I could achieve this behavior with the help of the *target* argument pass to *LRPClient.startAccess()* function.
I have try to execute the access callback with a subset of tag that have EPC number LSB byte matching 0x01. For this purpose I have filled the target dictionary as follow:

 ``` python
 target = {
           'MB': 1,
           'Pointer': 0,
           'MaskBitCount': 8,
           'TagMask': b'01',
           'DataBitCount': 0,
           'TagData': b''
           }
```

However during the execution sllurp return the following error:
``` xml
DEBUG:sllurp.llrp:LLRPMessage received in state 18: <ErrorMessage>
    <Type>100</Type>
    <LLRPStatus>
        <StatusCode>u'FieldError'</StatusCode>
        <ErrorDescription>'LLRP decode failure at byte offset 51 (status -217)'</ErrorDescription>
    </LLRPStatus>
</ErrorMessage>
```

I think that I misunderstood the configuration of *target* dictionary. Maybe you could give me some clues to solve my issue.
Thank you

rouxb, 2018-09-24:
Hello @ransford,
No, I have not figure this out.
For the moment I have circumvent this issue by using the java framework furnished by Impinj.
By the way, I'am always interested in solving this in Sllurp, because it will be easier to integrate a python app in the global automated test framework that I have to build.

Thank you for your support.

ransford, 2018-11-19:
Try the latest `develop` now that #113 is merged (thanks @thijmenketel).  Better?

rouxb, 2018-11-19:
Great jobs, it is perfect.
Tanks you for your support.

closed 2018-11-19 by rouxb

nickyandre, 2024-03-25:
> Great jobs, it is perfect.
> Tanks you for your support.

Hello i saw that you have succeed implementing filtering to launch access command to only subset of tag can you guide me how to do it in the custom code?. I hope i will get your response soon thank you very much.

thijmenketel, 2024-03-29:
Which custom code do you mean? I'm not I understand what you mean.

nickyandre, 2024-04-11:
> Which custom code do you mean? I'm not I understand what you mean.

Hello i am sorry for late reply in this case i've developed a code to do inventory and read a memory from the chip continously then log the value into a csv file please find below the code. I am using Impinj Speedway Revolution R420 UHF RFID reader. I have two question how to implement filter in this custom code and only do inventory and read memory only into this specific tag?. Second, how i can increase the read rate of the UHF RFID reader? What kind of parameters i can fix to increase the read rate?. Any suggestion to my code is appreciated 

`from sllurp import llrp
from twisted.internet import reactor
import logging
import csv
logging.getLogger().setLevel(logging.INFO)

READER_IP = '169.254.109.180'
#SCAN_TIME = 20

with open("RFIDlog.csv", "w", newline='') as file:
    writer = csv.writer(file)
    writer.writerow(['EPC', 'Memory'])
    
def access(proto):
    readSpecParam = None
    readSpecParam = {
            'OpSpecID': 0,
            'MB': 3, #3:user, 2:TID, 1:EPC
            'WordPtr': 261, #wordaddress to read
            'AccessPassword': 0,
            'WordCount': 1
    }
    return proto.startAccess(readWords=readSpecParam)

def tagreportcb(llrp_msg):
    global tagReport
    tags=llrp_msg.msgdict['RO_ACCESS_REPORT']['TagReportData']
    tag_list = set()
    for tag in tags:
#         tagReport += tag['TagSeenCount'][0]
        if "OpSpecResult" in tag:
            # copy the binary data to the standard output stream
            data = tag["OpSpecResult"].get("ReadData")
            datafixed = data.hex()                                                                                                                                                                                                     
            tag_list.add(datafixed)
            tag_list.add(tag.get('EPC-96',None))
            
    return tag_list

def report(llrp_msg):
    tag_list = tagreportcb(llrp_msg)
    tag_list = list(tag_list)
    with open("RFIDlog.csv", "a", newline='') as file:
        epc=None
        memory=None
        for item in tag_list:
            if len(item) > 23:
                epc = item
            else:
                memory = item
                if epc is not None:
                    writer = csv.writer(file)
                    writer.writerow([epc,memory])
                    print(f"EPC: {epc}, Memory: {memory}")
                
def shutdown(factory):
    return factory.politeShutdown()

def main():
    factory = llrp.LLRPClientFactory(antennas=[0], duration=0.5)
    factory.addTagReportCallback(report)
    factory.addStateCallback(llrp.LLRPClient.STATE_INVENTORYING, access)
    reactor.connectTCP(READER_IP, llrp.LLRP_PORT, factory, timeout=3)
    #reactor.callLater(SCAN_TIME, shutdown, factory)
    # catch ctrl-C and stop inventory before disconnecting
    reactor.addSystemEventTrigger('before', 'shutdown', shutdown, factory)
    reactor.run()

if __name__ == "__main__":
    main()`