sllurp/sllurp #109: Did anyone read phase and doppler frequency using the impinj extension in this code?

iadv, 2018-09-04:
I am not able to understand how to print the doppler frequency and the phase values. ['RO_ACCESS_REPORT']['TagReportData'] does not have these values and couldn't find any other dictionary where phase is stored. Is it an incomplete code or am I missing something?

iadv, 2018-09-04:
Also, I am passing impinj_search_mode = 1 as the help doesn't mention what arguments to pass neither Impinj's website. Default value is None

thijmenketel, 2018-09-05:
Try to pass the search mode value 2 (dual target, see [here](https://support.impinj.com/hc/en-us/articles/202756158-Understanding-EPC-Gen2-Search-Modes-and-Sessions)).

Furthermore, to get the extensions added also pass the option --impinj-reports.

Note on those values: due to the nature of the RFID standard, the reader will hop randomly through the channels. This hopping induces a phase shift so keep that in mind.

thijmenketel, 2018-09-11:
@iadv Any success? If so, please close the issue ;)

ransford, 2018-09-23:
@iadv assuming you got this to work; reopen if still having trouble!

closed 2018-09-23 by ransford