sllurp/sllurp #115: No Tag Reads from Zebra FX-9600

AaronHuus, 2019-01-02:
When connecting to the Zebra FX-9600 and registering a TagReportCallback, I am not seeing any tags be detected.  I create the factory as follows:

```
    factory: LLRPClientFactory = LLRPClientFactory(
        antennas=[1],
        tx_power=0,
        start_inventory=True
    )
    factory.addTagReportCallback(tag_callback)
    reactor.connectTCP(READER_IP_ADDRESS, LLRP_PORT, factory)
    reactor.run()
```

And have defined the callback as:

```
def tag_callback(tag_report):
    tags = tag_report.msgdict['RO_ACCESS_REPORT']['TagReportData']
    print(f'tags: {tags}')
```

All that is printed to the console is `tags: []` twice.  And then nothing after that even as I move a tag in and out of range.  When using the readers web interface and sample 'read' application it is able to see the tags.  Any suggestions would be appreciated, thanks!

ransford, 2019-01-02:
Hello! Could you post a debug-level log (`sllurp -d -l inventory.log 192.168.0.234`)? I'd like to dissect the protocol messages going back and forth.

AaronHuus, 2019-01-02:
```
connecting to 192.168.0.234:5084...
Building protocol for 192.168.0.234:5084
Antennas in buildProtocol: [1]
192.168.0.234:5084 start_inventory: True
will reset reader state on connect
will start inventory on connect
using antennas: [1]
transmit power: {1: 0}
connected to ('192.168.0.234', 5084) (192.168.0.234:5084)
got 32 bytes from reader: b'043f000000200000001b00f600160080000c00057e7755c77a2f010000060000'
expect 32 bytes (have 32)
deserializing READER_EVENT_NOTIFICATION command
decode_ReaderEventNotification
decode_ReaderEventNotificationData (type=246 len=22)
decode_UTCTimestamp
decode_UTCTimestamp (type=128 len=12)
decode_ConnectionAttemptEvent
decode_ConnectionAttemptEvent (type=256 len=6)
decode_AntennaEvent
done deserializing READER_EVENT_NOTIFICATION command
LLRPMessage received in state 1: <READER_EVENT_NOTIFICATION>
	<Ver>1</Ver>
	<Type>63</Type>
	<ID>27</ID>
	<ReaderEventNotificationData>
		<ConnectionAttemptEvent>
			<Status>'Success'</Status>
		</ConnectionAttemptEvent>
	</ReaderEventNotificationData>
</READER_EVENT_NOTIFICATION>
starting message callbacks for READER_EVENT_NOTIFICATION
done with message callbacks for READER_EVENT_NOTIFICATION
in handleMessage(READER_EVENT_NOTIFICATION), there are 0 Deferreds
serializing GET_READER_CAPABILITIES command
serialized bytes: b'04010000000b0000000000'
done serializing GET_READER_CAPABILITIES command
state change: STATE_DISCONNECTED -> STATE_SENT_GET_CAPABILITIES
got 10 bytes from reader: b'043e0000000a0000001c'
expect 10 bytes (have 10)
deserializing KEEPALIVE command
done deserializing KEEPALIVE command
LLRPMessage received in state 21: <KEEPALIVE>
	<Ver>1</Ver>
	<Type>62</Type>
	<ID>28</ID>
</KEEPALIVE>
starting message callbacks for KEEPALIVE
done with message callbacks for KEEPALIVE
serializing KEEPALIVE_ACK command
serialized bytes: b'04480000000a00000000'
done serializing KEEPALIVE_ACK command
got 1228 bytes from reader: b'040b00000ce000000000011f0008000000000089007200084000000000a1000177080008322e372e31392e30008b000800000000008d000800040004008c00090001000101008c00090002000101008c00090003000101008c00090004000101008c00090005000101008c00090006000101008c00090007000101008c00090008000101008e001cd80700000000004000000010000000010000000400000008008f0bad0000000100900ba500910008000003e800910008000103f200910008000203fc00910008000304060091000800040410009100080005041a0091000800060424009100080007042e0091000800080438009100080009044200910008000a044c00910008000b045600910008000c046000910008000d046a00910008000e047400910008000f047e00910008001004880091000800110492009100080012049c00910008001304a600910008001404b000910008001504ba00910008001604c400910008001704ce00910008001804d800910008001904e200910008001a04ec00910008001b04f600910008001c050000910008001d050a00910008001e051400910008001f051e00910008002005280091000800210532009100080022053c00910008002305460091000800240550009100080025055a0091000800260564009100080027056e0091000800280578009100080029058200910008002a058c00910008002b059600910008002c05a000910008002d05aa00910008002e05b400910008002f05be00910008003005c800910008003105d200910008003205dc00910008003305e600910008003405f000910008003505fa0091000800360604009100080037060e0091000800380618009100080039062200910008003a062c00910008003b063600910008003c064000910008003d064a00910008003e065400910008003f065e00910008004006680091000800410672009100080042067c00910008004306860091000800440690009100080045069a00910008004606a400910008004706ae00910008004806b800910008004906c200910008004a06cc00910008004b06d600910008004c06e000910008004d06ea00910008004e06f400910008004f06fe00910008005007080091000800510712009100080052071c00910008005307260091000800540730009100080055073a0091000800560744009100080057074e0091000800580758009100080059076200910008005a076c00910008005b077600910008005c078000910008005d078a00910008005e079400910008005f079e00910008006007a800910008006107b200910008006207bc00910008006307c600910008006407d000910008006507da00910008006607e400910008006707ee00910008006807f8009100080069080200910008006a080c00910008006b081600910008006c082000910008006d082a00910008006e083400910008006f083e00910008007008480091000800710852009100080072085c00910008007308660091000800740870009100080075087a0091000800760884009100080077088e009100080078089800910008007908a200910008007a08ac00910008007b08b600910008007c08c000910008007d08ca00910008007e08d400910008007f08de00910008008008e800910008008108f200910008008208fc0091000800830906'
expect 3296 bytes (have 1228)
got 1228 bytes from reader: b'0091000800840910009100080085091a0091000800860924009100080087092e0091000800880938009100080089094200910008008a094c00910008008b095600910008008c096000910008008d096a00910008008e097400910008008f097e00910008009009880091000800910992009100080092099c00910008009309a600910008009409b000910008009509ba00910008009609c400910008009709ce00910008009809d800910008009909e200910008009a09ec00910008009b09f600910008009c0a0000910008009d0a0a00910008009e0a1400910008009f0a1e0091000800a00a280091000800a10a320091000800a20a3c0091000800a30a460091000800a40a500091000800a50a5a0091000800a60a640091000800a70a6e0091000800a80a780091000800a90a820091000800aa0a8c0091000800ab0a960091000800ac0aa00091000800ad0aaa0091000800ae0ab40091000800af0abe0091000800b00ac80091000800b10ad20091000800b20adc0091000800b30ae60091000800b40af00091000800b50afa0091000800b60b040091000800b70b0e0091000800b80b180091000800b90b220091000800ba0b2c0091000800bb0b360091000800bc0b400091000800bd0b4a0091000800be0b540091000800bf0b5e0091000800c00b680091000800c10b720091000800c20b7c0091000800c30b860091000800c40b900091000800c50b9a0091 [... cut ...]
done deserializing READER_EVENT_NOTIFICATION command
LLRPMessage received in state 18: <READER_EVENT_NOTIFICATION>
	<Ver>1</Ver>
	<Type>63</Type>
	<ID>100</ID>
	<ReaderEventNotificationData>
	</ReaderEventNotificationData>
</READER_EVENT_NOTIFICATION>
starting message callbacks for READER_EVENT_NOTIFICATION
done with message callbacks for READER_EVENT_NOTIFICATION
Got reader event notification
lost connection: Connection to the other side was lost in a non-clean fashion: Connection lost.

Process finished with exit code 0
```

AaronHuus, 2019-01-02:
[Here](https://www.zebra.com/content/dam/zebra_new_ia/en-us/manuals/rfid/rfid%20software%20interface-cg-en.pdf) is some documentation on the Zebra FX9600 LLRP interface.

ransford, 2019-01-02:
In this log, the reader sends exactly one tag report, and it's empty:

```
got 10 bytes from reader: b'043d0000000a00000000'
expect 10 bytes (have 10)
deserializing RO_ACCESS_REPORT command
decode_ROAccessReport
decode_TagReportData
done deserializing RO_ACCESS_REPORT command
LLRPMessage received in state 5: <RO_ACCESS_REPORT>
	<Ver>1</Ver>
	<Type>61</Type>
	<ID>0</ID>
	<TagReportData>[]</TagReportData>
</RO_ACCESS_REPORT>
starting message callbacks for RO_ACCESS_REPORT
done with message callbacks for RO_ACCESS_REPORT
tags: []
```

Here's the part that looks interesting, though:
```
state change: STATE_SENT_ENABLE_ROSPEC -> STATE_INVENTORYING
got 39 bytes from reader: b'043f000000270000006400f6001d0080000c00057e7755dae23b00f9000d000000000100000000'
expect 39 bytes (have 39)
deserializing READER_EVENT_NOTIFICATION command
decode_ReaderEventNotification
decode_ReaderEventNotificationData (type=246 len=29)
decode_UTCTimestamp
decode_UTCTimestamp (type=128 len=12)
decode_ConnectionAttemptEvent
decode_AntennaEvent
Unprocessed bytes in READER_EVENT_NOTIFICATION: b'00f9000d000000000100000000'
done deserializing READER_EVENT_NOTIFICATION command
LLRPMessage received in state 18: <READER_EVENT_NOTIFICATION>
	<Ver>1</Ver>
	<Type>63</Type>
	<ID>100</ID>
	<ReaderEventNotificationData>
	</ReaderEventNotificationData>
</READER_EVENT_NOTIFICATION>
```

The "Unprocessed bytes" are what I'll take a look at next. By eye I'm not sure what the reader is telling us!

AaronHuus, 2019-01-02:
I may have made some progress, i set `session=0` and things started to work more like I was expecting.

AaronHuus, 2019-01-03:
After reading a little more, i don't think that is what we wanted.  I am looking into how to implement something similar to `impinj-search-mode` but for the Zebra/Motorola RFID reader.

ransford, 2019-01-03:
I manually dissected the "Unprocessed bytes" from your log (using the [LLRP spec](docs/llrp_1_1-standard-20101013.pdf)) and figured out that your reader was saying the ROSpec (essentially, the highest-level set of instructions you've given the reader) was preempted by another ROSpec. Is there another read activity running, say, via the reader's web interface? What if you `sllurp reset` your reader before trying `sllurp inventory`? Or this preemption may be a red herring. I'm not sure how to interpret that that report.

```
043f00000027   -- ReaderEventNotification of length 39 (including this header)
00000064       -- Message ID = 100
  00f6001d     -- ... ReaderEventNotificationData param of length 29 (including this header)
    0080000c   -- UTCTimestamp param of length 12 (including this header)
      00057e7755dae23b -- Microseconds = 1546425890169403
    00f9000d   -- ROSpecEvent param of length 13 (including this header)
      00       -- ROSpec event type = 0
      00000001 -- ROSpec ID = 1
      00000000 -- Preempting ROSpec ID = 0
```

Happy to offer pointers if you're implementing new functionality, but it sounds like you're already on the right track following the Impinj example. Pull requests supporting more reader families are always welcome!

ransford, 2019-01-03:
Oh no, ignore my interpretation of the "preemption" message. According to section 14.2.6.4 of the aforementioned LLRP spec, ROSpec event type = 0 means that the ROSpec given by "ROSpec ID" _started_, not that it was preempted. So that message is benign.

AaronHuus, 2019-01-03:
Thank you for looking into that! So I have it sort of doing what i want, but i need to set `report_every_n_tags=1`. This gets it close, but really what I want is a list of the tags in the event repot, not 1 tag per report.  Do you know of a way to get back every tag that was seen during each inventory scan? It may be 1 tag, it may be 0, it may be N tags.

ransford, 2019-01-03:
Try `sllurp inventory -t 1` to get a report every 1 second. How to get a report every N seconds is left as an exercise for the reader. :-)

(That part of the sllurp codebase isn't great, as there are a zillion moving parts controlling timeouts and such. Pull requests or financial support welcome if you want to use that option in production.)

ransford, 2019-02-13:
Closing this issue because I'm hopeful the last update solved the problem. Please reopen it if we didn't quite get there, @AaronHuus!

closed 2019-02-13 by ransford

pBogey, 2019-09-06:
Hello,

I am trying to write an app for the Zebra FX9500 reader and I am having the same issue as Aaron had: tag list returns empty on the first run. 

Tried the example code from the Readme file:
`# Minimal example; see inventory.py for more.
from sllurp import llrp
from twisted.internet import reactor
import logging
logging.getLogger().setLevel(logging.INFO)

def cb (tagReport):
    tags = tagReport.msgdict['RO_ACCESS_REPORT']['TagReportData']
    print 'tags:', tags

factory = llrp.LLRPClientFactory()
factory.addTagReportCallback(cb)
reactor.connectTCP('169.254.1.1', llrp.LLRP_PORT, factory)
reactor.run()`

With the code above, the result is "tags = []" and the reader keeps reading, but the program does not return anything else. I've checked out inventory.py and tried to understand how it works. I have the modified the example code to the following state:

`# Minimal example; see inventory.py for more.
from sllurp import llrp
from twisted.internet import reactor, defer
import logging
import time
import pprint
logging.getLogger().setLevel(logging.INFO)
logger = logging.getLogger(__name__)

READER_IP_ADDRESS = '169.254.1.1'
numtags = 0

def finish(*args):
    print("*"*20," Running Inventory.finish","*"*20)
    runtime = monotonic() - start_time
    logger.info('total # of tags seen: %d (%d tags/second)', numtags,
                numtags/runtime)
    if reactor.running:
        reactor.stop()

def shutdown(factory):
    print("*"*40," Running Inventory.shutdown","*"*40)
    return factory.politeShutdown()

d = defer.Deferred()
d.addCallback(finish)
factory_args = dict(
    onFinish=d,
    )

def cb(llrp_msg):
    global numtags
    print("*"*20," Running cb","*"*20)
    tags = llrp_msg.msgdict['RO_ACCESS_REPORT']['TagReportData']
    print ('tags:', tags)
    if len(tags):
        logger.info('saw tag(s): %s', pprint.pformat(tags))
        for tag in tags:
            numtags += tag['TagSeenCount'][0]
    else:
        logger.info('no tags seen')
        print("-"*10," Exit cb ", "-"*10)
        return

factory = llrp.LLRPClientFactory()
factory.addTagReportCallback(cb)
reactor.connectTCP(READER_IP_ADDRESS, llrp.LLRP_PORT, factory)
reactor.addSystemEventTrigger('before', 'shutdown', shutdown, factory)
[... 1622 lines cut ...]
2019-09-06 14:05:43,752 sllurp.llrp_proto: DEBUG: decode_param pardata: b'\x00\x03\x00\xdf\x00\x06\x00\x01\x00\xe0\x00\n\x00\x01\x00\x01\x00\xc8\x01J\x00\x1d\x00\x01O\x00\x08\x00\x040\xd4\x01P\x00\x10@\x00\x14\x00\x00\x00\x00\x01Q\x00\x05\x00'
2019-09-06 14:05:43,754 sllurp.llrp_proto: DEBUG: decode_param data: b"\x00\xde\x003\x00\x04\x00\xdf\x00\x06\x00\x01\x00\xe0\x00\n\x00\x01\x00\x01\x00\xc8\x01J\x00\x1d\x00\x01O\x00\x08\x00\x040\xd4\x01P\x00\x10@\x00\x14\x00\x00\x00\x00\x01Q\x00\x05\x00\x00\xf4\x00C\x00\xf5\x00\x07\x00\x01\x00\x00\xf5\x00\x07\x00\x02\x00\x00\xf5\x00\x07\x00\x06\x00\x00\xf5\x00\x07\x00\x00\x00\x00\xf5\x00\x07\x00\x03\x00\x00\xf5\x00\x07\x00\x04\x00\x00\xf5\x00\x07\x00\x05\x00\x00\xf5\x00\x07\x00\x08\x00\x00\xf5\x00\x07\x00\x07\x00\x00\xed\x00\x12\x00\x00\x00\x00\xee\x00\x0b\x00\x00\x01\\\x00\x05\x00\x00\xef\x00\x05\x00\x00\xd9\x00\x08\x00\x00\x01\x80\x00\xdc\x00\t\x01\x00\x00'\x10\x00\xe1\x00\x08\x00\x01\x00\x01\x00\xe1\x00\x08\x00\x02\x00\x01\x00\xe1\x00\x08\x00\x03\x00\x01\x00\xe1\x00\x08\x00\x04\x00\x01\x00\xdb\x00\x07\x00\x01\x00\x00\xdb\x00\x07\x00\x02\x00\x00\xdb\x00\x07\x00\x03\x00\x00\xdb\x00\x07\x00\x04\x00\x00\xe2\x00\x05\x80\x03\xff\x00\r\x00\x00\x00\xa1\x00\x00\x00e\x00\x03\xff\x00\x10\x00\x00\x00\xa1\x00\x00\x00\xff\x00\x00\x00\x00\x03\xff\x00\r\x00\x00\x00\xa1\x00\x00\x01^\x00\x03\xff\x00\x9c\x00\x00\x00\xa1\x00\x00\x00f\x80\x00\xb1\x00\x8f\x00\x00\x03\xe9\x07\x00\x00\xb2\x00\x12\x00\xb3\x00\x05\x01\x00\xb6\x00\t\x00\x00\x00\x00\x00\x00\xb7\x00?\x00\x01\x00\x00\x00\xb8\x00\t\x00\x00\x00\x00\x00\x00\xba\x00\x07\x03\xe9\x01\x03\xff\x00\r\x00\x00`\xff\x00\x00\x01,\x80\x03\xff\x00\r\x00\x00`\xff\x00\x00\x01-\x01\x03\xff\x00\r\x00\x00`\xff\x00\x00\x01,\x80\x00\xed\x004\x00\x00\x00\x00\xee\x00\x0b\x97\x80\x01\\\x00\x05\xc0\x03\xff\x00\x15\x00\x00\x00\xa1\x00\x00\x00y\x01\x00\x00\x02\x1f@\x01\x00\x00\x03\xff\x00\r\x00\x00\x00\xa1\x00\x00\x00z\x01\x03\xff\x00\x10\x00\x00\x00\xa1\x00\x00\x01\xd2\x00\x00\x00\x00"
2019-09-06 14:05:43,755 sllurp.llrp_proto: DEBUG: decode_param pardata: b'\x00\x04\x00\xdf\x00\x06\x00\x01\x00\xe0\x00\n\x00\x01\x00\x01\x00\xc8\x01J\x00\x1d\x00\x01O\x00\x08\x00\x040\xd4\x01P\x00\x10@\x00\x14\x00\x00\x00\x00\x01Q\x00\x05\x00'
2019-09-06 14:05:43,756 sllurp.llrp_proto: DEBUG: decode_param data: b"\x00\xf4\x00C\x00\xf5\x00\x07\x00\x01\x00\x00\xf5\x00\x07\x00\x02\x00\x00\xf5\x00\x07\x00\x06\x00\x00\xf5\x00\x07\x00\x00\x00\x00\xf5\x00\x07\x00\x03\x00\x00\xf5\x00\x07\x00\x04\x00\x00\xf5\x00\x07\x00\x05\x00\x00\xf5\x00\x07\x00\x08\x00\x00\xf5\x00\x07\x00\x07\x00\x00\xed\x00\x12\x00\x00\x00\x00\xee\x00\x0b\x00\x00\x01\\\x00\x05\x00\x00\xef\x00\x05\x00\x00\xd9\x00\x08\x00\x00\x01\x80\x00\xdc\x00\t\x01\x00\x00'\x10\x00\xe1\x00\x08\x00\x01\x00\x01\x00\xe1\x00\x08\x00\x02\x00\x01\x00\xe1\x00\x08\x00\x03\x00\x01\x00\xe1\x00\x08\x00\x04\x00\x01\x00\xdb\x00\x07\x00\x01\x00\x00\xdb\x00\x07\x00\x02\x00\x00\xdb\x00\x07\x00\x03\x00\x00\xdb\x00\x07\x00\x04\x00\x00\xe2\x00\x05\x80\x03\xff\x00\r\x00\x00\x00\xa1\x00\x00\x00e\x00\x03\xff\x00\x10\x00\x00\x00\xa1\x00\x00\x00\xff\x00\x00\x00\x00\x03\xff\x00\r\x00\x00\x00\xa1\x00\x00\x01^\x00\x03\xff\x00\x9c\x00\x00\x00\xa1\x00\x00\x00f\x80\x00\xb1\x00\x8f\x00\x00\x03\xe9\x07\x00\x00\xb2\x00\x12\x00\ [... cut ...] \x00\x01\x00\x00\x00\xb8\x00\t\x00\x00\x00\x00\x00\x00\xba\x00\x07\x03\xe9\x01\x03\xff\x00\r\x00\x00`\xff\x00\x00\x01,\x80\x03\xff\x00\r\x00\x00`\xff\x00\x00\x01-\x01\x03\xff\x00\r\x00\x00`\xff\x00\x00\x01,\x80\x00\xed\x004\x00\x00\x00\x00\xee\x00\x0b\x97\x80\x01\\\x00\x05\xc0\x03\xff\x00\x15\x00\x00\x00\xa1\x00\x00\x00y\x01\x00\x00\x02\x1f@\x01\x00\x00\x03\xff\x00\r\x00\x00\x00\xa1\x00\x00\x00z\x01\x03\xff\x00\x10\x00\x00\x00\xa1\x00\x00\x01\xd2\x00\x00\x00\x00'
2019-09-06 14:05:43,790 sllurp.l

donkikos, 2021-04-30:
After setting the "session=0" I started getting all tags from Zebra FX9600.

donkikos, 2021-06-09:
Hi @schneiderB1 , try setting session=0. Your log shows that session is set to 2. From command line that would be

`sllurp inventory -s 0 -t 1 <IP>`

phuriphak, 2021-12-13:
@donkikos can i use real time output to csv file by column separate value.  i need to upload to MySQL

Brenwood911, 2025-10-29:
Bonjour, à partir du logiciel Windev de PCSOFT, j'essaie de récupérer les Tags lues par un FX9600 depuis plusieurs antennes mais il me manque des tags si l'on passe plus ou moins vite sous le portique. Je m'appuie sur le dotnet "Zebra-RFID-FXSeries-Host-DotNet-SDK_V107" mais à priori j'ai un problème de puissance et je n'arrive ni à lire les valeurs ni à pousser des valeurs. Avec 123RFID les tags remontent bien, existe t-il un fichier de config optimisé à pousser depuis windev afin de résoudre ce problème ? Merci pour votre aide