sllurp/sllurp #116: llrp.stopPolitely not stopping

djacobson, 2019-01-03:
Trying to reset and/or shutdown correctly; When calling stopPolitely, and adding a callback to the returned deferred object, my deferred callback never gets called, as described here:

        """Delete all active ROSpecs.  Return a Deferred that will be called
           when the DELETE_ROSPEC_RESPONSE comes back."""

All I see in the log is:

'will disconnect when stopped'

And, well, I guess something is not stopping? I know I need to wait until the deferred cb is called before I "reactor.stop()", or the Impinj will not be correctly reset (and, if it's not successfully reset, it literally heats up). :)

Anyways, any insight into debugging this would be appreciated.

djacobson, 2019-01-03:
It seems as though, in my app, in the following code:

```
    d = proto.stopPolitely(disconnect=True)
    d.addCallback(lambda _: reactor.stop())
```

...the callback just isn't called. If I skip the stopPolitely entirely, and just call reactor.stop(), it does shutdown. But, the reset state of the Impinj is suspect.

ransford, 2019-01-03:
Seems like a bug. I agree, the callback should fire once the DELETE_ACCESSSPEC_RESPONSE message arrives from the reader. Is that message actually arriving (check with debug-level logs)?

`sllurp/verb/inventory.py` calls `reactor.stop` via a callback (see `finish()` and `factory_args` in that file), but the way it wires up the callback is to pass an `onFinish` argument to the `LLRPClientFactory`. I don't know if that's helpful in your case. What a tangled web of callbacks we weave, when asynchronous code do we screve...

Do you have a tiny test case (like, a Python program that drives sllurp) that triggers the bug?

djacobson, 2019-01-04:
Thanks for the advice! Let me make sure the debug level is up to DEBUG and rerun the test.

djacobson, 2019-01-11:
So, I'm testing this with DEBUG level logging ON. I *think* I may be sniffing out a bug.

When I initiate the shutdown, I see the following normal log lines...

2019-01-11 10:42:28.379 [sllurp.llrp]: INFO: stopping politely
2019-01-11 10:42:28.379 [sllurp.llrp]: INFO: will disconnect when stopped
2019-01-11 10:42:28.380 [sllurp.llrp]: DEBUG: serializing DELETE_ACCESSSPEC command
2019-01-11 10:42:28.381 [sllurp.llrp]: DEBUG: serialized bytes: b'04290000000e0000000000000000'
2019-01-11 10:42:28.381 [sllurp.llrp]: DEBUG: done serializing DELETE_ACCESSSPEC command
2019-01-11 10:42:28.383 [sllurp.llrp]: DEBUG: state change: STATE_INVENTORYING -> STATE_SENT_DELETE_ACCESSSPEC
...

**But**, something to keep in mind here: I am continually reading tags when I try this shutdown. So, the next bunch of log lines have to to do with reading the tags...

2019-01-11 10:42:28.387 [sllurp.llrp_proto]: DEBUG: decode_ROAccessReport
2019-01-11 10:42:28.396 [sllurp.llrp_proto]: DEBUG: decode_TagReportData
2019-01-11 10:42:28.405 [sllurp.llrp_proto]: DEBUG: decode_EPCData (type=241 len=10)
2019-01-11 10:42:28.406 [sllurp.llrp_proto]: DEBUG: got EPCData; won't try EPC-96
2019-01-11 10:42:28.406 [sllurp.llrp_decoder]: DEBUG: found ROSpecID (type=9)
...

I never do see the expected log lines for a deferred shutdown:

"DEBUG: deserializing DELETE_ACCESSSPEC_RESPONSE command" + "DEBUG: done deserializing DELETE_ACCESSSPEC_RESPONSE command" (Note: I do see these log lines on start-up of the app, which is similar to a reset).

It almost seems as if continual tag reports callbacks "RO_ACCESS_REPORT" make the sllurp code "lose" the defferred shutdown callback... ??? All I see are continual:

"DEBUG: deserializing RO_ACCESS_REPORT command"

Does this help? I can search through the mass of debug log lines for other hints, but, this is the state I've been able to interpret so far.

ransford, 2019-01-12:
Are you seeing `DEBUG: done serializing DELETE_ACCESSSPEC command` at all once you've initiated the shutdown event? That's the message that will be logged right before the "DELETE_ACCESSSPEC" message gets shipped to the reader. Until it gets that message, the reader has no idea that it should stop reading tags.

djacobson, 2019-01-12:
Yes, see the 5th debug log line from the first set above (after the "stopping politely" log line (the shutdown is initiated).

```
2019-01-11 10:42:28.381 [sllurp.llrp]: DEBUG: done serializing DELETE_ACCESSSPEC command
```
It just doesn't seem to shutdown, goes straight into process more tag reports. Let me include a larger section of the log below starting from the "polite shutdown". The "done serializing DELETE_ACCESSSPEC" is still the 5th line below:
```
2019-01-11 10:42:28.379 [sllurp.llrp]: INFO: stopping politely
2019-01-11 10:42:28.379 [sllurp.llrp]: INFO: will disconnect when stopped
2019-01-11 10:42:28.380 [sllurp.llrp]: DEBUG: serializing DELETE_ACCESSSPEC command
2019-01-11 10:42:28.381 [sllurp.llrp]: DEBUG: serialized bytes: b'04290000000e0000000000000000'
2019-01-11 10:42:28.381 [sllurp.llrp]: DEBUG: done serializing DELETE_ACCESSSPEC command
2019-01-11 10:42:28.383 [sllurp.llrp]: DEBUG: state change: STATE_INVENTORYING -> STATE_SENT_DELETE_ACCESSSPEC
2019-01-11 10:42:28.387 [sllurp.llrp_proto]: DEBUG: decode_ROAccessReport
2019-01-11 10:42:28.396 [sllurp.llrp_proto]: DEBUG: decode_TagReportData
2019-01-11 10:42:28.405 [sllurp.llrp_proto]: DEBUG: decode_EPCData (type=241 len=10)
2019-01-11 10:42:28.406 [sllurp.llrp_proto]: DEBUG: got EPCData; won't try EPC-96
2019-01-11 10:42:28.406 [sllurp.llrp_decoder]: DEBUG: found ROSpecID (type=9)
2019-01-11 10:42:28.407 [sllurp.llrp_decoder]: DEBUG: found SpecIndex (type=14)
2019-01-11 10:42:28.408 [sllurp.llrp_decoder]: DEBUG: found InventoryParameterSpecID (type=10)
2019-01-11 10:42:28.409 [sllurp.llrp_decoder]: DEBUG: found AntennaID (type=1)
2019-01-11 10:42:28.409 [sllurp.llrp_decoder]: DEBUG: found PeakRSSI (type=6)
2019-01-11 10:42:28.410 [sllurp.llrp_decoder]: DEBUG: found ChannelIndex (type=7)
2019-01-11 10:42:28.410 [sllurp.llrp_decoder]: DEBUG: found FirstSeenTimestampUTC (type=2)
2019-01-11 10:42:28.411 [sllurp.llrp_decoder]: DEBUG: found LastSeenTimestampUTC (type=4)
2019-01-11 10:42:28.411 [sllurp.llrp_decoder]: DEBUG: found TagSeenCount (type=8)
2019-01-11 10:42:28.412 [sllurp.llrp_decoder]: DEBUG: found AccessSpecID (type=16)
2019-01-11 10:42:28.419 [sllurp.llrp_proto]: DEBUG: decode_OpSpecResult
2019-01-11 10:42:28.420 [sllurp.llrp_proto]: DEBUG: par={'ROSpecID': (1,), 'LastSeenTimestampUTC': (1547203327228048,), 'FirstSeenTimestampUTC': (1547203327228048,), 'PeakRSSI': (-38,), 'InventoryParameterSpecID': (1,), 'EPCData': {'EPC': b'6904032e', 'EPCLengthBits': 32}, 'TagSeenCount': (1,), 'AntennaID': (2,), 'AccessSpecID': (0,), 'ChannelIndex': (47,), 'SpecIndex': (1,)}
2019-01-11 10:42:28.426 [sllurp.llrp_proto]: DEBUG: decode_TagReportData
2019-01-11 10:42:28.427 [sllurp.llrp]: DEBUG: done deserializing RO_ACCESS_REPORT command
2019-01-11 10:42:28.428 [sllurp.llrp]: DEBUG: LLRPMessage received in state 20: <RO_ACCESS_REPORT>
```

ransford, 2019-01-12:
Oh ok, I guess I didn’t parse that properly earlier. Hmm. So it looks like
your program never receives a response to its request to delete the
AccessSpec. If you look at the traffic in wireshark (use ‘port 5084’ as the
capture filter) over the course of a session, what messages do you see
going back and forth around the time when you call stopPolitely()?

Wireshark’s protocol dissector for LLRP is pretty good!

djacobson, 2019-01-24:
So, I'm thinking about closing this issue because I'm having trouble reliably reproducing it. I did install Wireshark and have been using it to analyze port 5084... thx again for that advice. But, the app seems to properly shutdown (stopPolitely() with cb called) most of the time, with a curious delay (~ 5-10 secs). I'm suspecting I might see the issue again if I leave the app running and busily reading tags for an extended period of time (maybe that shutdown delay keeps getting longer?). I have the Wireshark capture files but, I suspect they look normal...

closed 2019-01-29 by djacobson

ransford, 2019-01-29:
You could turn up the log level and add more debugging messages to your application / to sllurp to diagnose that issue. Sorry it's hard to reproduce the failure; few things are more annoying than that!

edmond1992, 2020-04-17:
I got a similar problem, the shutdown task ran, but it never return to main despite it said STATE_DISCONNECTED and Connection was closed cleanly.

[debug.txt](https://github.com/ransford/sllurp/files/4494190/debug.txt)
```
#Some other stuff
SCAN_TIME = 5.0
def shutdown(factory):
    print('Shutting down')
    print(factory.politeShutdown())
    return 0

def main():
    factory = llrp.LLRPClientFactory(**readerargs)
    factory.addTagReportCallback(report)
    reactor.connectTCP(READER_IP_ADDRESS, llrp.LLRP_PORT, factory, timeout=3)
    reactor.addSystemEventTrigger('before', 'shutdown', shutdown, factory)
    # https://twistedmatrix.com/documents/current/core/howto/time.html
    reactor.callLater(SCAN_TIME, shutdown, factory)
    reactor.run()
    print("Back")
    return 0
```