sllurp/sllurp #118: LLRP ROSpec: RFTransmitter parameter, invalid HopTableID 1

anudeep0102, 2019-02-10:
I have been trying to connect LLRP compatible reader to the client(python) , which throws an error
**Error P_FieldError adding ROSpec: b'RFTransmitter parameter, invalid HopTableID (1)**.

Can any one help me to sort out this issue.?

Below is the response I get when I run the code given in the link,  https://github.com/ransford/sllurp

<RO_ACCESS_REPORT>
	<Ver>1</Ver>
	<Type>61</Type>
	<ID>1</ID>
	<TagReportData>[]</TagReportData>
</RO_ACCESS_REPORT>
<ROSpec>
	<ROSpecID>1</ROSpecID>
	<Priority>0</Priority>
	<CurrentState>'Disabled'</CurrentState>
	<ROBoundarySpec>
		<ROSpecStartTrigger>
			<ROSpecStartTriggerType>'Immediate'</ROSpecStartTriggerType>
		</ROSpecStartTrigger>
		<ROSpecStopTrigger>
			<ROSpecStopTriggerType>'Null'</ROSpecStopTriggerType>
			<DurationTriggerValue>0</DurationTriggerValue>
		</ROSpecStopTrigger>
	</ROBoundarySpec>
	<AISpec>
		<AntennaIDs>(1,)</AntennaIDs>
		<AISpecStopTrigger>
			<AISpecStopTriggerType>'Null'</AISpecStopTriggerType>
			<DurationTriggerValue>0</DurationTriggerValue>
		</AISpecStopTrigger>
		<InventoryParameterSpec>
			<InventoryParameterSpecID>1</InventoryParameterSpecID>
			<ProtocolID>1</ProtocolID>
			<AntennaConfiguration>
				<AntennaID>1</AntennaID>
				<RFTransmitter>
					<HopTableId>1</HopTableId>
					<ChannelIndex>1</ChannelIndex>
					<TransmitPower>0</TransmitPower>
				</RFTransmitter>
				<C1G2InventoryCommand>
					<TagInventoryStateAware>False</TagInventoryStateAware>
					<C1G2SingulationControl>
						<Session>2</Session>
						<TagPopulation>4</TagPopulation>
						<TagTransitTime>0</TagTransitTime>
					</C1G2SingulationControl>
				</C1G2InventoryCommand>
			</AntennaConfiguration>
		</InventoryParameterSpec>
	</AISpec>
	<ROReportSpec>
		<N>0</N>
		<ROReportTrigger>'Upon_N_Tags_Or_End_Of_AISpec'</ROReportTrigger>
		<TagReportContentSelector>
			<EnableROSpecID>True</EnableROSpecID>
			<EnableSpecIndex>False</EnableSpecIndex>
			<EnableInventoryParameterSpecID>False</EnableInventoryParameterSpecID>
			<EnableAntennaID>True</EnableAntennaID>
			<EnableChannelIndex>False</EnableChannelIndex>
			<EnablePeakRSSI>True</EnablePeakRSSI>
			<EnableFirstSeenTimestamp>False</EnableFirstSeenTimestamp>
			<EnableLastSeenTimestamp>True</EnableLastSeenTimestamp>
			<EnableTagSeenCount>True</EnableTagSeenCount>
			<EnableAccessSpecID>False</EnableAccessSpecID>
		</TagReportContentSelector>
	</ROReportSpec>
</ROSpec>

ransford, 2019-02-11:
What kind of reader? What country (matters because of local regulations dictating frequency ranges)?

Could you add this code to your program after the `logging.getLogger()...` line, and attach the resulting log file to this issue?

```python
import sllurp.log
sllurp.log.init_logging(debug=True, logfile='debug.txt')
```

anudeep0102, 2019-02-12:
The reader model is IF 2 from Intermec. This was purchased in U.S, but right now i have been testing it in INDIA.Please find the log file below in the below link.

[debug.txt](https://github.com/ransford/sllurp/files/2854170/debug.txt)

thijmenketel, 2019-02-12:
Might be a problem with the HopTableID as Impinj defaults to ID 1 it is also hardcoded to 1 in sllurp (see [line](https://github.com/ransford/sllurp/blob/develop/sllurp/llrp_proto.py#L3976)). However from your error messages I can see that the default HopTableID that your reader is reporting is 0, which means sllurp will give an error.

This might be fixable by a simple edit (might look at it if I have time).

anudeep0102, 2019-02-12:
@thijmenketel what value should i replace the HopTableID with?. I changed it to 0 and the error has been sorted out.But still i am unable to read the tags.

ransford, 2019-02-12:
I see your log file `debug.txt` ends with several failed `ConnectionAttemptEvent`s, indicating that something might be holding a connection open to your reader. If you're still getting that response from the reader, try power-cycling the reader or running `sllurp reset <reader IP>`.

Can you say more about "unable to read the tags"? Do you mean your callback's not being called? Does `sllurp inventory -n1 <reader IP>` report any tags?

anudeep0102, 2019-02-12:
@ransford Thanks for the reply. Yes, when i stop the existing connection and run the code,`ConnectionAttemptEvent `fails.This has been sorted out.

And regarding the callbacks, the tags are not read continuously. If i run the code, the reader reads the tags and displays tag info, tag count, rssi e.t.c.. Only if i run the code again the reader starts capturing the tag data.

Is there a setting or mode that i need to set to the reader,so that it sends the tag information continuously.?

ransford, 2019-02-12:
Try `sllurp inventory -n1 <reader IP>` for continuous output. The `-n1` parameter means "send a report for every 1 tag sighting." You can change the `-n` parameter to control how often you receive reports.

anudeep0102, 2019-02-13:
@thijmenketel thanks for help. Once i changed the HopTableID Value to 0 i am able to detect the tags.

anudeep0102, 2019-02-13:
@ransford instead of running the inventory script in terminal window, is there any other option to define the parameter n in script itself, so that I can read the tag continuously for defined duration using the minimal example script which is as follows,

```
# Minimal example; see inventory.py for more.
from sllurp import llrp
from twisted.internet import reactor
import logging
logging.getLogger().setLevel(logging.INFO)

def cb (tagReport):
    tags = tagReport.msgdict['RO_ACCESS_REPORT']['TagReportData']
    print 'tags:', tags

factory = llrp.LLRPClientFactory()
factory.addTagReportCallback(cb)
reactor.connectTCP('myreader', llrp.LLRP_PORT, factory)
reactor.run()
```

thijmenketel, 2019-02-13:
Take a look at [this](https://github.com/ransford/sllurp/blob/develop/sllurp/verb/inventory.py) inventory file, when you scroll down to line 64 the settings for the LLRP session are defined. You can play around with those as you like by creating your own factory arguments and passing is to the LLRPClientFactory (see same file, line 106).

Mahamutha, 2019-04-24:
> Might be a problem with the HopTableID as Impinj defaults to ID 1 it is also hardcoded to 1 in sllurp (see [line](https://github.com/ransford/sllurp/blob/develop/sllurp/llrp_proto.py#L3976)). [...]

Hi,

I am getting the error as 

`Error P_FieldError adding ROSpec: b'RFTransmitter parameter, invalid HopTableID (1)'`

I have changed the HopTableID to 0 in llrp_proto.py

```
antconf = {
                'AntennaID': antid,
                'RFTransmitter': {
                    'HopTableId': 0,
                    'ChannelIndex': 1,
                    'TransmitPower': transmit_power,
                },
```

and similarly in impinj-r1000-hybrid-mode-rospec.xml
```

          <AntennaID>1</AntennaID>
          <RFTransmitter>
            <HopTableID>0</HopTableID>
            <ChannelIndex>0</ChannelIndex>
            <TransmitPower>61</TransmitPower>
          </RFTransmitter>

        <AntennaID>2</AntennaID>
          <RFTransmitter>
            <HopTableID>0</HopTableID>
            <ChannelIndex>0</ChannelIndex>
            <TransmitPower>61</TransmitPower>
          </RFTransmitter>

         <AntennaID>3</AntennaID>
          <RFTransmitter>
            <HopTableID>0</HopTableID>
            <ChannelIndex>0</ChannelIndex>
            <TransmitPower>61</TransmitPower>
          </RFTransmitter>

         <AntennaID>4</AntennaID>
          <RFTransmitter>
            <HopTableID>0</HopTableID>
            <ChannelIndex>0</ChannelIndex>
            <TransmitPower>61</TransmitPower>
          </RFTransmitter>
```

Whereas still I am facing the issue. Could you please help me out to resolve this?

thijmenketel, 2019-04-24:
Sllurp doesn't do anything with impinj-r1000-hybrid-mode-rospec.xml file, its more for reference.

Can you add some debug information (see above comments for how to)?

Mahamutha, 2019-04-24:
Sure, Thanks for the clarification for impinj-r1000-hybrif-mode-rospec.xml. 

I have given the command,
 `sllurp inventory -n1 -t1 192.168.1.106`

and the response I got is,

```
 (venv) zoid@zoid-Inspiron-5379:~/zoid_docs/PycharmProjects/sllurp$ sllurp inventory -n1 -t1 192.168.1.106
2019-04-24 14:13:15,318 sllurp.verb.inventory: INFO: Antenna map: {'192.168.1.106': {'1': 'Antenna 1'}}
2019-04-24 14:13:15,319 sllurp.llrp: INFO: connecting to 192.168.1.106:5084...
2019-04-24 14:13:15,327 sllurp.llrp: INFO: will reset reader state on connect
2019-04-24 14:13:15,327 sllurp.llrp: INFO: will start inventory on connect
2019-04-24 14:13:15,327 sllurp.llrp: INFO: using antennas: [1]
2019-04-24 14:13:15,327 sllurp.llrp: INFO: transmit power: {1: 0}
2019-04-24 14:13:15,328 sllurp.llrp: INFO: connected to ('192.168.1.106', 5084) (192.168.1.106:5084)
2019-04-24 14:13:15,635 sllurp.llrp: INFO: using reader mode: None
2019-04-24 14:13:15,686 sllurp.verb.inventory: INFO: no tags seen
2019-04-24 14:13:15,734 sllurp.llrp: INFO: stopping politely
2019-04-24 14:13:15,941 sllurp.verb.inventory: INFO: saw tag(s): [{'AntennaID': (1,),
  'EPC-96': b'e200383d1149810975a59589',
  'LastSeenTimestampUTC': (946704138507196,),
  'PeakRSSI': (-50,),
  'ROSpecID': (1,),
  'TagSeenCount': (1,)},
 {'AntennaID': (1,),
  'EPC-96': b'e200383d11499ac975a595f0',
  'LastSeenTimestampUTC': (946704237479189,),
  'PeakRSSI': (-55,),
  'ROSpecID': (1,),
  'TagSeenCount': (1,)},
 {'AntennaID': (1,),
  'EPC-96': b'e200383d1149198975a593eb',
  'LastSeenTimestampUTC': (946704237454189,),
  'PeakRSSI': (-56,),
  'ROSpecID': (1,),
  'TagSeenCount': (1,)},
 {'AntennaID': (1,),
  'EPC-96': b'e200383d1149178975a593e3',
  'LastSeenTimestampUTC': (946704138713202,),
  'PeakRSSI': (-58,),
  'ROSpecID': (1,),
  'TagSeenCount': (1,)},
 {'AntennaID': (1,),
  'EPC-96': b'e200383d114905c975a5939c',
  'LastSeenTimestampUTC': (946704237373198,),
  'PeakRSSI': (-60,),
  'ROSpecID': (1,),
  'TagSeenCount': (6,)},
 {'AntennaID': (1,),
  'EPC-96': b'e200383d1149114975a593ca',
  'LastSeenTimestampUTC': (946704237322231,),
  'PeakRSSI': (-55,),
  'ROSpecID': (1,),
  'TagSeenCount': (3,)},
 {'AntennaID': (1,),
  'EPC-96': b'e200383d1149134975a593d2',
  'LastSeenTimestampUTC': (946704237403140,),
  'PeakRSSI': (-54,),
  'ROSpecID': (1,),
  'TagSeenCount': (2,)},
 {'AntennaID': (1,),
  'EPC-96': b'e200383d114901c975a5938c',
  'LastSeenTimestampUTC': (946704138603199,),
  'PeakRSSI': (-55,),
  'ROSpecID': (1,),
  'TagSeenCount': (1,)},
 {'AntennaID': (1,),
  'EPC-96': b'e200383d114994c975a595d8',
  'LastSeenTimestampUTC': (946704148089302,),
[... 720 lines cut ...]
  'LastSeenTimestampUTC': (946704379632266,),
  'TagSeenCount': (1,)}]
2019-04-24 14:15:12,138 sllurp.verb.inventory: INFO: saw tag(s): [{'ChannelIndex': (3,),
  'EPC-96': b'e200383d1149198975a593eb',
  'LastSeenTimestampUTC': (946704379651280,),
  'TagSeenCount': (1,)}]
^C2019-04-24 14:15:24,749 sllurp.llrp: INFO: stopping politely
2019-04-24 14:15:24,749 sllurp.llrp: INFO: will disconnect when stopped
2019-04-24 14:15:24,830 sllurp.llrp: INFO: lost connection: Connection to the other side was lost in a non-clean fashion: Connection lost.
2019-04-24 14:15:24,830 sllurp.verb.inventory: INFO: total # of tags seen: 205 (1 tags/second)
2019-04-24 14:15:24,830 sllurp.llrp: INFO: disconnecting
```
I have just made a change (HopTableId from "1" to "0") in llrp_proto.py

```
antconf = {
                'AntennaID': antid,
                'RFTransmitter': {
                    'HopTableId': 0,
                    'ChannelIndex': 1,
                    'TransmitPower': transmit_power,
                },
```

Is there any possibility to get the Reader ID?

thijmenketel, 2019-04-24:
It seems to be working correctly now right?

What do you mean by reader ID?

Mahamutha, 2019-04-24:
> It seems to be working correctly now right?
> 
> What do you mean by reader ID?

Actually my question is, I am listening the data with the help of reader's ip address. Is there any option to listen the data from RFID reader in server?

I am using intermec IF2A000014.

ransford, 2019-04-25:
@Mahamutha I think we're not sure what you mean.

If you mean "can I use a hostname like `my.reader.com` instead of referring to that reader by its IP address," yes you can.

If you are wondering how to get more information out of each tag report, you probably want to write your own callback. That should be pretty easy if you follow the README to do it.

closed 2021-01-24 by ransford