sllurp/sllurp #119: ErrorMessage while enabling Impinjextensions

Mahamutha, 2019-02-13:
Hi,

I have intermec IF2A000014 RFID Reader with 902-928 MHz. Can you please help me out, to enable impinjextension,

`2019-02-13 09:54:27,947 sllurp.verb.inventory: INFO: Antenna map: {u'192.168.1.10': {'1': 'Antenna 1'}}
2019-02-13 09:54:27,947 sllurp.llrp: INFO: connecting to 192.168.1.10:5084...
2019-02-13 09:54:27,953 sllurp.llrp: INFO: will reset reader state on connect
2019-02-13 09:54:27,953 sllurp.llrp: INFO: will start inventory on connect
2019-02-13 09:54:27,953 sllurp.llrp: INFO: using antennas: [1]
2019-02-13 09:54:27,953 sllurp.llrp: INFO: transmit power: {1: 0}
2019-02-13 09:54:27,953 sllurp.llrp: INFO: connected to (u'192.168.1.10', 5084) (192.168.1.10:5084)
2019-02-13 09:54:27,978 sllurp.llrp: ERROR: unexpected response ErrorMessage while enabling Impinjextensions
^C2019-02-13 09:56:43,547 sllurp.llrp: INFO: stopping politely
2019-02-13 09:56:43,547 sllurp.llrp: INFO: will disconnect when stopped
2019-02-13 09:56:43,587 sllurp.llrp: INFO: lost connection: Connection to the other side was lost in a non-clean fashion: Connection lost.
2019-02-13 09:56:43,588 sllurp.verb.inventory: INFO: total # of tags seen: 0 (0 tags/second)
2019-02-13 09:56:43,588 sllurp.llrp: INFO: disconnecting
`

ransford, 2019-02-13:
Do Intermec readers support Impinj extensions? I don't know. Aren't they different companies?

Mahamutha, 2019-02-13:
Is there any provision to disable impinj extensions?

ransford, 2019-02-13:
Yes, if you are able to provide a [more specific bug report](https://www.chiark.greenend.org.uk/~sgtatham/bugs.html), we may be able to help. In #118 we are resolving a minor issue with Intermec readers. The reporter of #118 is using a command like `sllurp inventory -n1 <reader IP address>` to see tags through an Intermec reader (the same model as yours, I believe).

First question: what exactly are you running on your command line that produces the above output? If we're attempting to enable Impinj extensions by default, that's a bug.

Mahamutha, 2019-02-13:
I have followed the below procedure,

> $ git clone https://github.com/ransford/sllurp.git
$ cd sllurp
$ virtualenv .venv
$ source .venv/bin/activate
$ pip install .
$ sllurp inventory ip.add.re.ss

So, when I give "sllurp inventory IPAdress", I am just getting the below log

> (.venv) root@zoid-Inspiron-5379:/home/zoid/Downloads/sllurp# sllurp inventory 192.168.1.10
2019-02-13 11:32:33,808 sllurp.verb.inventory: INFO: Antenna map: {u'192.168.1.10': {'1': 'Antenna 1'}}
2019-02-13 11:32:33,809 sllurp.llrp: INFO: connecting to 192.168.1.10:5084...
2019-02-13 11:32:33,908 sllurp.llrp: INFO: will reset reader state on connect
2019-02-13 11:32:33,908 sllurp.llrp: INFO: will start inventory on connect
2019-02-13 11:32:33,908 sllurp.llrp: INFO: using antennas: [1]
2019-02-13 11:32:33,909 sllurp.llrp: INFO: transmit power: {1: 0}
2019-02-13 11:32:33,910 sllurp.llrp: INFO: connected to (u'192.168.1.10', 5084) (192.168.1.10:5084)
2019-02-13 11:32:34,511 sllurp.llrp: ERROR: unexpected response ErrorMessage while enabling Impinjextensions
^C2019-02-13 11:32:40,306 sllurp.llrp: INFO: stopping politely
2019-02-13 11:32:40,306 sllurp.llrp: INFO: will disconnect when stopped
2019-02-13 11:32:40,333 sllurp.llrp: INFO: lost connection: Connection to the other side was lost in a non-clean fashion: Connection lost.
2019-02-13 11:32:40,333 sllurp.verb.inventory: INFO: total # of tags seen: 0 (0 tags/second)
2019-02-13 11:32:40,333 sllurp.llrp: INFO: disconnecting
(.venv) root@zoid-Inspiron-5379:/home/zoid/Downloads/sllurp#

ransford, 2019-02-13:
Thanks. I think you found a bug, and I think I just fixed it. Could you run:
```shell
$ cd /path/to/sllurp
$ git fetch -p
$ git checkout bugfix/119-impinj-defaults
$ pip install -U -e .
$ sllurp inventory -n1 <reader IP>
```

It seems we were accidentally enabling Impinj extensions even when we shouldn't have.

Mahamutha, 2019-02-13:
@ransford, Following log for your reference,

**$git fetch -p**

> (.venv) root@zoid-Inspiron-5379:/home/zoid/Downloads/sllurp# git fetch -p
remote: Enumerating objects: 10, done.
remote: Counting objects: 100% (10/10), done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 10 (delta 5), reused 10 (delta 5), pack-reused 0
Unpacking objects: 100% (10/10), done.
From https://github.com/ransford/sllurp
   4754930..3ec9c04  develop                    -> origin/develop
 * [new branch]      bugfix/119-impinj-defaults -> origin/bugfix/119-impinj-defaults
   3579404..e00422f  master                     -> origin/master
 * [new tag]         v0.5.0                     -> v0.5.0

**$ git checkout bugfix/119-impinj-defaults**

> (.venv) root@zoid-Inspiron-5379:/home/zoid/Downloads/sllurp# git checkout bugfix/119-impinj-defaults
error: Your local changes to the following files would be overwritten by checkout:
	setup.py
	sllurp/llrp.py
Please commit your changes or stash them before you switch branches.
Aborting

**$ pip install -U -e .**

> (.venv) root@zoid-Inspiron-5379:/home/zoid/Downloads/sllurp# pip install -U -e .
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Obtaining file:///home/zoid/Downloads/sllurp
Requirement already satisfied, skipping upgrade: click in ./.venv/lib/python2.7/site-packages (from sllurp==0.4.3) (7.0)
Requirement already satisfied, skipping upgrade: twisted in ./.venv/lib/python2.7/site-packages (from sllurp==0.4.3) (18.9.0)
Requirement already satisfied, skipping upgrade: monotonic in ./.venv/lib/python2.7/site-packages (from sllurp==0.4.3) (1.5)
Requirement already satisfied, skipping upgrade: PyHamcrest>=1.9.0 in ./.venv/lib/python2.7/site-packages (from twisted->sllurp==0.4.3) (1.9.0)
Requirement already satisfied, skipping upgrade: zope.interface>=4.4.2 in ./.venv/lib/python2.7/site-packages (from twisted->sllurp==0.4.3) (4.6.0)
Requirement already satisfied, skipping upgrade: hyperlink>=17.1.1 in ./.venv/lib/python2.7/site-packages (from twisted->sllurp==0.4.3) (18.0.0)
Requirement already satisfied, skipping upgrade: Automat>=0.3.0 in ./.venv/lib/python2.7/site-packages (from twisted->sllurp==0.4.3) (0.7.0)
Requirement already satisfied, skipping upgrade: attrs>=17.4.0 in ./.venv/lib/python2.7/site-packages (from twisted->sllurp==0.4.3) (18.2.0)
Requirement already satisfied, skipping upgrade: incremental>=16.10.1 in ./.venv/lib/python2.7/site-packages (from twisted->sllurp==0.4.3) (17.5.0)
Requirement already satisfied, skipping upgrade: constantly>=15.1 in ./.venv/lib/python2.7/site-packages (from twisted->sllurp==0.4.3) (15.1.0)
Requirement already satisfied, skipping upgrade: six in ./.venv/lib/python2.7/site-packages (from PyHamcrest>=1.9.0->twisted->sllurp==0.4.3) (1.12.0)
Requirement already satisfied, skipping upgrade: setuptools in ./.venv/lib/python2.7/site-packages (from PyHamcrest>=1.9.0->twisted->sllurp==0.4.3) (40.8.0)
Requirement already satisfied, skipping upgrade: idna>=2.5 in ./.venv/lib/python2.7/site-packages (from hyperlink>=17.1.1->twisted->sllurp==0.4.3) (2.8)
Installing collected packages: sllurp
  Found existing installation: sllurp 0.4.3
    Uninstalling sllurp-0.4.3:
      Successfully uninstalled sllurp-0.4.3
  Running setup.py develop for sllurp
Successfully installed sllurp

**$ sllurp inventory -n1 <reader IP>**

> (.venv) root@zoid-Inspiron-5379:/home/zoid/Downloads/sllurp# sllurp inventory -n1 192.168.1.10
2019-02-13 12:29:53,321 sllurp.verb.inventory: INFO: Antenna map: {u'192.168.1.10': {'1': 'Antenna 1'}}
2019-02-13 12:29:53,322 sllurp.llrp: INFO: connecting to 192.168.1.10:5084...
2019-02-13 12:29:53,340 sllurp.llrp: INFO: will reset reader state on connect
2019-02-13 12:29:53,340 sllurp.llrp: INFO: will start inventory on connect
2019-02-13 12:29:53,340 sllurp.llrp: INFO: using antennas: [1]
2019-02-13 12:29:53,340 sllurp.llrp: INFO: transmit power: {1: 0}
2019-02-13 12:29:53,340 sllurp.llrp: INFO: connected to (u'192.168.1.10', 5084) (192.168.1.10:5084)
2019-02-13 12:29:53,368 sllurp.llrp: ERROR: unexpected response ErrorMessage while enabling Impinjextensions

Mahamutha, 2019-02-13:
@ransford, Thanks for your suggestion. It works after following the below steps,
`$ cd /path/to/sllurp
$ git fetch -p
$ git checkout bugfix/119-impinj-defaults
$ pip install -U -e .
$ sllurp inventory -n1 <reader IP>`

Now, when I give 

> $sllurp inventory -n1 -t1 IPAddress

I am getting the below log,

> (.venv) root@zoid-Inspiron-5379:/home/zoid/PycharmProjects/sllurp# sllurp inventory -n1 -t1 192.168.1.10
2019-02-13 15:43:11,329 sllurp.verb.inventory: INFO: Antenna map: {u'192.168.1.10': {'1': 'Antenna 1'}}
2019-02-13 15:43:11,330 sllurp.llrp: INFO: connecting to 192.168.1.10:5084...
2019-02-13 15:43:11,359 sllurp.llrp: INFO: will reset reader state on connect
2019-02-13 15:43:11,359 sllurp.llrp: INFO: will start inventory on connect
2019-02-13 15:43:11,359 sllurp.llrp: INFO: using antennas: [1]
2019-02-13 15:43:11,359 sllurp.llrp: INFO: transmit power: {1: 0}
2019-02-13 15:43:11,360 sllurp.llrp: INFO: connected to (u'192.168.1.10', 5084) (192.168.1.10:5084)
2019-02-13 15:43:11,473 sllurp.llrp: INFO: using reader mode: None
2019-02-13 15:43:11,510 sllurp.verb.inventory: INFO: no tags seen
2019-02-13 15:43:11,567 sllurp.llrp: INFO: stopping politely
2019-02-13 15:43:11,606 sllurp.llrp: INFO: Impinj search mode? None
2019-02-13 15:43:11,606 sllurp.llrp_proto: INFO: will report every ~N=1 tags
2019-02-13 15:43:11,606 sllurp.llrp: INFO: starting inventory
2019-02-13 15:44:15,601 sllurp.verb.inventory: INFO: saw tag(s): [{u'ChannelIndex': (12,),
  u'EPC-96': 'e200383d1149b30975a59651',
  u'LastSeenTimestampUTC': (946708708987663,),
  u'TagSeenCount': (1,)}]
2019-02-13 15:44:18,472 sllurp.verb.inventory: INFO: saw tag(s): [{u'ChannelIndex': (12,),
  u'EPC-96': 'e200383d1148fc0975a59375',
  u'LastSeenTimestampUTC': (946708711769671,),
  u'TagSeenCount': (1,)}]
2019-02-13 15:44:24,912 sllurp.verb.inventory: INFO: saw tag(s): [{u'ChannelIndex': (6,),
  u'EPC-96': 'e200383d1149a90975a59629',
  u'LastSeenTimestampUTC': (946708718117656,),
  u'TagSeenCount': (1,)}]
2019-02-13 15:45:50,735 sllurp.verb.inventory: INFO: saw tag(s): [{u'ChannelIndex': (39,),
  u'EPC-96': 'e200383d1149a90975a59629',
  u'LastSeenTimestampUTC': (946708803131659,),
  u'TagSeenCount': (1,)}]
2019-02-13 15:45:50,757 sllurp.verb.inventory: INFO: saw tag(s): [{u'ChannelIndex': (46,),
  u'EPC-96': 'e200383d1149a90975a59629',
  u'LastSeenTimestampUTC': (946708803191666,),
  u'TagSeenCount': (1,)}]
2019-02-13 15:45:51,032 sllurp.verb.inventory: INFO: saw tag(s): [{u'ChannelIndex': (26,),
  u'EPC-96': 'e200383d1149a90975a59629',
  u'LastSeenTimestampUTC': (946708803415650,),
  u'TagSeenCount': (1,)}]

I have 50 tags in front of reader, I am not able to read all the 50 tags for every 1 sec interval.

Mahamutha, 2019-02-13:
Hi,

Could you please help me out to read all the tags in an environment (i.e., UHF Passive Tags and compatible with Intermec RFID Reader) continuously. I have followed your suggestion and by giving the below command, I can able to read a tag continuously say for ex.2 to 3 mins only after that I am not able to read the same. Could you please help me out in this? 

`$sllurp inventory -n1 -t1 IPAdress`

[Sllurp_llrp_log.txt](https://github.com/ransford/sllurp/files/2860660/Sllurp_llrp_log.txt)
[Log_Test_File.txt](https://github.com/ransford/sllurp/files/2860667/Log_Test_File.txt)

ransford, 2019-02-13:
Looks like your reader saw 18 tags, one much more than the others:
```shell
$ grep EPC-96 Sllurp_llrp_log.txt | sort | uniq -c | sort -nr
 180   u'EPC-96': 'e200383d11497f0975a59581',
   4   u'EPC-96': 'e200383d1149ac4975a59636',
   1   u'EPC-96': 'e200383d1149b30975a59651',
   1   u'EPC-96': 'e200383d1149b08975a59647',
   1   u'EPC-96': 'e200383d1149990975a595e9',
   1   u'EPC-96': 'e200383d1149968975a595df',
   1   u'EPC-96': 'e200383d114992c975a595d0',
   1   u'EPC-96': 'e200383d1149910975a595c9',
   1   u'EPC-96': 'e200383d11498a4975a595ae',
   1   u'EPC-96': 'e200383d1149804975a59586',
   1   u'EPC-96': 'e200383d11497ec975a59580',
   1   u'EPC-96': 'e200383d1149260975a5941d',
   1   u'EPC-96': 'e200383d1149198975a593eb',
   1   u'EPC-96': 'e200383d1149194975a593ea',
   1   u'EPC-96': 'e200383d1149158975a593db',
   1   u'EPC-96': 'e200383d1149114975a593ca',
   1   u'EPC-96': 'e200383d114901c975a5938c',
   1   u'EPC-96': 'e200383d1148fa0975a5936d'
```

I would suggest experimenting with different transmit power levels (`-X`) and sessions (`-s`) when running the `inventory` tool. Unfortunately there are lots of variables here; you'll have to try a few different combinations of values.

Thank you for testing the `bugfix/119-impinj-defaults` branch. I'm going to merge it into `develop` momentarily.

Mahamutha, 2019-02-14:
@ransford Sure, Thanks for your suggestion.

closed 2021-01-24 by ransford