sllurp/sllurp #121: EPC Registration (Multiple Masking)

cooolallen, 2019-02-22:
Hi Ransford,
Thanks for supporting this package. I really have a good time playing around with this. According to #113 , it can mask tags by starting few bits (or full bits). I am wondering if there a way I can do things like registration? When I mention registration I mean given a list of masks, the tags fit any of the masks will response. For example. sllurp inventory <reader_ip> --tag-filter-mask '123, 666', only tags starting with 123 or 666 in their epc-id to respond.

Or is there any other way I can achieve the goal with a different approach?
Thanks!

ransford, 2019-02-24:
I think the easiest way to filter with multiple tag masks is to use several concurrent C1G2Filter specs. @thijmenketel what do you think?

I just drafted an implementation of this in the `feature/121-multi-filter` branch. Would you please try it, @cooolallen? All you have to do if you're using `sllurp inventory` is to provide multiple `--tag-filter-mask <mask>` arguments (so `--tag-filter-mask A --tag-filter-mask B [...]`).

ransford, 2019-02-24:
(created #123 for my `feature/121-multi-filter` branch)

cooolallen, 2019-02-25:
Hi Ransford,
Thanks for the quick supporting!
I just tested on that branch and had this error message when I pass the only one mask.

```
(venv) cooolallens-MacBook-Pro:sllurp cooolallen$ sllurp inventory 192.168.1.109 --tag-filter-mask 300833b2 
2019-02-25 16:13:56,117 sllurp.verb.inventory: INFO: Antenna map: {'192.168.1.109': {'1': 'Antenna 1'}}
2019-02-25 16:13:56,117 sllurp.llrp: INFO: connecting to 192.168.1.109:5084...
2019-02-25 16:13:56,119 sllurp.llrp: INFO: will reset reader state on connect
2019-02-25 16:13:56,119 sllurp.llrp: INFO: will start inventory on connect
2019-02-25 16:13:56,119 sllurp.llrp: INFO: using antennas: [1]
2019-02-25 16:13:56,119 sllurp.llrp: INFO: transmit power: {1: 0}
2019-02-25 16:13:56,120 sllurp.llrp: INFO: connected to ('192.168.1.109', 5084) (192.168.1.109:5084)
2019-02-25 16:13:56,408 sllurp.llrp: INFO: using reader mode: None
2019-02-25 16:13:56,792 sllurp.llrp: INFO: stopping politely
2019-02-25 16:13:56,806 sllurp.llrp: INFO: Impinj search mode? None
2019-02-25 16:13:56,806 sllurp.llrp: INFO: starting inventory
2019-02-25 16:13:56,806 sllurp.llrp: ERROR: can only concatenate tuple (not "str") to tuple
Traceback (most recent call last):
  File "/Users/cooolallen/GoogleDrive/RFID/sllurp/venv/lib/python3.7/site-packages/sllurp/llrp.py", line 860, in send_ADD_ROSPEC
    'ROSpec': rospec,
  File "/Users/cooolallen/GoogleDrive/RFID/sllurp/venv/lib/python3.7/site-packages/sllurp/llrp.py", line 1294, in sendMessage
    llrp_msg = LLRPMessage(msgdict=msg_dict)
  File "/Users/cooolallen/GoogleDrive/RFID/sllurp/venv/lib/python3.7/site-packages/sllurp/llrp.py", line 38, in __init__
    self.serialize()
  File "/Users/cooolallen/GoogleDrive/RFID/sllurp/venv/lib/python3.7/site-packages/sllurp/llrp.py", line 58, in serialize
    data = encoder(self.msgdict[name])
  File "/Users/cooolallen/GoogleDrive/RFID/sllurp/venv/lib/python3.7/site-packages/sllurp/llrp_proto.py", line 487, in encode_AddROSpec
    return encode('ROSpec')(msg['ROSpec'])
  File "/Users/cooolallen/GoogleDrive/RFID/sllurp/venv/lib/python3.7/site-packages/sllurp/llrp_proto.py", line 1632, in encode_ROSpec
    data += encode('AISpec')(par['AISpec'])
  File "/Users/cooolallen/GoogleDrive/RFID/sllurp/venv/lib/python3.7/site-packages/sllurp/llrp_proto.py", line 2262, in encode_AISpec
    data += encode('InventoryParameterSpec')(par['InventoryParameterSpec'])
  File "/Users/cooolallen/GoogleDrive/RFID/sllurp/venv/lib/python3.7/site-packages/sllurp/llrp_proto.py", line 2368, in encode_InventoryParameterSpec
    data += encode('AntennaConfiguration')(antconf)
  File "/Users/cooolallen/GoogleDrive/RFID/sllurp/venv/lib/python3.7/site-packages/sllurp/llrp_proto.py", line 2398, in encode_AntennaConfiguration
    data += encode('C1G2InventoryCommand')(par['C1G2InventoryCommand'])
  File "/Users/cooolallen/GoogleDrive/RFID/sllurp/venv/lib/python3.7/site-packages/sllurp/llrp_proto.py", line 2472, in encode_C1G2InventoryCommand
    data += encode('C1G2Filter')(filt)
  File "/Users/cooolallen/GoogleDrive/RFID/sllurp/venv/lib/python3.7/site-packages/sllurp/llrp_proto.py", line 2513, in encode_C1G2Filter
    par['C1G2TagInventoryMask'])
  File "/Users/cooolallen/GoogleDrive/RFID/sllurp/venv/lib/python3.7/site-packages/sllurp/llrp_proto.py", line 2534, in encode_C1G2TagInventoryMask
    par['TagMask'] += '0'           # pad with zero
TypeError: can only concatenate tuple (not "str") to tuple
^C2019-02-25 16:14:19,156 sllurp.llrp: INFO: stopping politely
2019-02-25 16:14:19,156 sllurp.llrp: INFO: will disconnect when stopped
2019-02-25 16:14:19,176 sllurp.llrp: INFO: lost connection: Connection to the other side was lost in a non-clean fashion: Connection lost.
2019-02-25 16:14:19,176 sllurp.verb.inventory: INFO: total # of tags seen: 0 (0 tags/second)
2019-02-25 16:14:19,176 sllurp.llrp: INFO: disconnecting
```

Different error messages are given if I pass two masks.  

```
(venv) cooolallens-MacBook-Pro:sllurp cooolallen$ sllurp inventory 192.168.1.109 --tag-filter-mask 300833b2 --tag-filter-mask 3009
2019-02-25 16:17:47,289 sllurp.verb.inventory: INFO: Antenna map: {'192.168.1.109': {'1': 'Antenna 1'}}
2019-02-25 16:17:47,290 sllurp.llrp: INFO: connecting to 192.168.1.109:5084...
2019-02-25 16:17:47,291 sllurp.llrp: INFO: will reset reader state on connect
2019-02-25 16:17:47,291 sllurp.llrp: INFO: will start inventory on connect
2019-02-25 16:17:47,291 sllurp.llrp: INFO: using antennas: [1]
2019-02-25 16:17:47,291 sllurp.llrp: INFO: transmit power: {1: 0}
2019-02-25 16:17:47,291 sllurp.llrp: INFO: connected to ('192.168.1.109', 5084) (192.168.1.109:5084)
2019-02-25 16:17:47,575 sllurp.llrp: INFO: using reader mode: None
2019-02-25 16:17:47,936 sllurp.llrp: INFO: stopping politely
2019-02-25 16:17:47,949 sllurp.llrp: INFO: Impinj search mode? None
2019-02-25 16:17:47,949 sllurp.llrp: INFO: starting inventory
2019-02-25 16:17:47,949 sllurp.llrp: ERROR: argument should be bytes, buffer or ASCII string, not 'tuple'
Traceback (most recent call last):
  File "/Users/cooolallen/GoogleDrive/RFID/sllurp/venv/lib/python3.7/site-packages/sllurp/llrp.py", line 860, in send_ADD_ROSPEC
    'ROSpec': rospec,
[... 8 lines cut ...]
  File "/Users/cooolallen/GoogleDrive/RFID/sllurp/venv/lib/python3.7/site-packages/sllurp/llrp_proto.py", line 1632, in encode_ROSpec
    data += encode('AISpec')(par['AISpec'])
[... cut ...]
TypeError: argument should be bytes, buffer or ASCII string, not 'tuple'
^C2019-02-25 16:17:54,460 sllurp.llrp: INFO: stopping politely
2019-02-25 16:17:54,461 sllurp.llrp: INFO: will disconnect when stopped
2019-02-25 16:17:54,474 sllurp.llrp: INFO: lost connection: Connection to the other side was lost in a non-clean fashion: Connection lost.
2019-02-25 16:17:54,474 sllurp.verb.inventory: INFO: total # of tags seen: 0 (0 tags/second)
2019-02-25 16:17:54,474 sllurp.llrp: INFO: disconnecting
```

Am I do something wrong with the inventory?

Thanks!

ransford, 2019-02-27:
Oh, I see I didn't implement quite far enough. `llrp_proto.encode_C1G2TagInventoryMask()` needs to be updated further. I'll try to get to that soon, but if you're willing to take a crack at it, you might make faster progress.

ransford, 2019-03-01:
OK, try that branch again, please. I fixed a use of the wrong variable and added a test case.

Regina8023, 2019-03-01:
Hi ransford,

Thank you for your effort! I am working with cooolalle. When I tried using the new version, it seems that it works well in one tag and two tags. But when I put three tag-mask-filter in argument, I got the following error: 

```
(.venv) (base) Reginas-Mac:sllurp-feature-121-multi-filter Regina8023$ sllurp --debug inventory 192.168.1.109 --tag-filter-mask 00 --tag-filter-mask 30 --tag-filter-mask 00 --impinj-search-mode 2 -t 2
............
2019-03-01 16:41:47,282 sllurp.llrp: INFO: starting inventory
2019-03-01 16:41:47,282 sllurp.llrp: DEBUG: made enabled_rospec
2019-03-01 16:41:47,282 sllurp.llrp: DEBUG: made added_rospec
2019-03-01 16:41:47,282 sllurp.llrp: DEBUG: about to send_ADD_ROSPEC
2019-03-01 16:41:47,282 sllurp.llrp: DEBUG: serializing ADD_ROSPEC command
2019-03-01 16:41:47,282 sllurp.llrp_proto: DEBUG: encoding AntennaConfiguration: {'AntennaID': 1, 'RFTransmitter': {'HopTableId': 1, 'ChannelIndex': 1, 'TransmitPower': 81}, 'C1G2InventoryCommand': {'TagInventoryStateAware': False, 'C1G2SingulationControl': {'Session': 2, 'TagPopulation': 4, 'TagTransitTime': 0}, 'C1G2Filter': [{'C1G2TagInventoryMask': {'MB': 1, 'Pointer': 32, 'TagMask': '00'}}, {'C1G2TagInventoryMask': {'MB': 1, 'Pointer': 32, 'TagMask': '30'}}, {'C1G2TagInventoryMask': {'MB': 1, 'Pointer': 32, 'TagMask': '00'}}], 'ImpinjInventorySearchModeParameter': 2}}
2019-03-01 16:41:47,283 sllurp.llrp: DEBUG: serialized bytes: b'0414000000a60000000800b1009c00000001000000b2001200b300050100b6000901000007d000b700730001000100b8000901000007d000ba006200010100de005b000100e0000a000100010051014a004b00014b000f00014c000a400020000800014b000f00014c000a400020000830014b000f00014c000a4000200008000150000b8000040000000003ff000e0000651a00000017000200ed000d01000000ee00060980'
2019-03-01 16:41:47,283 sllurp.llrp: DEBUG: done serializing ADD_ROSPEC command
2019-03-01 16:41:47,283 sllurp.llrp: DEBUG: sent ADD_ROSPEC
2019-03-01 16:41:47,283 sllurp.llrp: DEBUG: state change: STATE_CONNECTED -> STATE_SENT_ADD_ROSPEC
2019-03-01 16:41:47,285 sllurp.llrp: DEBUG: got 102 bytes from reader: b'041e0000006600000008011f005c006900244c4c5250205b3339385d203a202f2f4331473246696c746572203a206f766572666c6f770121003000b100c80121002800b700c80121002000ba00c80121001800de00c801210010014a00c801210008014b00cd'
2019-03-01 16:41:47,285 sllurp.llrp: DEBUG: expect 102 bytes (have 102)
2019-03-01 16:41:47,285 sllurp.llrp: DEBUG: deserializing ADD_ROSPEC_RESPONSE command
2019-03-01 16:41:47,286 sllurp.llrp_proto: DEBUG: decode_AddROSpecResponse
2019-03-01 16:41:47,288 sllurp.llrp_proto: DEBUG: decode_LLRPStatus
2019-03-01 16:41:47,288 sllurp.llrp_proto: DEBUG: decode_LLRPStatus: b'011f005c006900244c4c5250205b3339385d203a202f2f4331473246696c746572203a206f766572666c6f770121003000b100c80121002800b700c80121002000ba00c80121001800de00c801210010014a00c801210008014b00cd'
2019-03-01 16:41:47,290 sllurp.llrp_proto: DEBUG: decode_LLRPStatus (type=287 len=92)
2019-03-01 16:41:47,291 sllurp.llrp_proto: DEBUG: decode_FieldError
2019-03-01 16:41:47,291 sllurp.llrp_proto: DEBUG: no FieldError
2019-03-01 16:41:47,293 sllurp.llrp_proto: DEBUG: decode_ParameterError
2019-03-01 16:41:47,294 sllurp.llrp_proto: DEBUG: decode_ParameterError (type=289 len=48 data=b'\x00\xb1\x00\xc8\x01!\x00(\x00\xb7\x00\xc8\x01!\x00 \x00\xba\x00\xc8\x01!\x00\x18\x00\xde\x00\xc8\x01!\x00\x10\x01J\x00\xc8\x01!\x00\x08\x01K\x00\xcd')
2019-03-01 16:41:47,296 sllurp.llrp_proto: DEBUG: decode_FieldError
2019-03-01 16:41:47,297 sllurp.llrp_proto: DEBUG: decode_ParameterError
2019-03-01 16:41:47,299 sllurp.llrp_proto: DEBUG: decode_ParameterError (type=289 len=40 data=b'\x00\xb7\x00\xc8\x01!\x00 \x00\xba\x00\xc8\x01!\x00\x18\x00\xde\x00\xc8\x01!\x00\x10\x01J\x00\xc8\x01!\x00\x08\x01K\x00\xcd')
2019-03-01 16:41:47,302 sllurp.llrp_proto: DEBUG: decode_FieldError
2019-03-01 16:41:47,306 sllurp.llrp_proto: DEBUG: decode_ParameterError
2019-03-01 16:41:47,308 sllurp.llrp_proto: DEBUG: decode_ParameterError (type=289 len=32 data=b'\x00\xba\x00\xc8\x01!\x00\x18\x00\xde\x00\xc8\x01!\x00\x10\x01J\x00\xc8\x01!\x00\x08\x01K\x00\xcd')
2019-03-01 16:41:47,310 sllurp.llrp_proto: DEBUG: decode_FieldError
2019-03-01 16:41:47,312 sllurp.llrp_proto: DEBUG: decode_ParameterError
2019-03-01 16:41:47,313 sllurp.llrp_proto: DEBUG: decode_ParameterError (type=289 len=24 data=b'\x00\xde\x00\xc8\x01!\x00\x10\x01J\x00\xc8\x01!\x00\x08\x01K\x00\xcd')
2019-03-01 16:41:47,315 sllurp.llrp_proto: DEBUG: decode_FieldError
2019-03-01 16:41:47,317 sllurp.llrp_proto: DEBUG: decode_ParameterError
2019-03-01 16:41:47,319 sllurp.llrp_proto: DEBUG: decode_ParameterError (type=289 len=16 data=b'\x01J\x00\xc8\x01!\x00\x08\x01K\x00\xcd')
2019-03-01 16:41:47,321 sllurp.llrp_proto: DEBUG: decode_FieldError
2019-03-01 16:41:47,323 sllurp.llrp_proto: DEBUG: decode_ParameterError
2019-03-01 16:41:47,327 sllurp.llrp_proto: DEBUG: decode_ParameterError (type=289 len=8 data=b'\x01K\x00\xcd')
2019-03-01 16:41:47,329 sllurp.llrp_proto: DEBUG: decode_FieldError
2019-03-01 16:41:47,331 sllurp.llrp_proto: DEBUG: decode_ParameterError
2019-03-01 16:41:47,331 sllurp.llrp: DEBUG: done deserializing ADD_ROSPEC_RESPONSE command
2019-03-01 16:41:47,331 sllurp.llrp: DEBUG: LLRPMessage received in state 15: <ADD_ROSPEC_RESPONSE>
	<Ver>1</Ver>
	<Type>30</Type>
	<ID>8</ID>
	<LLRPStatus>
		<StatusCode>'OverflowParameter'</StatusCode>
		<ErrorDescription>b'LLRP [398] : //C1G2Filter : overflow'</ErrorDescription>
		<ParameterError>
			<ParameterType>177</ParameterType>
[... cut ...]
2019-03-01 16:41:47,331 sllurp.llrp: DEBUG: in handleMessage(ADD_ROSPEC_RESPONSE), there are 1 Deferreds
2019-03-01 16:41:47,331 sllurp.llrp: CRITICAL: Error OverflowParameter adding ROSpec: b'LLRP [398] : //C1G2Filter : overflow'
```

From the error, it seems the argument parsing is correct and the message is sent to reader. But when deserialize message from reader, error occurs. Cooolallen and I guess this is the problem of reader: maybe it cannot deal with so many settings? 

What do you think about it? 

Thank you!

ransford, 2019-03-02:
Yes, the word "overflow" in your message certainly makes it seem likely that the reader can't handle so many filters. Maybe it would work to send the reader multiple AISpecs, each with a different set of filters (up to, it seems, 2). Pull request for that welcome! :-)

ransford, 2019-03-08:
It looks like the multiple-filter code worked as intended, so I'm going to merge #123 and close this issue. I believe @Regina8023's issue is related to a limit that the reader imposes.

PR #123 (2019-03-08): Multiple --tag-filter-mask params
... probably by turning each into a `C1G2Inventory > C1G2Filter` parameter.

Fixes #121
closed 2019-03-08 by ransford