sllurp/sllurp #122: Not aggregate data in tag report and query tag report from reader to increase read rate

cooolallen, 2019-02-24:
Hi Ransford, thanks for your support so that we can play around with this. 

I am wondering if there a way we can return a multiple seen count without aggregating data? For example, instead of PeakRSSI, I want to get RSSIs as an array so that we can do further analysis. To increase reading rate, I prefer not setting --report-every-n-tags to 1. 

Or is there a way we can pull the tag reports from the reader rather than the reader push the report back? I know there is some functionality in C# like reader.querytag() and ReportMode.WaitForQuery. I assume this function will increase the read rate and it is more intuitive that user query tag report from the reader. Perhaps it already exists but I missed it. But I think it is great to have this function here! 

Thanks

ransford, 2019-02-24:
I think I know what you mean. Check out section 14.2.3 (p. 85) of the LLRP spec in `docs/llrp_1_1-standard-20101013.pdf` to read about `TagReportData` and you'll see what the constraints are at the protocol level.

I don't know anything about `reader.querytag()`... is that manufacturer specific? More information on whatever mechanisms that's using would be helpful. Same for `ReportMode.WaitForQuery`.

cooolallen, 2019-02-25:
Thanks for giving the guide to read the LLRP spec.
However, I didn't get much information from that document. Are you trying to say that it is possible that this cannot be achieved because of the constraints at the protocol level?

I will try to gather more information with `reader.querytag()` and `ReportMode.WaitForQuery` for you and try to figure out how they do that.

Thank you

thijmenketel, 2019-03-08:
So at the moment the low read rate has a couple of different causes: 
 - The current version of sllurp on the develop branch is using stack inspection for debug purposes.
 - The report_every_n_tags option causes the reader to send a tcp frame for every tag report it receives, this slows down the inventory operation. 

Take a look at issue #106 where a lot of this is talked about. Furthermore, pr #117 has a fix for the stack inspection but reworks most of the code to make it happen. I'm currently using a version that @fviard created with a different report trigger (see [line](https://github.com/ransford/sllurp/blob/5d89870aeb04ca11f82d786c0203efd65c4e7002/sllurp/llrp_proto.py#L3952) and [comment](https://github.com/ransford/sllurp/issues/106#issuecomment-418732661)).

closed 2021-01-24 by ransford