sllurp/sllurp #124: sllurp does not allow reader_mode 5

djacobson, 2019-03-06:
Not a huge priority but, we have been testing all of an Impinj's reader modes as listed on page 19 of the Impinj Octane LLRP Guide ( https://support.impinj.com/hc/article_attachments/360000792260/Impinj_Octane_LLRP.pdf ) ... and, yep, there's a reader mode = 5 ("Dense Reader 2 M=4") we were kind of curious about testing. Below is the log error when I attempt to set that using sllurp:

2019-03-06 11:44:28.643 [sllurp.llrp]: ERROR: Capabilities mismatch
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.5/site-packages/sllurp/llrp.py", line 321, in parseCapabilities
    if mo['ModeIdentifier'] == self.mode_identifier][0]
IndexError: list index out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.5/site-packages/sllurp/llrp.py", line 469, in handleMessage
    self.parseCapabilities(self.capabilities)
  File "/home/pi/.local/lib/python3.5/site-packages/sllurp/llrp.py", line 327, in parseCapabilities
    raise ReaderConfigurationError(errstr)
sllurp.llrp_errors.ReaderConfigurationError: Invalid mode_identifier; valid mode_identifiers are [0, 1, 2, 3, 4, 1000, 1002, 1003, 1004]
2019-03-06 11:44:28.644 [sllurp.llrp]: ERROR: Failed to decode LLRPMessage; will not decode 2010 remaining bytes
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.5/site-packages/sllurp/llrp.py", line 321, in parseCapabilities
    if mo['ModeIdentifier'] == self.mode_identifier][0]
IndexError: list index out of range

thijmenketel, 2019-03-08:
Okay, so this means that your reader doesn't support mode 5 as the mode table is taken from the reader capabilities response itself. Depending on which version of the firmware you're running it's supported or not (the document you attached is specifically for octane 5.14). When you go the the homepage of the reader in the browser the firmware version is also shown.

djacobson, 2019-03-08:
@thijmenketel I didn't know this... it looked like a static list to me. I will def read this section closer. I do know that our R420 has been updated to the latest firmware version. Thank you. I'll confirm and close this issue.

closed 2019-03-11 by djacobson