sllurp/sllurp #125: Sllurp reader settings not matching what the Impinj reader shows

djacobson, 2019-03-23:
I am using sllurp to set the following reader settings on my Impinj R420:

Reader Mode = 1003
Search Mode = 5
Session = 1
Tag Population = 100

...and I believe the following tagreader.log log lines show that the settings were successfully made:

```
2019-03-23 PM 12:18:28.050 [sllurp.llrp]: INFO: connecting to speedwayr-10-3f-a8:5084...
2019-03-23 PM 12:18:29.687 [sllurp.llrp]: INFO: will reset reader state on connect
2019-03-23 PM 12:18:29.688 [sllurp.llrp]: INFO: will start inventory on connect
2019-03-23 PM 12:18:29.689 [sllurp.llrp]: INFO: Enabling Impinj extensions
2019-03-23 PM 12:18:29.689 [sllurp.llrp]: INFO: using antennas: [1, 2, 3, 4]
2019-03-23 PM 12:18:29.691 [sllurp.llrp]: INFO: transmit power: {1: 0, 2: 0, 3: 0, 4: 0}
2019-03-23 PM 12:18:29.694 [sllurp.llrp]: INFO: connected to ('speedwayr-10-3f-a8', 5084) (10.0.0.2:5084)
2019-03-23 PM 12:18:31.349 [sllurp.llrp]: INFO: using reader mode: {'PIE': 1500, 'R': 0, 'MaxTari': 6250, 'C': 0, 'Mod': 0, 'BDR': 40000, 'ModeIdentifier': 1003, 'M': 0, 'MinTari': 6250, 'StepTari': 0, 'FLM': 0}
2019-03-23 PM 12:18:31.433 [sllurp.llrp]: INFO: stopping politely
2019-03-23 PM 12:18:31.504 [sllurp.llrp]: INFO: Impinj search mode? 5
2019-03-23 PM 12:18:31.505 [sllurp.llrp_proto]: INFO: impinj_search_mode: 5
2019-03-23 PM 12:18:31.506 [sllurp.llrp_proto]: INFO: impinj_search_mode: 5
2019-03-23 PM 12:18:31.506 [sllurp.llrp_proto]: INFO: impinj_search_mode: 5
2019-03-23 PM 12:18:31.506 [sllurp.llrp_proto]: INFO: impinj_search_mode: 5
2019-03-23 PM 12:18:31.507 [sllurp.llrp_proto]: INFO: will report every ~N=1 tags
2019-03-23 PM 12:18:31.507 [sllurp.llrp]: INFO: starting inventory
```

*BUT*, when I ssh into the Impnj reader while my app is running and run the following command:

`show rfid llrp config`

...I see the following sections in the XML output, which does not reflect my settings (in fact it never changes). Am I missing something here, interpreting this command wrong? Maybe something is getting reset when I ssh into the reader?

```
...
    <C1G2InventoryCommand>
      <TagInventoryStateAware>0</TagInventoryStateAware>
      <C1G2RFControl>
        <ModeIndex>1000</ModeIndex>
        <Tari>0</Tari>
      </C1G2RFControl>
      <C1G2SingulationControl>
        <Session>1</Session>
        <TagPopulation>32</TagPopulation>
        <TagTransitTime>0</TagTransitTime>
      </C1G2SingulationControl>
      <Impinj:ImpinjInventorySearchMode>
        <Impinj:InventorySearchMode>Reader_Selected</Impinj:InventorySearchMode>
      </Impinj:ImpinjInventorySearchMode>
...
```

djacobson, 2019-03-23:
Below is my LLRP client object init. I've confirmed the values of the params:

```
			fac = LLRPClientFactory(
				report_every_n_tags=configParams['report_every_n_tags'],
				antennas=configParams['antennas'],
				tx_power=configParams['tx_power'],
				# modulation=configParams['modulation'],
				mode_identifier=configParams['mode_identifier'],
				start_inventory=configParams['start_inventory'],
				tari=configParams['tari'],
				session=configParams['session'],
				tag_population=configParams['tag_population'],
				impinj_search_mode=configParams['impinj_search_mode'],
				tag_content_selector={
					'EnableROSpecID': True,
					'EnableSpecIndex': True,
					'EnableInventoryParameterSpecID': True,
					'EnableAntennaID': True,
					'EnableChannelIndex': True,
					'EnablePeakRRSI': True,
					'EnableFirstSeenTimestamp': True,
					'EnableLastSeenTimestamp': True,
					'EnableTagSeenCount': True,
					'EnableAccessSpecID': True,
				},
				# impinj_tag_content_selector={
				# 	'EnableImpinjExtensions': True,  # Not implemented yet
				# 	'EnablePeakRRSI': True,  # Redundant?
				# },
				reconnect=configParams['reconnect']
			)
```

djacobson, 2019-03-23:
Just an update, I set `logger.setLevel(logging.DEBUG)` and the config response I got back in the log file correctly matched my settings. So, it just seems the Impinj's interactive 'show rfid llrp config` command disagrees / does not report the config that the llrp client is using... ? I'm really puzzled here. It **seems** my settings are taking effect due to the reader's behavior.

thijmenketel, 2019-03-26:
So at the moment Sllurp doesn't support other search modes than single target or dual target (see cli.py, impinj search mode). 

The tag population part seems weird indeed, I have no explanation for that!

djacobson, 2019-03-26:
@thijmenketel Thank you for pointing that out. I was looking at llrp.py and it was not clear to me that it was restricted to just those 2 values; I thought it was just passing that value through to the API. But cli.py parse args makes that a little more clear, although maybe it should be logged as an error instead of INFO?

Also, any insight into why the reader mode would not match? Could it be that all settings are being discarded due to one being erroneous? Would you say sllurp is defaulting the Impinj reader to search_mode = 1 ? We have been testing by setting different reader modes and getting significant behavior changes; I'm assuming our search mode has been defaulting to something when we've used a value other than 1 or 2?

Thank you for your insight!

djacobson, 2019-04-03:
I'm going to close this issue because it seems to be an Impinj specific peculiarity, not necessarily based on the sllurp lib. If anyone does have any insight or can confirm / reproduce this diff on their Impinj reader + sllurp combo, please let me know.

closed 2019-04-03 by djacobson