sllurp/sllurp #126: EPC on tags read by sllurp doesn't match with EPC read by Impinj itemtest or Impinj multireader software

iadv, 2019-03-27:
Today I noticed that EPC read by sllurp doesn't match with that read by the Impinj software itemtest and multi-reader. I am using allien squiggle tags[](url). I have no idea how this can happen. There is no relation/pattern between the two EPC's shown. Also, I have just one tag in the vicinity, but the output shows there two tags with slightly different EPCs. This makes me think that there is a bug in the decoder function in sllurp. Does anyone else have the same problem? Any solution?
[image]
[image]

ransford, 2019-03-27:
I have never seen that happen; we're really not doing anything fancy to decode TagReport messages. Can you post a debug-level log (see this project's top-level README for [instructions](https://github.com/ransford/sllurp/#if-you-find-a-bug)) and the sllurp command you're using?

thijmenketel, 2019-03-27:
That seems kind of weird, I also have Alien squiggle tags here and the EPC is similar to yours (in the bottom picture). Which value is Sllurp reporting (which picture is sllurp and which is ItemTest)? 

Also, the values in the top picture don't really look like hex values, are you doing some converting?

iadv, 2019-03-27:
@thijmenketel The top picture is sllurp. Bottom picture is Multi-reader software by Impinj (they match with itemtest software). No I am not doing anything with conversion at all. Also, it's weird if you see the top picture has two EPCs in the list, where as I have just one tag. Something is not right with decoding EPC bits. I will post the debug-level log as ransford asked now

iadv, 2019-03-27:
[github.txt](https://github.com/ransford/sllurp/files/3013541/github.txt)

@ransford here is the debug file. I notice some "failed to decode EPCData" messages, but the code automatically rectifies it by getting EPC-96 automatically. Is this probably the issue?

thijmenketel, 2019-03-27:
Hmmm that's weird indeed.. I have no issues on my side with the same code. From the debug messages I can see that there are way more bytes coming in then Sllurp is expecting, so I think something might be wrong there.

iadv, 2019-03-27:
@thijmenketel yes to speedup the read rate, but I do not notice much difference. However, changing the --mode -identifier to 4 gives maximum speed

thijmenketel, 2019-03-27:
So this setting has some issues:
 - Sllurp currently does a lot of stack inspections for debug reasons (even if debugging is turned off, see issue #106). This means that the code is slow and cannot handle very high throughput.
 - When sllurp can’t handle throughput, weird things happen. 

Furthermore, mode 4 is always slower than mode 0.

I tried with multiple versions of sllurp with about a 100 tags, no problems. 

I would suggest to change the N value back to 0 and try again (try multiple modes). If you want to get high throughput (which I need) I would suggest to use the [untwisted](https://github.com/ransford/sllurp/tree/fviard-develop_untwisted?files=1) version with N value 1, this one has a different implementation for debug reporting and will be much faster (I get over 1000 reads per second on multiple tags).

djacobson, 2019-03-29:
@iadv I'm just curious how your are parsing your tag data. Below is how I'm doing it in my custom app (Note: not a sllurp verb example):

```
...
rawtagReport = llrpMsg.msgdict['RO_ACCESS_REPORT']
...

...
tagReport = rawtagReport
tagData = tagReport['TagReportData']
if tagData:

for tagdict in tagData:
	for tagelem_key in tagdict.keys():
		tkey = tagelem_key.lower()
		tvalue = tagdict[tagelem_key]

		if tkey == "epcdata":
			epcdata = tvalue['EPC']
			if epcdata:
				tagNumber = int(epcdata[4:], 16)  # Base 16 (hexadecimal)
				...

		elif tkey == "epc-96":
			epc = tvalue
			tagNumber = int(epc[4:], 16)  # Base 16 (hexadecimal)

		elif tkey == "firstseentimestamputc":
			(firstSeenTimestampUTC,) = tvalue

		elif tkey == "antennaid":
			(antennaID,) = tvalue

		elif tkey == "peakrssi":
			(peakRSSI,) = tvalue

		...
```

thijmenketel, 2019-04-02:
@iadv Any progress on this issue?

iadv, 2019-04-02:
@thijmenketel No. The EPC is still off from the actual values. I took a short cut method right now as I am more interested in processing the phase information. I will look into the EPC thing after phase is figured out. 
@djacobson I am using the sllurp verb files. I haven't changed anything related to parsing yet.

closed 2021-01-24 by ransford