sllurp/sllurp #127: Printing frequency hop table

iadv, 2019-03-27:
One thing I am curious to implement in sllurp is making frequency information available to the user in addition to the channel index. In FCC region, the frequency table is randomly created as opposed to what I earlier thought to be as channel index = o => f = 902.25 MHz. I was wondering if there is any way to print the frequency table every time the program is run. I can see the able in the debug mode, however, it is not efficient to run in debug every time someone is interested in phase. If we can print the frequency table, I think the phase can be adjusted to the frequency hopping effects. 
I can help implement this however I am not sure how to access this in the inventory.py file. 

For example, tagreportdata can be accessed by using llrp_msg.msgdict['RO_ACCESS_REPORT']['TagReportData']. What is the analogous thing for 
['FrequencyInformation] or ['RegulatoryCapabilities']? Is it capdict['RegulatoryCapabilities']? I looked at the whole debug code, but cannot interpret it. I want to be able to call capdict in the inventory.py.

iadv, 2019-03-28:
I got the frequency hop table values. However, even after adjusting phase for different channels, the phase isn't consistent. For example, if I get phase 1 and phase2 corresponding to channel 1 and channel 2 ,respectively, I am calculating the phase 2_new in the same frequency reference as phase 1. But phase 1 and phase2_new are not even close. I am not sure why the phase (after correcting for hopping frequency) isn't same (not even within some error range).

thijmenketel, 2019-03-28:
I see you found the way to get the hopping table from sllurp, little bit of a hack but it works.

Now frequency hopping. So this is problem that I have tried to tackle as well, what is your approach for compensating? An example approach I've tried is described in [this paper[(https://twei7.com/papers/TWei_MobiCom16_Tagyro.pdf) (section 4.5), however this approach wasn't really very accurate when the tag starts moving. For now I just use phase readings per single channel and normalise them, this doesn't remove the hopping though. Furthermore, the time for the reader to change channel is about 10 ms, which is quite a lot if you're running at higher read rates.

If you have a more modern version of the impinj readers you might be able to change the region, which allows you to select the hopping table/operating frequency (see PR #112).

iadv, 2019-03-28:
Two comments - 
1) Yes I did the same as the one in the paper by multiplying and dividing the phase with corresponding frequencies. However, I am not sure about the offset term they added in the equation. I will read it properly in some time
2) See the below pictures, one is from the octane implementation and the other is from the sllurp. I sorted the phase values from the octane implementation as per channel indexes and plotted them. Here I took the channel indexes as per impinj's documentation. Channel 1 is 902.25 and increments so on. The result is as expected: a continuous change in phase. However, when I plot the phase readings from sllurp and sorting them I get random plot. Here, I tried taking channel index as per the above as well as as per the frequency hop table. I think this should match the readings from octane if it were correct. So I am not sure how to get the corresponding frequency for a channel index shown in SLLURP. Any thoughts on how to get the corresponding frequency for channel indexes?
[image]
[image]

thijmenketel, 2019-03-28:
1. Yeah, the paper only supplies this as a possible solution for the frequency hopping, in their actual implementation they fix the frequency to a single channel. In equation 11 the general mathematics is shown, however I still keep some errors unfortunately.

2. So the capabilities command will show two tables: the hop-table and the frequency list. I did this by adding ```self.hoptable = capdict['RegulatoryCapabilities']['UHFBandCapabilities']['FrequencyInformation']['FrequencyHopTable0']``` on line 362 (roughly) in llrp.py and then after the inventory starts I get the table from ```llrp.hoptable``` (for the Impinj r420 this works, but it is hacky). Looking at the graphs I think you just have the frequency list which is indeed a linear list.

fviard, 2019-03-28:
@iadv do you use N=1 and have a count of 1 per tag read?
Because there is a little trick if you get "accumulated" tag reads from the reader that the reported frequency channel will just be the one when the tag was the first seen but the other values will be of the last observation.

iadv, 2019-03-28:
@fviard The default is 1 right? In the inventory.py we have ## parser.add_argument('-n', '--report-every-n-tags', default=1, type=int, dest='every_n', metavar='N', help='issue a TagReport every N tags')##

So I shouldn't have that issue

iadv, 2019-03-28:
@thijmenketel I add the print command to manually cross-check in the same line: print(capdict['RegulatoryCapabilities']['UHFBandCapabilities']['FrequencyInformation']['FrequencyHopTable0'])

I get single table not two tables like you mentioned: What I get is here below - 
{u'Frequency32': 916750, u'Frequency33': 911250, u'HopTableId': 1, u'Frequency28': 913750, u'Frequency27': 926750, u'Frequency26': 912250, u'Frequency25': 908750, u'Frequency24': 907750, u'Frequency23': 923750, u'Frequency22': 911750, u'Frequency21': 920750, u'Frequency20': 924750, u'Frequency45': 922750, u'Frequency44': 914750, u'Frequency47': 909750, u'Frequency46': 906250, u'Frequency41': 902750, u'Frequency40': 924250, u'Frequency43': 915250, u'Frequency42': 905750, u'Frequency49': 921250, u'Frequency48': 905250, u'Frequency29': 918750, u'Frequency4': 925750, u'Frequency5': 907250, u'Frequency6': 921750, u'Frequency7': 904750, u'Frequency1': 920250, u'Frequency2': 916250, u'Frequency3': 903250, u'Frequency38': 918250, u'Frequency8': 917250, u'Frequency9': 923250, u'Frequency39': 919250, u'Frequency16': 909250, u'Frequency17': 914250, u'Frequency14': 910250, u'Frequency15': 919750, u'Frequency12': 915750, u'Frequency13': 917750, u'Frequency10': 910750, u'Frequency11': 922250, u'Frequency34': 908250, u'Frequency35': 926250, u'Frequency36': 906750, u'Frequency37': 912750, u'Frequency30': 903750, u'Frequency31': 927250, u'Frequency18': 925250, u'Frequency19': 904250, u'NumHops': 50, u'Frequency50': 913250}

_

Do you get something like this? I the graphs I use these as frequencies for corresponding channel indexes

iadv, 2019-03-28:
While investigating this a little further into this, I am thinking may be I am reading the channel index versus frequency wrong. There are three scenarios: 
a) Hoptable is printed as in the above post starting from Freq 32 to Freq 50. So here the channel index 1 is the u'Frequency32':
b) The channelindex is printed as in the debug file. Copied below - 
u'UHFBandCapabilities': {u'FrequencyInformation': {u'FrequencyHopTable0': {u'Frequency1': 923250,
                                                                                                        u'Frequency10': 918250,
                                                                                                        u'Frequency11': 917250,
                                                                                                        u'Frequency12': 914750,
                                                                                                        u'Frequency13': 912750,
                                                                                                        u'Frequency14': 908250,
                                                                                                        u'Frequency15': 906250,
                                                                                                        u'Frequency16': 908750,
                                                                                                        u'Frequency17': 911250,
                                                                                                        u'Frequency18': 910750,
                                                                                                        u'Frequency19': 907750,
                                                                                                        u'Frequency2': 927250,
                                                                                                        u'Frequency20': 924750,
                                                                                                        u'Frequency21': 913250,
                                                                                                        u'Frequency22': 904250,
                                                                                                        u'Frequency23': 909250,
                                                                                                        u'Frequency24': 924250,
                                                                                                        u'Frequency25': 907250,
                                                                                                        u'Frequency26': 925750,
                                                                                                        u'Frequency27': 905750,
                                                                                                        u'Frequency28': 918750,
                                                                                                        u'Frequency29': 915750,
                                                                                                        u'Frequency3': 920250,
                                                                                                        u'Frequency30': 905250,
                                                                                                        u'Frequency31': 915250,
                                                                                                        u'Frequency32': 912250,
                                                                                                        u'Frequency33': 919250,
                                                                                                        u'Frequency34': 921750,
                                                                                                        u'Frequency35': 906750,
                                                                                                        u'Frequency36': 913750,
                                                                                                        u'Frequency37': 922250,
                                                                                                        u'Frequency38': 916250,
                                                                                                        u'Frequency39': 910250,
                                                                                                        u'Frequency4': 926750,
                                                                                                        u'Frequency40': 911750,
                                                                                                        u'Frequency41': 909750,
                                                                                                        u'Frequency42': 902750,
                                                                                                        u'Frequency43': 903750,
                                                                                                        u'Frequency44': 904750,
                                                                                                        u'Frequency45': 926250,
[... cut ...]
                                                                                                        u'NumHops': 50},
                                                                                u'Hopping': True},
Here the Channel index 1 is the one with u'Frequency1' and channel index 2 is u'Frequency2' and so on
3) seems more straightforward - 
channel index 1 is u'Frequency1' and Channel index 2 is u'Frequency2' and Channel index 50 is u'Frequency50'
What should be the correct way of interpreting the channel index frequency? I think case 3, but the plots are not getting better

thijmenketel, 2019-04-02:
Sorry for the late reply.

Your assumption that Frequency1 is channel 1 etc. is correct. It is weird that the plots are not correct. Any luck so far with compensating for the hopping?

iadv, 2019-04-02:
Yes, I did a small experiment to confirm the channel versus frequency between the sllurp library's frequency hop table and using my earlier octane implementation in Eclipse. 
1. Yes, channel 1's frequency corresponds to the frequency against u'frequency1' in the table
2. I found an online documentation from impinj which lists the channel (all 50 channels) v/s frequency. Which is then wrong
3. Impinj reader adds 180 deg of random phase ambiguity. So I took the 0-2pi scale phase data and converted to 0-pi. 
4. Reader then adds phase offset at each frequency so a simple compensation wouldn't work
5. Data so far is very noisy and I am trying techniques to filter the data. Will update on this soon

thijmenketel, 2019-04-03:
1. Correct.

2. Well its not really wrong, but it just lists the possible frequencies allowed in FCC regions. However, the protocol dictates that on startup of the reader, the hoptable is to be randomized, which happens on the reader and on which we have no control (in FCC regions at least, ETSI allows for fixed frequency or preset frequencies).

3. I found that this ambiguity only happens when Miller encoding is used, FM0 doesn't seem to produce this ambiguity. 

4. My current solution for compensating for the hopping is either just measuring the shift and subsequently subtracting it or normalising per channel (which is not the nicest way but works for my research). 

5. Try a median filter, worked for me. The noise is dependent on the environment that you're using the reader and the reflection characteristics of the tag. The Alien Squiggle tags that you're using are very sensitive, which is good for range but not for noise. Monza tags tend to be less noisy.

thijmenketel, 2019-04-11:
@iadv Any progress?

closed 2021-01-24 by ransford