sllurp/sllurp #128: Support for other Impinj Search Modes

djacobson, 2019-04-03:
Currently Sllurp only appears to support Search Modes 1 & 2 (Single and Dual Target). There are three more Search Modes available on Impinj readers. Would it be a difficult change to allow the sllurp lib to optionally set these modes?

https://support.impinj.com/hc/en-us/articles/202756158-Understanding-EPC-Gen2-Search-Modes-and-Sessions

ransford, 2019-04-03:
I just took a quick look and it appears that you should be able to do this with one change: in `sllurp/cli.py`, change this:

```python
@click.option('--impinj-search-mode', type=click.Choice(['1', '2']),
              help=('Impinj extension: inventory search mode '
                    ' (1=single, 2=dual)'))
```

... by adding to that `click.Choice` array and documenting your additions in the `help` string. Let us know how it goes!

ransford, 2019-04-03:
If you chase that `impinj_search_mode` argument all the way to `llrp_proto.py`, you'll see that whatever you provide in that command-line argument is passed (after conversion to integer) directly to `C1G2InventoryCommand > ImpinjInventorySearchModeParameter`. That's why I suggest it should be pretty straightforward.

djacobson, 2019-04-04:
@ransford Thank you for confirming! I used to do exactly that, but, one of the reasons I submitted this was because of some odd behavior we were seeing in our tests and the sllurp tagreader log (see closed issue [https://github.com/ransford/sllurp/issues/125](125) ... and @thijmenketel 's comment on that issue. Isn't that repeated log line odd? I will test this more...

ransford, 2019-04-04:
Oh, I see! I don't know to what extent the reader tries to DWIM. With all the various settings and parameters, the space of possible configurations is so huge that it's *possible* it tries to provide failsafe settings in the case of certain misconfigurations? OTOH, these readers are pretty good at providing a helpful error message when there really is an error.

I trust `show rfid llrp config` output to be the "ground truth" for reader settings. One handy trick is to dump XML output to a file, then `diff` among these files while fiddling with different behaviors.

```shell
$ ssh <reader> show rfid llrp config > config.xml
$ ssh <reader> show rfid llrp rospec 0 > rospec0.xml
...
```

I have an R420 in case you want me to test a specific piece of Python code...

djacobson, 2019-04-04:
@ransford One simple thing I would love if you could confirm for me on your R420; if you set the following settings in your sllurp sample (traditional valid settings according to the sllurp demo):
```
_DEFAULT_READER_MODE = 4  # 4 = Max Miller, 1003 = AutoSet Static Fast
_DEFAULT_IMPINJ_SEARCH_MODE = 2
_DEFAULT_SESSION_MODE = 1  # Session 1 = Search mode 'single target with suppression'
```
...does your:
```
$ ssh <reader> show rfid llrp config > config.xml
```
...command match the settings you just set? And then, after that, if you change the reader mode to 1003?
```
_DEFAULT_READER_MODE = 1003
```
My R420 seems to be stuck on that:
```
<ModeIndex>1000</ModeIndex>
```
...setting, according to the `show rfid llrp config` command???

tanderl, 2019-04-04:
I agree that the `show rfid llrp config` should be the config that the reader is actually using, so it appears that it's ignoring the `SET_READER_CONFIG` command.  I glanced at `llrp.py` and it appears that if the reader responded with something other than `Success` that this would show up in the log, is my understanding correct?

Something else you might try is looking at the logs on the reader itself.  I think you should be able to:

    > show logging events app 20

To get the last 20 log entries on the reader.  I think you should also be able to adjust the logging verbosity with:

    > config logging rfid debug

For reference, I got those from the "Impinj RShell Reference Manual" that should be downloadable from https://support.impinj.com/hc/en-us/articles/202755298-Reader-Documentation.

closed 2019-09-05 by djacobson