sllurp/sllurp #129: Default mode

iadv, 2019-06-06:
Just a quick question - 

What is the default mode while using inventory if any mode-identifier is not specified

ransford, 2019-06-07:
`LLRPROSpec.__init__()` omits "reader mode" from its serialized message if the user hasn't provided a mode identifier. So the behavior is whatever your reader decides to do when you give it an ROSpec with no mode identifier included — whatever default mode the reader chooses.

Mode selection is complicated because the reader has to announce the modes it supports; see `LLRPClient.parseCapabilities()` for the subtle details.

iadv, 2019-06-07:
That's an interesting situation. I would like to do study on how the mode impacts read range and tag detectability. Yes, for Impinj reader you could use something like 0,1,2,3,1001... (one can check with the command -mode-identifier --help). I was wondering if you or someone knows what is that "default reader mode" for impinj Rev 420?

ransford, 2019-06-07:
Probably 0, but I'm not sure. If you look at section 17.3.1.2.1 of `docs/llrp_1_1-standard-20101013.pdf` in this repo, you'll see that `C1G2RFControl` (the parent of `ModeIndex`) is an optional parameter, and section 16.2.1.2.1 just says "if absent, use default."

Here's something you could do to find the default your reader uses: comment out the part of the `LLRPROSpec` constructor that adds the `C1G2RFControl` bits so that this parameter isn't sent to the reader as part of the ROSpec, then start `sllurp inventory`, and while that's running, ssh to the reader and run `show rfid llrp rospec 0` and see if the ModeIndex is listed there. *Maybe* that will help?

ransford, 2019-06-07:
I wonder if @tanderl knows the answer to the above question?

tanderl, 2019-06-07:
I believe the default is mode 1000 as described in table 8.1 on page 151 of the "Octane LLRP Guide" available from https://support.impinj.com/hc/en-us/articles/202755298-Reader-Documentation.  Note that this is an "Autoset" mode so the reader will adjust link parameters as it sees fit based on how it senses the environment.  If you want modes with fixed values for parameters such as Tari, PIE, BLF, etc, you will want to stick with the non-AutoSet modes.  See table 3.2 on page 19 of that same document.

closed 2021-01-24 by ransford