sllurp/sllurp #134: Trouble reading TID

romansapia, 2019-08-29:
Hello!

I'm developing my own app, but as the problem is exactly the same using the command line I'll use it as example. I just need to read TID, so I run this:

`sllurp access -X 1 -mb 2 -r 6  <my reader ip>`

I got the tag data, but so many times and so fast that looks like some buffer is filling faster that it empties. I intuit that because longer I leave the TAG in range, more time I'll still receive tag data even when the tag is out of range. 

As I said I'm developing my own app, so I'm not restricted to the command line. If I just create one callback: 

`factory.addTagReportCallback(tagReportCallback)`

Everyting goes fine (even using report_every_n_tags=1). But as I need to access MB:2 I add another callback:

```
factory.addStateCallback(llrp.LLRPClient.STATE_INVENTORYING, access)

def access(proto):
    readSpecParam = None
    if args.read_words:
        readSpecParam = {
            'OpSpecID': 0,
            'MB': args.mb,
            'WordPtr': args.word_ptr,
            'AccessPassword': args.access_password,
            'WordCount': args.read_words
        }

    accessStopParam = {
        'AccessSpecStopTriggerType': 1 if args.count > 0 else 0,
        'OperationCountValue': args.count,
    }

    return proto.startAccess(readWords=readSpecParam,writeWords=writeSpecParam,accessStopParam=accessStopParam)
```

But I'm facing the same issue: I got the tag data, but I still receive tag long time after the tag is out of range. I suspect that the problem is with "accessStopParam"... but not sure how it should be. 

Thank you for reading me, and thank you to the community for this great work.

Román

Reader: Speedway R220
Software Version: 5.12.2.240 (Build cbc9ad1d0d1)

ransford, 2019-09-03:
Have you tried setting `count` to something other than `0`? Does that do anything?

If your CPU is slow, it's plausible that your application isn't draining its input buffer fast enough. If you look at a `DEBUG`-level log, you can estimate how quickly you're consuming bytes from the input buffer and compare to the rate at which they're coming in.

romansapia, 2019-09-04:
Thank you ransford!

Yes, I tried to set count to other than cero. But the issue if I do that is I don't get the OpSpecResult (but only in the first one). Below two responses; first one count=0, and second with count=1, and in both cases with report-every-n-tags=5. Please note that TagSeenCount is 1 in the first example (that is something I don't understand)

> [{u'AntennaID': (1,), u'FirstSeenTimestampUTC': (1567600573219872L,), u'TagSeenCount': (1,), u'PeakRSSI': (-34,), u'EPC-96': '000000000000000090005815', u'LastSeenTimestampUTC': (1567600573219872L,), u'AccessSpecID': (1,), u'OpSpecResult': {u'ReadDataWordCount': 6, u'ReadData': '\xe2\x004\x12\x017\xf8\x00\x0c\x1a\x8dW', u'Result': 0, u'OpSpecID': 0}}]

> [{u'AntennaID': (1,), u'FirstSeenTimestampUTC': (1567600670254329L,), u'TagSeenCount': (5,), u'PeakRSSI': (-37,), u'EPC-96': '000000000000000090005815', u'LastSeenTimestampUTC': (1567600670354317L,), u'AccessSpecID': (0,)}]

Also I think you are right about the CPU. No sure how to check how it is consuming the buffer (below I pasted the DEBUG log if it helps), but the core running the python is at 100%. I'm using a Quadcore Cortex-A53 (Raspberry Pi 3+), it's not a very fast computer, but should handle the load; something isn't right.   
So it's possible to reduce the rate of the reading?... should be done increasing report-every-n-tags. But that only works with count!=0... and if I do that I'm not getting the OpSpecResult which I need to read TID

Thank you again!

Román

```
DEBUG:sllurp.llrp:starting message callbacks for RO_ACCESS_REPORT
[{u'AntennaID': (1,), u'FirstSeenTimestampUTC': (1567600892262609L,), u'TagSeenCount': (1,), u'PeakRSSI': (-31,), u'EPC-96': '000000000000000064221842', u'LastSeenTimestampUTC': (1567600892262609L,), u'AccessSpecID': (1,), u'OpSpecResult': {u'ReadDataWordCount': 6, u'ReadData': '\xe2\x004\x12\x014\xff\x00\x02\x18\xb0\x8a', u'Result': 0, u'OpSpecID': 0}}]
e203412134ff0218b08a 000000000000000064221842 1567600892262609 -31 1
DEBUG:sllurp.llrp:done with message callbacks for RO_ACCESS_REPORT
DEBUG:sllurp.llrp:in handleMessage(RO_ACCESS_REPORT), there are 0 Deferreds
DEBUG:sllurp.llrp:expect 79 bytes (have 12227)
DEBUG:sllurp.llrp:deserializing RO_ACCESS_REPORT command
DEBUG:sllurp.llrp_proto:decode_ROAccessReport
DEBUG:sllurp.llrp_proto:decode_TagReportData
DEBUG:sllurp.llrp_proto:failed to decode EPCData; trying EPC-96
DEBUG:sllurp.llrp_proto:decode_EPC96 (type=13 len=13)
DEBUG:sllurp.llrp_proto:EPC-96: 000000000000000064221842
DEBUG:sllurp.llrp_decoder:found AntennaID (type=1)
DEBUG:sllurp.llrp_decoder:found PeakRSSI (type=6)
DEBUG:sllurp.llrp_decoder:found FirstSeenTimestampUTC (type=2)
DEBUG:sllurp.llrp_decoder:found LastSeenTimestampUTC (type=4)
DEBUG:sllurp.llrp_decoder:found TagSeenCount (type=8)
DEBUG:sllurp.llrp_decoder:found AccessSpecID (type=16)
DEBUG:sllurp.llrp_proto:decode_OpSpecResult
DEBUG:sllurp.llrp_proto:par={u'AntennaID': (1,), u'FirstSeenTimestampUTC': (1567600892309465L,), u'TagSeenCount': (1,), u'PeakRSSI': (-31,), u'EPC-96': '000000000000000064221842', u'LastSeenTimestampUTC': (1567600892309465L,), u'AccessSpecID': (1,), u'OpSpecResult': {u'ReadDataWordCount': 6, u'ReadData': '\xe2\x004\x12\x014\xff\x00\x02\x18\xb0\x8a', u'Result': 0, u'OpSpecID': 0}}
DEBUG:sllurp.llrp_proto:decode_TagReportData
DEBUG:sllurp.llrp:done deserializing RO_ACCESS_REPORT command
DEBUG:sllurp.llrp:LLRPMessage received in state 18: <RO_ACCESS_REPORT>
        <Ver>1</Ver>
        <Type>61</Type>
        <ID>1044721874</ID>
        <TagReportData>
                <EPC-96>'000000000000000064221842'</EPC-96>
                <AntennaID>(1,)</AntennaID>
                <PeakRSSI>(-31,)</PeakRSSI>
                <FirstSeenTimestampUTC>(1567600892309465L,)</FirstSeenTimestampUTC>
                <LastSeenTimestampUTC>(1567600892309465L,)</LastSeenTimestampUTC>
                <TagSeenCount>(1,)</TagSeenCount>
                <AccessSpecID>(1,)</AccessSpecID>
                <OpSpecResult>
                        <Result>0</Result>
                        <OpSpecID>0</OpSpecID>
                        <ReadDataWordCount>6</ReadDataWordCount>
                        <ReadData>'\xe2\x004\x12\x014\xff\x00\x02\x18\xb0\x8a'</ReadData>
                </OpSpecResult>
        </TagReportData>
</RO_ACCESS_REPORT>

DEBUG:sllurp.llrp:starting message callbacks for RO_ACCESS_REPORT
[{u'AntennaID': (1,), u'FirstSeenTimestampUTC': (1567600892309465L,), u'TagSeenCount': (1,), u'PeakRSSI': (-31,), u'EPC-96': '000000000000000064221842', u'LastSeenTimestampUTC': (1567600892309465L,), u'AccessSpecID': (1,), u'OpSpecResult': {u'ReadDataWordCount': 6, u'ReadData': '\xe2\x004\x12\x014\xff\x00\x02\x18\xb0\x8a', u'Result': 0, u'OpSpecID': 0}}]
e203412134ff0218b08a 000000000000000064221842 1567600892309465 -31 1
DEBUG:sllurp.llrp:done with message callbacks for RO_ACCESS_REPORT
DEBUG:sllurp.llrp:in handleMessage(RO_ACCESS_REPORT), there are 0 Deferreds
DEBUG:sllurp.llrp:expect 10 bytes (have 12148)
DEBUG:sllurp.llrp:deserializing RO_ACCESS_REPORT command
DEBUG:sllurp.llrp_proto:decode_ROAccessReport
DEBUG:sllurp.llrp_proto:decode_TagReportData
DEBUG:sllurp.llrp:done deserializing RO_ACCESS_REPORT command
DEBUG:sllurp.llrp:LLRPMessage received in state 18: <RO_ACCESS_REPORT>
[... 22 lines cut ...]
DEBUG:sllurp.llrp_proto:decode_OpSpecResult
[... cut ...]
                <LastSeenTimestampUTC>(1567600892346640L,)</LastSeenTimestampUTC>
                <TagSeenCount>(1,)</TagSeenCount>
                <AccessSpecID>(1,)</AccessSpecID>
                <OpSpecResult>
                        <Result>0</Result>
                        <OpSpecID>0</OpSpecID>
                        <ReadDataWordCount>6</ReadDataWordCount>
                        <ReadData>'\xe2\x004\x12\x014\xff\x00\x02\x18\xb0\x8a'</ReadData>
                </OpSpecResult>
        </TagReportData>
</RO_ACCESS_REPORT>
```

closed 2021-01-24 by ransford