sllurp/sllurp #135: ReceiverSensitivity setting question

djacobson, 2019-09-05:
Quick Q: is the antenna config setting 'ReceiverSensitivity' part of the LLRP standard? And if so, does sllurp set it to any value? Can I set it? If not, what does it default to?

I was thinking maybe it was part of the Impinj extensions, but, it looks like it is part of the LLRP standard antenna config section. Under the following doc.

Read under the following doc:

Octane LLRP Guide
Page 21, section 3.1.7
https://support.impinj.com/hc/article_attachments/360003975300/Impinj_Octane_LLRP.pdf

Thank you for any insight!

ransford, 2019-09-06:
Yes, it's an LLRP (not Impinj) thing. I'd refer you to these parts of the LLRP 1.1 standard (in the `docs/` directory in this repo):
- section 10.2.1, where the reader announces its capabilities, including receiver sensitivity
- section 13.2.6, where the client (e.g., sllurp) sets the sensitivity (per antenna?) to one of the values provided by the reader in the aforementioned blob of capabilities

There's no built-in facility for the user of sllurp to provide an RFReceiver parameter. That's one design wart: it's not easy to assign a value to a single parameter of your choosing out of the bazillions of LLRP parameters. It's a huge protocol! Anyhow, you can look at `LLRPROSpec.__init__()` to see how we currently build a set of LLRP messages based on the arguments provided to this Python function. You could add a parameter to that function and munge the outgoing AntennaConfiguration message accordingly, similar (in some ways) to the way we optionally configure impinj_fixed_frequency_param here: https://github.com/ransford/sllurp/blob/a6079d1681bcd01d643219832498ab12384ebe73/sllurp/llrp_proto.py#L4022

djacobson, 2019-09-06:
Thanks Ransford! I would like to submit this as a feature request. I'll play with a branch myself.

Slightly off topic, but, are you aware of ways to change those persistent default settings on Impinj's?

ransford, 2019-09-06:
Pull request(s) welcome!

I'm not sure I understand your question about persistent settings on Impinj readers.

djacobson, 2019-09-07:
On the Impinj question,  think I found what I was looking for... I believe it is this:

https://support.impinj.com/hc/en-us/articles/202756258-Saving-Reader-Settings

But, I was hoping for a built-in cmd in their ssh shell rather than a programming example (not that I mind that much). I'm still looking at extending the Rospec settings section of sllurp.

closed 2021-01-24 by ransford