sllurp/sllurp #136: Can't figure how to stop the inventory or the reactor.

pBogey, 2019-09-10:
Hello,

I will open a new issue, in hope someone will have an answer. I am using a Zebra FX9500 RFID reader. I don't know if my issue is reader specific or this is just how sllurp works. I am using the follwoing code:

```
# Minimal example; see inventory.py for more.
from sllurp import llrp
from twisted.internet import reactor, defer
import logging
import pprint
logging.getLogger().setLevel(logging.INFO)
logger = logging.getLogger(__name__)

READER_IP_ADDRESS = '169.254.1.1'

def shutdown(factory):
    return factory.politeShutdown()

def tag_report_cb(llrp_msg):
    tags = llrp_msg.msgdict['RO_ACCESS_REPORT']['TagReportData']
    print ('tags:', tags)

factory = llrp.LLRPClientFactory()
factory.addTagReportCallback(tag_report_cb)
reactor.connectTCP(READER_IP_ADDRESS, llrp.LLRP_PORT, factory)
reactor.addSystemEventTrigger('before', 'shutdown', shutdown, factory)
reactor.run()
```

Running this, the code will return an empty tag list (`tags = []`), then the reader will enter an infinite inventory reading loop (`STATE_INVENTORYING` from the LLRPClient), but it seems that no messages are coming from the reader, except for the Keepalive ones (the `handleMessage` method is not ran during this phase). Only after using CTRL+C to interrupt the code, `politeShutdown` will run and call `tag_report_cb`, which prints the inventory. If I comment the `reactor.addSystemEventTrigger('before', 'shutdown', shutdown, factory)`, no tags will be printed and the only output will be the first empty tag list.

How could I enter the STATE_INVENTORYING state and leave at will, or after a defined period of time?

Thanks,
Bogdan Prădatu

ransford, 2019-09-14:
Hello! Try passing a `report_every_n_tags` argument to your reader to have the reader emit a report every N tag sightings. Like this:

```python
factory = llrp.LLRPClientFactory(report_every_n_tags=10)
```

This will behave similar to `sllurp inventory -n 10`.

closed 2019-09-14 by ransford

pBogey, 2019-09-26:
> Hello! Try passing a `report_every_n_tags` argument to your reader to have the reader emit a report every N tag sightings. Like this:
> 
> ```python
> factory = llrp.LLRPClientFactory(report_every_n_tags=10)
> ```
> 
> This will behave similar to `sllurp inventory -n 10`.

Ok, this will flood with reports. Just like time will give a report every n seconds, but it will not stop the inventory process.

How can I stop the process?

ransford, 2019-09-30:
The error you sent makes it sound like the reader didn't support the value you provided in the TriggerType field. Support for different features varies from reader to reader. If you search the LLRP specifications under the `docs/` directory in this repo, you might find something about support for different TriggerTypes.

pBogey, 2019-10-01:
> The error you sent makes it sound like the reader didn't support the value you provided in the TriggerType field. Support for different features varies from reader to reader. If you search the LLRP [...]

Ok, I understand now. I was reading the LLRP v1.1 standard where `TagObservationTrigger Parameter` is defined as:

> TriggerType: Integer
> Possible Values: Value Modulation
>  ------ ------------
>  0 Upon seeing N tag observations, or timeout. The
>  definition of an "observation" is vendor specific.
>  1 Upon seeing no more new tag observations for T ms,
>  or timeout. The definition of an "observation" is
> vendor specific.
>  2 N attempts to see all tags in the FOV, or timeout.
>  3 Upon seeing N unique tag observations, or timeout.
>  4 Upon seeing no more new unique tag observations for
>  T ms, or timeout.

whereas in the v1.0.1 standard the same parameter is defined as:

> TriggerType: Integer
> Possible Values:
>  Value Modulation
>  ------ ------------
>  0 Upon seeing N tag observations, or timeout 
> Copyright © 2005-2007 EPCglobal®, All Rights Reserved. Page 59 of 194
>  1 Upon seeing no more new tag observations for t ms,
>  or timeout
>  2 N attempts to see all tags in the FOV, or timeout

The standard also specifies that each message has a ”Version Field”

> Version Field Protocol Spec Version
> 1 LLRP v1.0.1
> 2 LLRP v1.1

I guess sllurp uses 1 in the version field. Tried to change [line 71](https://github.com/ransford/sllurp/blob/a6079d1681bcd01d643219832498ab12384ebe73/sllurp/llrp_proto.py#L71) from llrp_proto, but had no success. Could you guide me to the setting which defines the LLRP protocol in your code?

ransford, 2019-10-01:
I'm afraid you'll have to experiment to see what your reader supports. LLRP is kind of a mess; you'd certainly expect to see "maximum version of LLRP I support" as one of the Capabilities fields, but it's not there. There may be subtle differences in Capabilities blobs that give you a hint when you connect to a reader, but exploring that would require more time than I have available.

If you wanted to hack sllurp to avoid hard-coding the protocol version, you could certainly make it a per-connection setting in `LLRPProto`.

pBogey, 2019-10-03:
I've implemented a `GET_SUPPORTED_VERSION` message and got 
> Error in message received: unsupported version (108)
So I guess my reader only supports version 1.0.1.

pBogey, 2019-10-04:
I have finally found a way to stop the reactor, after a predefined period of time. After unsuccessfully trying to modify the `startInventory` method of the LLRPClient and adding deferreds, I found a very simple solution:

I've added `reactor.callLater(20, shutdown, factory)` just before `reactor.run()` in the code from my first post.