sllurp/sllurp #137: Connection to the other side was lost in a non-clean fashion: Connection lost

OlegAstrakhantsev, 2019-09-26:
Hey. I ran into an incomprehensible problem. The SLLURP worked and suddenly stopped working. The built-in reader program works all the time. What could it be?

built-in reader program:
[image]

SLLURP consols:
[image]

Log:
INFO:sllurp.llrp:connecting to 95.84.154.133:5084...
INFO:sllurp.llrp:will reset reader state on connect
INFO:sllurp.llrp:will start inventory on connect
INFO:sllurp.llrp:using antennas: (1,)
INFO:sllurp.llrp:transmit power: {1: 0}
INFO:sllurp.llrp:connected to ('95.84.154.133', 5084) (95.84.154.133:5084)
INFO:sllurp.llrp:using reader mode: None
INFO:sllurp.llrp:lost connection: Connection to the other side was lost in a non-clean fashion: Connection lost.

OlegAstrakhantsev, 2019-09-27:
Everything is fine. It was a reader failure. After the reboot, everything worked again.

ransford, 2019-09-27:
Oh good... that would have been hard to debug. :-)

closed 2019-09-27 by ransford