sllurp/sllurp #138: Connectivity isuue when netwok restarts.

hashirsayeed, 2019-09-30:
I am connecting ZEBRA FX7500 RFID reader and running an inventory every second. It works fine but when network restarts, the whole process of reading tags stops and program is unable to read the tags again. I have to restart my reactor manually to read tags again. 
Is there any way i can restart the reactor when the network restarts or any flag the indicates the connectivity status?

ransford, 2019-10-01:
If you're using the built-in `sllurp inventory` command, you can use the `-r` (reconnect) option to attempt to reconnect if the underlying network library (Twisted) notices that the socket has been disconnected.

In general it's difficult for a TCP client to know when the server has abruptly disappeared (which I think is what you're describing); that's more of a [Twisted question](https://twistedmatrix.com/documents/current/core/howto/clients.html#reconnection) than a sllurp question.

closed 2019-10-01 by ransford

hashirsayeed, 2019-10-03:
Thank you for the solution. @ransford