sllurp/sllurp #141: TX power configuration compatibility with Zebra FX9600 reader

avasygn, 2019-10-17:
Hello,

Zebra reader returns this tx_power_table :

```
tx_power_table = [10.0, 10.1, ..., 29.2, 0]
```

Impinj Speedway returns this tx_power_table :

``` 
tx_power_table = [0, 10.0, 10.25, ..., 30.0]
```

With Zebra, 0 (which means max power) is at the end of the tx_power_table list even though it is at the beginning of the tx_power_table. Thus, there is a compatibility problem with the function llrp.py::LLRPClient.get_tx_power(tx_power) : 

- for Zebra, when tx_power (which is an index) is 0, it should set 10.0dBm even though the function will get 0 as the max power to set. 
What do you think about the following fix:

``` python
def get_tx_power(self, tx_power):
     [...]
     for antid, tx_power in tx_power.items():
          ### version to fix
          if tx_power == 0:
               # tx_power = 0 means max power
          ### new version
          if tx_power == min_power:
               # tx_power = min_power index (index's value is 0) means max power

               ### no changes from here
               max_power_dbm = max(self.tx_power_table)
               tx_power = self.tx_power_table.index(max_power_dbm)
               ret[antid] = (tx_power, max_power_dbm)
```

ransford, 2019-10-17:
Thank you for finding that issue! A PR to fix this problem would we welcome.

This part of reader configuration is tricky because the LLRP client
(sllurp) has to tell the reader "use the following index into an array you
gave me." But the user doesn't know this index in advance unless they've
previously inspected the transmit power array from the reader.

avasygn, 2019-10-18:
Thank you for answering! I created the PR here: #142

closed 2021-01-24 by ransford