sllurp/sllurp #148: how to read data from sensormatic IDX-2000?

crssulton, 2019-12-04:


crssulton, 2019-12-04:
[image]
I can connect, but how do I get the data?

ransford, 2019-12-04:
The `inventory` command is just for demonstration purposes. If you run `sllurp inventory -n1 192.168.200.110:21`, do you see tag reports?

(Also, 21 is a strange port for LLRP -- 5084 is the standard port -- but maybe you have your reader configured differently.)

crssulton, 2019-12-26:
Oh I understand. but, when I try again and find a new error.. how with this?

crssulton, 2019-12-26:
[image]

ransford, 2019-12-31:
What revision of sllurp? `git describe --always` in the sllurp directory will tell you what version you are using.

avasygn, 2020-01-27:
Hello @crssulton,

Maybe your reader does not respect LLRP strictly because the error occurs when sllurp try to decode a reader response. In my case, I cannot use sllurp with Kathrein reader although it should be compliant with LLRP. But I can drive Kathrein reader from the software of the manufacturer (which seems uses LLRP).

Try to run Wireshark to know more about the response of the reader which lead to an error.

crssulton, 2020-02-24:
> Hello @crssulton,
>
> Maybe your reader does not respect LLRP strictly because the error occurs when sllurp try to decode a reader response. In my case, I cannot use sllurp with Kathrein reader [...]

thanks you

closed 2021-01-24 by ransford

seba30, 2026-09-14:
Maybe see #160?