sllurp/sllurp #149: Big refactor of Sllurp as a possible Sllurp v2?

fviard, 2019-12-09:
Hi @ransford ,

For a few weeks, I have worked on a big refactor of the encoding/decoding of LLRP in Sllurp, going one huge step further than my "untwisted" branch.
https://github.com/fviard/sllurp/tree/develop_untwisted_next

I hope, that you can get an idea of the biggest part of the changes by looking at llrp_proto.py.
Now, in my opinion, it should be more easier and faster to add encoder or decoder for new fields, including "custom" ones.

The performance is also even a lot better than the "untwisted" branch.
For example, for the Impinj R420, i can go from around 1100 tags/s to 1200+ tags/s.
Testing the speed of decoding ROREPORT msg on my computer only, it goes from 10k tags/s (untwisted) to around 45k tags/s.

@thijmenketel I think you might be interested to try this branch also :-)
(Note: the self['ROSpec']['ROReportSpec']['N'] = report_every_n_tags hack is still needed for the moment, if you still want to test the max performance)

So, @ransford , if you agree with that, my plan would be to push this branch to your repo in a staging branch like "fviard-sllurp-v2", or something like that.
And then let it be developed and battle tested from there, in order to become the official "v2" once good enough?

ransford, 2019-12-09:
This is very cool. That performance improvement sounds amazing.

I see a lot of new stuff in llrp_proto.py that's easy to read. It seems
like you've preserved the basic structure we had before, where message
structs represented as dictionaries carried their own pointers to
encoder/decoder methods.

Not quite sure how I feel about the fact that it spawns a thread on the
user's behalf via a function called `connect()`, as it's easy to lose track
of a thread you didn't know was created. How difficult would it be to let
the user provide the thread (perhaps require that they call `connect()`
from it)?

I want to bring this to @ukrutt's attention too as another contributor.

fviard, 2019-12-09:
By the way, I forgot to add that I have implemented the whole set of parameters to decode get_capabilities and get_config.

Yes, I kept the overall idea of the basic structure, but I separated the Message_struct and a Param_struct, so there is not any more a conflict of "type" ids.
The best things would have been to transform each "fields" entry in a kind of tuple list, or something with a dict content for each "key" to be able to save the additional info (like if it is a simple or multiple entry field, ...).
But I did not do that to keep it easy to read and to add fields.

Also, I have changed so that, "multiple entries" fields, will correspond to a "list" on the par dict side, instead of a direct dict entry.
So that we can support things like "multiple AISpecs" in the ROSpec.

For the thread creation, that looks like standard to me for things interacting with devices (serials, ...). Probably, that should be documented.
I don't think that it would be a great idea to get a thread from the user.

I did think about the question, and looked if it was possible to ensure that the sllurp thread will exit automatically if the user main_thread exit.
Sadly, it is not possible to do that in a clean and reliable way.
- Thread could be "daemon", but that is not great that there is a big chance that sllurp thread/connection does not have time to gracefully exit most of the time.

Still, I pushed a few improvements:
- First, "sllurp" and reader host/port are added to the thread name, so they can easily be identified.
- I added a  way to have a "blocking" disconnect behavior when needed.
- I added the staticmethod: disconnect_all_readers() so that can be used by the user script as "cleanup" instead of having to keep track of each reader instance to disconnect them individually.

thijmenketel, 2020-01-15:
@fviard @ransford This looks very promising indeed! My apologies for the absent on this repo, I've finished my MSc degree and unfortunately have no access anymore to a reader. This makes further collaboration somewhat harder!

I'm looking to rework the GUI I made and add it to sllurp as a simple front to start and stop inventories and export tagdata. It's not the nicest solution but it works, I'll try to get it done soon.

fviard, 2020-01-22:
@thijmenketel Even if imperfect, that would still be nice. I will be testing for you if that can help.

ransford, 2020-01-27:
UI would be awesome! I'm rearranging a home network and would be happy to expose the LLRP port of an Impinj reader if that would be helpful.

fviard, 2020-03-11:
I feel that my "untwisted" version is good enough to be upstream.
So I have pushed it into a new branch into @ransford repository:
https://github.com/ransford/sllurp/tree/fviard-develop-v2

That, if it fits to you, should be the new target for devs and pull requests.
This sllurp has the package version 2.0.
:smiley: 

@thijmenketel Don't forgot your GUI, I'm impatient to be able to test it one day !

closed 2020-03-11 by fviard

avasygn, 2020-03-12:
Hello @fviard , I can design a GUI. I get use to use pyqt but I can try to design a pure python GUI with tkinter. What would you prefer ?

fviard, 2020-03-12:
@avasygn Would be nice to have, but in theory, @thijmenketel did one already.
So, would be great to hear from him, if it is still its plan to share it, to avoid duplicating the work :-p

Regarding the GUI framework, I think @thijmenketel did it with pyqt.
Otherwise, i have no opinion on which one is the best:
- tkinter: lighter, very few dependencies, easier to package with cxfreeze, but lots of frustrations.
- pyqt: allows to do more nicer things and more features in the GUI

fviard, 2020-04-27:
@avasygn @thijmenketel : If you are interested to contribute to develop a gui,
@papapel has submitted one that I have merged on my side:
https://github.com/fviard/sllurp/tree/fviard-develop-v2/examples/qtgui

I would like to pep-8 clean it a little before pushing it here, but I think that, as simple as it is, it is already working well :-)
But, from that base, I think a lot of improvements can already be done ^^

ransford, 2020-04-27:
Since sllurp is meant to be a simple library for reader control that acts as a conduit between tag populations and arbitrary applications, I would suggest putting GUI code into a separate project that imports sllurp as a dependency. That would allow you to change the GUI in any way you want without affecting projects that use sllurp as a (non-graphical) library.

In the past, it was difficult to use sllurp on Windows because of Twisted; I would have similar concerns about any new dependency on a GUI framework.

papapel, 2020-04-27:
Hello,

What about the GUI within the example folder with a readme to explain which dependencies are required? It does not affect the sllurp dependencies. But if the GUI is leaded to grow-up, it is a good/ the best suggestion.

Also, avasygn and papapel, it's me, I create a non-professional account (fviard, I forgot to notice you:slightly_smiling_face:)

fviard, 2020-04-27:
@papapel Ahhh, so that makes more sense!

@ransford On the global idea I agree with you, but so I think that it is quite good to be in the example folder like it is. (Renamed from gui to qtgui also btw).
So, it is not only a possible gui, but also an example of how to use sllurp with Qt. In similar way that someone can add a Tkinter gui.

Also, the dependencies requirements needed for that are also kept located within the examples/qtgui folder also.
In a similar way that it is done for the tornado example.
The goal being, as I like it, like you, that sllurp dependencies themselves stay minimal, like now, with only the minimum needed by sllurp itself.

So it is not, the sllurp gui, or something that has to be installed like the "sllurp cmdline", but an example.
In a first time, that would be lot easier to modify it little by little at the same time as sllurp lib is modified.
But, if it evolves well, and give something that is really "polished"/functional/stable, probably it will be better to move it to a dedicated project.

@ransford What do you think about that logic?

thijmenketel, 2020-06-27:
@ransford @ransford I'm really sorry about the long silent period the past months, I did not have much time aside from working.

On my fork of [fiard-develop-v2](https://github.com/thijmenketel/sllurp/tree/fviard-develop-v2) you can find some files that can form the gui. Unfortunately I've not had the change to test with the latest version of @fviard 's version of SLLURP and as I don't have any access anymore to a reader I'm not able to do so anymore. I would ask you guys to try and see if you like the gui and way I've build it. I do realize that the code is quite messy and maybe even a little hard to read, so if you have questions I might be able to answer them here!

To run please use the run_gui.sh script, it build a venv and removes it after running (did it like that while developing, not the nicest or efficient way I know).

I hope this gui serves as an addition to the SLLURP repository and can help people with less experience use it.

fviard, 2020-06-29:
@thijmenketel Thanks for sharing your work, even if not in a perfect state, it is a nice contribution.
If I find time, I will try to see how we can merge your gui with the one of @avasygn ^^

ransford, 2020-06-30:
I really do think the GUIs are cool, and I'd like to link to them from the main sllurp README so that people can find them. I am still not inclined to merge UI code into the sllurp library, however. Think of sllurp (the library) as the plumbing underneath your excellent projects.

papapel, 2020-07-01:
Hello all,

I run @thijmenketel gui. The run_gui.sh works well. I will test it with tags later.

I am OK to contribute to merge the guis with @fviard (I can work on inventory and tag memory access but I have never worked on phase rotation, frequency doopler etc.). But before, it seems we should take a decision:

- gui is an example of sllurp, within sllurp project

or

- gui is a sllurp based external project

thijmenketel, 2020-07-01:
One of the changes I forgot to make to the gui code is removing the calibration mode for the phase angle which is only relevant for frequency hopping readers and doesn't really work that well. For my MSc thesis I used a whole different way of compensation for the hopping. 

Furthermore, this calibration mode uses a start inventory callback ([line 758](https://github.com/thijmenketel/sllurp/blob/c1966ac55d66c7c8fa00c33b0de3e84fd66316c6/gui/sllurp_gui.py#L758)) where I extract the hoptable from the reader, I don't know if that has been implemented in main sllurp or I hacked it in some time ago so I might remove it at all.

ransford, 2020-07-02:
@papapel this is the better path:
> gui is a sllurp based external project

fviard, 2020-07-02:
@papapel Thanks.
Yes, the best would be to create a new project.
Except if you want to do it, i will try to create it with the basic merge of the two to start.

papapel, 2020-07-02:
Sounds good. You can create the new project.

fviard, 2020-07-28:
So, I created the new project and went down the rabbit hole to merge the two GUIs and improve that:
https://github.com/fviard/sllurp-gui

I think that it starts to work well for basic features even if the code is still not so clean and so much could still be done.
Performance are still quite good in my opinion as I can still saturate a reader and history can be enabled for some time without issue.

There is still the original gui of @thijmenketel around in the repo as the "export" feature was not backported yet.

The most important basic remaining thing, in my opinion, is to rework the backend code of configuration to be separated from display, and to be able to save and restore it.
What I have in my that would be nice is to:
- Have multiple profiles and a list box at the top to select which one to use.
- A few buttons next to the check list for "add a new profile" (ie config/reader), "update/overwrite the current profile with current settings".

@thijmenketel I did not look in details at the algorithms, you created, but by testing the output graph of them, they look to be very nice!

Here are a few screenshots of the integrated graphes:

(In the following graphes, the tags are statics and then briefly moved a few times.

[image]

[image]

[image]

[image]

thijmenketel, 2020-08-02:
@fviard I see you made something very nice out of the bowl of spaghetti I left! 

Thanks for the compliment, it took some time to figure out how to work with the values the reader spits out so its nice to read that the algo's are still doing their job.

If you have any issues with exporting data (which I assume you will not have given your experience) I'm open for questions. 

I'll be looking here and on your repo for updates as it's looking to be a nice addition to SLLURP!